pvvx/ZigbeeTLc #101: Can't set Zigbee Attributes

harrismck, 2024-06-04:
According to the docs, I should be able to set min/max comfort temp/humidity values via Zigbee attributes, but those attributes don't seem to exist. In ZHA, looking at the UserInterface cluster, I only see these three attributes:
[image]
I tried setting the value manually with the zha.set_zigbee_cluster_attribute service call, but I get this error:
`Unexpected error for call_service at pos 1: Cluster 204 not found on endpoint 1 while writing attribute 102 with value 1800`
[image]

My device is a LYWSD03MMC-z by Xiaomi, and I have just flashed the latest firmware

pvvx, 2024-06-04:
ClusterID: 0x0204 (HEX) -> 516 (DEC)
Attr: 0x0102 (HEX) -> 258 (DEC)

```
service: zha.set_zigbee_cluster_attribute
data:
  cluster_type: out
  value: "1800"
  ieee: a4:c1:38:xx:xx:xx:xx:xx
  endpoint_id: 1
  cluster_id: 516
  attribute: 258
```

harrismck, 2024-06-05:
Ah, of course the HA values aren't in hex, woops.
However, even after fixing that, I still get the error `Cluster 516 not found on endpoint 1 while writing attribute 258 with value 1800`
I tried your yaml for ZHA exactly (except the ieee).

I got ZHA toolkit, and the response had the error: `DeliveryError('Failed to deliver message: <EmberStatus.DELIVERY_FAILED:
    102>')`

pvvx, 2024-06-05:
[image]

[image]

harrismck, 2024-06-05:
[image]
[image]
Hmmm

harrismck, 2024-06-07:
I haven't changed Measurement Interval. Checking it with ZHA toolkit's read attribute, it's set to 10.
I'm using a HomeAssistant SkyConnect (AKA a CP2102N from Silicon Labs).

Bizarrely, I was able to set the min comfort temp just now through ZHA toolkit.... by removing the quotes around the value. Confirmed it has been set by reading the attribute.
Doesn't explain why the ZHA service call doesn't work, nor why these attributes don't show in the Manage Zigbee Device UI.

Here's an example of succesfully setting it to 17 degrees:
[image]

pvvx, 2024-06-07:
>Doesn't explain why the ZHA service call doesn't work, nor why these attributes don't show in the Manage Zigbee Device UI.

These attributes are not standard in the Zigbee 3.0 specification.

harrismck, 2024-06-08:
Ahhh that makes sense! Thanks. Not entirely sure why removing the quotes around the value was important, but hey, it works.

closed 2024-06-08 by harrismck