pvvx/ZigbeeTLc #102: Addr of lcd and sensor lost

dodgerz2000, 2024-06-06:
[image]
[image]
After flashing the firmware with ZigBee with the display turned off, after a while the sensor stopped working, although the device itself responded to requests, it also became impossible to turn on the display, the command was accepted but the display did not turn on. After flashing it to ble, it became clear that the device had lost the addresses of the display and sensor.  Can this be fixed somehow?

pvvx, 2024-06-06:
Most likely this is a hardware problem. Look at the printed circuit board and the soldering of the elements.

dodgerz2000, 2024-06-07:
Everything is fine with the contacts, the sensor worked for several years before. I also noticed that the hw version in the firmware information and on the board is different.  1.4 on the board and 1.6 in the information.
[image]

pvvx, 2024-06-07:
The HW version is determined by the I2C addresses of the sensor and display controller.

Write the device’s response to https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html to command “02”:
[image]
```
15:19:05: Hardware Revision String: B1.4
15:19:05: Software Revision String: V4.4
15:19:05: Firmware Revision String: github.com/pvvx
15:19:05: Detected custom Firmware
15:19:05: Hardware Version: LYWSD03MMC B1.4, Software Version: 4.4, Sensor: SHTC3 (SHTV3)
15:19:05: Custom config HEX string: 55051000004004a9313180e1
15:19:11: Settings 02 was sent successfully
15:19:12: Sensor: I2C addres 0x70, LCD driver: I2C addres 0x3C
```
And in the "console" (Chrome F12) using the command "03".
[image]

pvvx, 2024-06-07:
Specially checked on HW B1.4:
1. Turned off the display and turned off - turned on the power.
2. Connecting and turning on the display. Disconnect.

Everything is fine.

---

Consumption in sleep mode with display off:
[image]
Consumption in sleep mode with display on:
[image]