pvvx/ZigbeeTLc #105: LYWSD03MMC B1.5 Zigbee problem

BmdOnline, 2024-07-02:
Hi,
I've ordered 5 devices LYWSD03MMC (Hardware B1.5).
I've successfully replaced original BT firmware 130 with Zigbee one 1.20 for 4 ones.
I'm using them with Home Assistant / ZHA.

But, I was unable to upgrade the 5th one :
I've replaced original firmware with Zigbee 1.20 through OTA with success, but I was unable to add it to Home Assistant.
It wasn't detected.
I've tried with FW 1.20 or 1.21. I've tried through OTA, I've tried with UART connection... Nothing works.
After upgrade, I was unable to pair/detect the device to Home Assistant / ZHA.

I don't know what to do.
Hardware issue ? Any way to diagnose ?

Thanks.

pvvx, 2024-07-02:
Have you tried using a USB-UART programmer to erase all Flash and then write any firmware?

BmdOnline, 2024-07-02:
Yes, I've tried too.

BmdOnline, 2024-07-02:
I've tried again :

erase flash
disconnect / reconnect device. Having a blank screen
flash firmware, tried with several Baudrate (from normal to lowest), several Atime (from normal to highest)
disconnect / reconnect device. Having a normal screen

but I was unable to pair device with ZHA.

pvvx, 2024-07-02:
> Hardware issue ? Any way to diagnose ?

To check the thermometer, you can flash the official firmware from Xiaomi and check the BLE connection.
https://github.com/pvvx/ATC_MiThermometer/raw/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin

BmdOnline, 2024-07-03:
Hi,
I've installed original firmware, used Xiaomi Home and paired successfully the device.
App seems OK. Temp & Hum are displayed on my phone.

I've tried OTA upgrade from Xiaomi Home, from official v1.0.0_0130 to v2.1.1_0159.
Upgrade done, app seem OK too.

pvvx, 2024-07-03:
Zigbee coordinators have a limit on the number of devices they can support. (?)

There is also this firmware: https://github.com/pvvx/BZdevice

BmdOnline, 2024-07-03:
In fact, I've tried to pair this device as my 4th temperature device. And because of these problems I've paired my 5th one first.

Now, I'm trying to make this (the 4th one) works.
I don't think it was caused by number of supported devices.
I'm trying to analyse ZHA debug log, but can't see any pertinent data.

> There is also this firmware: https://github.com/pvvx/BZdevice

I will try this firmware.

BmdOnline, 2024-07-03:
> There is also this firmware: https://github.com/pvvx/BZdevice

BT is working with this firmware (ATCzB31BB9 is detected) , but not Zigbee.
It really looks like I have a hardware problem with this device.

pvvx, 2024-11-14:
> I have one of my devices which only connects to Zigbee once it is less than 5 cm away from coordinator or router

Zigbee is an intermediate modulation for BLE.
BLE has a PHY type of 2M, 1M, 125K (Coded PHY).
Zigbee has only 250K.
Most likely, you just have a defective device.