pvvx/ZigbeeTLc #107: Another TY0201 by _TZ3000_bjawzodf with display

diggit, 2024-07-12:
Hi,
I have this Zigbee sensor with display: `TY0201 by _TZ3000_bjawzodf`

Module: ZTU
Sensor: AHT20 o flex cable
Display driver: [BL55072](https://www.belling.com.cn/media/file_object/bel_product/BL55072A/datasheet/BL55072A_V1.02_en.pdf)

SWS is not available on any test point or trace so disassemble is necessary for flashing. 

Original FW readout (read out 2 times and it matched): [TY0201-by-TZ3000_bjawzodf_0x10013001.zip](https://github.com/user-attachments/files/16195805/TY0201-by-TZ3000_bjawzodf_0x10013001.zip)

<details>

<summary>Photos - click to expand</summary>

(On top left, there is also signal bar graph - not connected at photo)
[image]
Position of plastic locks. Best way to start opening the case is from the bottom with thin metal shim. Once the first lock opens, continue with plastic prying bar all around.
[image]
[image]
[image]

</details>

I assume next steps would:
- implement display driver
- find segment mapping
- find button and LED GPIOs

I am a bit lazy to setup whole TLSR8 IDE and write pure C code, but I can definitely find out those GPIOs...

pvvx, 2024-07-13:
The ZTU module uses a TLSR8258 chip with built-in 1 Megabyte SPI-Flash memory.

diggit, 2024-07-13:
1 - updated first post with full 1M dump, I am using `TlsrComSwireWriter` so it may not be 100% reliable
 2 , 3 - thanks, good resources where to start
 4 - what is the impact of this?

I'll try to build do some FW experiments later, but first I have to re-check pinout. Those i2c pins don't make sense.

pvvx, 2024-07-14:
> Module: ZTU 

https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp
```
ZTU is embedded with a low-power 32-bit CPU, 1024-KB flash , 64-KB RAM, and rich peripheral resources.
```
But you have a ZTU module with 512 kilobytes of Flash! 

> 4 - what is the impact of this?

The checksum only indicates that some other SDK was used to build the firmware. Most likely outdated.

Typically Tuya's Zigbee firmware comes with BootLoader. The main program downloaded via OTA is then located at address 0x8000. This affects the fw OTA build type.

pvvx, 2024-07-14:
OTA bin for [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) with adjusted checksum.
[TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip](https://github.com/user-attachments/files/16216483/TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip)

diggit, 2024-07-14:
> OTA bin for [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) with adjusted checksum.
[TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip](https://github.com/user-attachments/files/16216483/TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip)

My device has Zigbee FW, so I guess [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) is not usable here.

What does this OTA contain?

It also looks like this one will need some HW modifications too. i2c seems to be on pins which can't be switched to i2c mode if I read datasheet correctly. There is footprint for some similar chip like in ZTH01/ZTH02 have, but not populated. To connect sensor and LCD driver to the normal i2c pins, purple connections would have to be added.
[image]

pvvx, 2024-07-14:
First you need to make BLE firmware.
You can't debug anything in Zigbee.

TheNitek, 2024-12-02:
I have one of those bricked after a (failed) zigbee2mqtt OTA update. Should a recovery work via https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html using the binary from the first post?

pvvx, 2024-12-03:
There is 1MB - it takes too long to flash it using such a programmer.
And the MAC address of the device will be the same as **diggit**.

selvalt7, 2024-12-08:
I can confirm that by bridging pads like in the picture below, results in working temperature functionality. 
[image]
It also requires modifying firmware to use the correct pin for the button ( B4) and the correct sensor (AHT2X3X).
I didn't bother to add display functionality, so it stays blank.

> It also looks like this one will need some HW modifications too. i2c seems to be on pins which can't be switched to i2c mode if I read datasheet correctly. There is footprint for some similar chip like in ZTH01/ZTH02 have, but not populated. To connect sensor and LCD driver to the normal i2c pins, purple connections would have to be added. ![TY0201_by_TZ3000_bjawzodf_i2c](https://private-user-images.githubusercontent.com/512025/348553070-7dd3cfb3-0c4a-42bd-934a-d79d21d19462.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM2NjEwMTUsIm5iZiI6MTczMzY2MDcxNSwicGF0aCI6Ii81MTIwMjUvMzQ4NTUzMDcwLTdkZDNjZmIzLTBjNGEtNDJiZC05MzRhLWQ3OWQyMWQxOTQ2Mi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwOFQxMjI1MTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ODkzODY3ZjhkOTNjYjhkMzViMGJlM2Q5MDdmOTQ3NWZhMzIxMzc4NTIwMDRiNTY3N2JhYjhhNTJhMzYwZjZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.hzNnEDIDvP-n3A_B_YndgzVhbHqR16GlZ9-fkY0dAVM)

pvvx, 2024-12-15:
Flashing the file provided by @diggit into the ZTU module showed that the file does not work!

diggit, 2024-12-15:
Interesting. I did several matching readouts using uart method and your script, I even fixed one timing issue and backed up and successfully restored other ZTU based sensor this way. (Did not try it with this one though.) Mabe you have different HW or there is some device specific part or dump ma be corrupted as you say.

mechanig, 2024-12-27:
Hi, @pvvx  can you check if the dump I attached is working? This is a dump from a brand new device, made via **TlsrComSwireWriter**. I have a broken device due to an upgrade from z2m. And I can't get it to work when I deploy this dump.

[TY0201_TZ3000_bjawzodf.zip](https://github.com/user-attachments/files/18263766/TY0201_TZ3000_bjawzodf.zip)

pvvx, 2024-12-28:
The firmware checksum does not match tl_check_fw2.exe.
The firmware can be tied to the Flash serial number. This has happened with Zigbee before.

pvvx, 2024-12-28:
[TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip](https://github.com/user-attachments/files/16216483/TY0201-by-TZ3000_bjawzodf_0x10013001_ota.zip) and [TY0201_TZ3000_bjawzodf.zip](https://github.com/user-attachments/files/18263766/TY0201_TZ3000_bjawzodf.zip) firmware behave the same.
They output some signal to PA1 and that's it.

ouinouin, 2025-01-07:
@pvvx I have a bit of time and a lot of those devices to test, i bought some adapters to flash with TB 03F like shown here : https://github.com/doctor64/tuyaZigbee/blob/master/docs/flash.md .
i lack competencies for coding, but i m good at logic analyzers/ scopes , which firmware to flash and what are the steps to find the LCD mapping with this firmware ?

pvvx, 2025-01-08:
It is necessary to create a new firmware with BLE. Taking into account the PCB layout (used GPIO).
FW BLE has a function to create a map of screen segments and other debugging functionality.
This is not possible on Zigbee.

closed 2026-01-30 by pvvx

votblah, 2026-01-31:
Successfully OTA-updated one more of my sensors from original Tuya firmware to ZigbeeTlc using [1141-0203-10923001-ZTY0201z_v0134.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-0203-10923001-ZTY0201z_v0134.zigbee). The top button shall be continuosly clicked during OTA for it to progress (~15 minutes).

**NOTE**: There are (at least) 2 versions of this sensor. Do not go for OTA blindly without checking what [version you have](https://pvvx.github.io/TY0201_TZ3000_bjawzodf/), unless you are prepared, if you are unlucky, to do some soldering to return it to working condition.