pvvx/ZigbeeTLc #108: Another ZTH01

diggit, 2024-07-12:
Hi,
I just wan to let you know, I have found slightly different ZTH01 than on pictures [here](https://pvvx.github.io/TS0601_TZE200_zth01/).
It reported av `TS0601` by `_TZE200_yjjdcqsq` IIRC.
FW was flashed via SWS.
Extra chip bypass can be seen here
[image]

pvvx, 2024-07-13:
https://pvvx.github.io/TS0601_TZE200_zth01/

closed 2024-07-13 by diggit

DChe47, 2024-09-26:
Hi
I have several of these devices and they do not work very well. Sometimes they disappear from the zigbee network ("freeze").
This project interested me. How do these devices work after the change?
What are the recommendations? What to do first to remove the small chip or change the software?
Is it programmable via RS232? How to connect for programming? Which software tool should I use to download software?
Thanks in advance to everyone.

pvvx, 2024-09-26:
> How do these devices work after the change?

Statistics are still small.
ZTH01 work on the street:
[image]
[image]

> Is it programmable via RS232? 

Yes.

pvvx, 2024-09-26:
> How do these devices work after the change?

The program and modification are similar to [ZTH02](https://pvvx.github.io/TS0601_TZE200_zth02/).
But ZTH02 has a different battery.
But my ZTH02 works on [BLE firmware](https://github.com/pvvx/ATC_MiThermometer), in "LE long range" mode. Because it is located far away in the barn and Zigbee connection does not reach there.
[image]

DChe47, 2024-09-27:
Thank you for the answer!

pvvx, 2024-09-27:
> Why are you removing the little chip? What does it change?

This chip is connected to sensors, and it transmits data to the ZTU module via UART interface.
It transmits truncated data, at low speed and whenever it wants - once an hour. At the same time, it consumes energy as the ZTU module itself.

DChe47, 2025-01-06:
I'm already trying to modify the second device, but the result is bad.
First, I make changes to HW as shown in the picture. Then I solder four additional wires to BAT-, BAT+, SWS and RST.
To download the software, I use this: https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
I'm connecting: GND -> -Vbat, +3.3V -> +Vbat, TX -> SWS.
I choose the speed 2000000 baud, Firmware file: ZTH01Z_v0123.bin and my COM3.
Then I consistently made: Erase All Flash, Write to Flash, Soft Reset MCU. 
At each of these three steps, at the beginning, I did a hardware Reset by closing the RST at +3.3v.

Here is the result:
14:26:37: File was selected, size: 126244 bytes
14:26:49: USB-COM opened.
14:26:53: Reset DTR/RTS (100 ms)
14:26:53: Activate (5 sec)...
14:26:58: Flash Erase All (3.5 sec)...
14:27:01: Done.
14:27:06: Reset DTR/RTS (100 ms)
14:27:06: Activate (5 sec)...
14:27:11: Write 126244 bytes in to Flash...
14:27:40: Done (34.149 sec).
14:27:40: Soft Reset MCU
14:27:56: Reset DTR/RTS (100 ms)
14:27:56: Activate (5 sec)...
14:28:01: Soft Reset MCU
14:28:01: Done.

Before starting the software download, the LED on the sensor was on. 
After the download is finished, I disconnect the sensor from the USB-COM converter and turn on the normal power (batteries). When the button is pressed, the LED does not blink and the Join does not work.
What am I doing wrong and why might it not work?