pvvx/ZigbeeTLc #109: MHO-C122: No info on LCD

FaBjE, 2024-07-26:
I've updated one of my MHO-C122 to the ZigBee firmware as a test.
After the upgrade there is nothing on the LCD. 

I did a quick diff between the ZigBee LCD driver:
https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/src/lcd_mho_c122.c
and the BT LCD driver: https://raw.githubusercontent.com/pvvx/ATC_MiThermometer/master/src/lcd_mho_c122.c

I noticed some differences.
- Datatype name differences (should not matter)
- The "send_to_lcd" function is now a "critical section" (isr are disabled) 
- [LCD init delay is disabled](https://github.com/pvvx/ZigbeeTLc/blob/master/src/lcd_mho_c122.c#L134) Is this on purpose or?

The big question is: Should this be working (theoretically) or is the LCD not implemented yet in this beta firmware?

Upgrade log:
```
12:21:23 PM: Searching for devices
12:22:33 PM: Connecting to: MHO-C122
12:22:44 PM: Hardware Revision String: V1.1-202106
12:22:44 PM: Software Revision String: V4.6
12:22:44 PM: Firmware Revision String: github.com/pvvx
12:22:44 PM: Detected custom Firmware
12:22:44 PM: Hardware Version: MHO-C122 V1.1-202106, Software Version: 4.6, Sensor: SHTC3 (SHTV3)
12:22:44 PM: Custom config HEX string: 55051000002804a931318bb4
12:23:10 PM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZMHOC122_v0122.bin'...
12:23:11 PM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZMHOC122_v0122.bin
12:23:11 PM: File size: 128756 bytes
12:23:11 PM: Count: 8048
12:23:14 PM: Start DFU
12:24:25 PM: Update done after 70.282 seconds
12:24:29 PM: Disconnected.
```

(Maybe to refresh your memory, I added the [LCD driver for the MHO-C122](https://github.com/pvvx/ATC_MiThermometer/issues/339)
The BLE firmware is still working very well!))

pvvx, 2024-07-26:
I don't have MHO-C122 and have no way to check its functionality.

FaBjE, 2024-07-27:
Thanks for your response. 
I tried setting the "screen off" attribute to 0.

```
# Display On
service: zha.set_zigbee_cluster_attribute
data:
  ieee: 'a4:c1:38:94:93:??:??:??'
  endpoint_id: 1
  cluster_id: 0x0204
  cluster_type: in
  attribute: 0x0106
  value: 0

```
It was confirmed as successful
```
DEBUG (MainThread) [zigpy.zcl] [0xC86E:1:0x0204] Decoded ZCL frame: UserInterface:Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)])
DEBUG (MainThread) [homeassistant.components.zha.core.device] [0xC86E](MHO-C122-z): set: 0 for attr: 262 to cluster: 516 for ept: 1 - res: Write_Attributes_rsp(status_records=[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)])
```
But no change unfortunately.

> On [C122 I don't know](https://github.com/pvvx/ZigbeeTLc/blob/master/src/board_mho_c122.h#L43) what elements are soldered on the printed circuit board on I2C bus

On the PCB I can not identify any diode or resistor. Only capacitors and a few inductors: https://pvvx.github.io/MHO_C122/
Any limits will be in the IC's itself.

But apart from that, the I2C speed is [still configured at 100 Khz](https://github.com/pvvx/ZigbeeTLc/blob/master/src/board_mho_c122.h#L43) as far as I can see. So that should not be a problem.

Unfortunately I am quite limited in my debug options here. (I don´t have a debugger) or any connection points to connect a logic analyzer to the I2C bus.
How would you debug an issue like this? Is there any debug UART/console available?

> The [display controller is similar to LYWSD03MMC (B1.4, B1.7, B2.0)](https://github.com/pvvx/ZigbeeTLc/blob/master/src/lcd_03mmc.c#L242). The only difference is the LCD panel segments.
When porting the MHO-C122 I just flashed the LYWSD03MMC and by trial and error I was able to map the LCD.
As I see nothing now, I think there is more of a low-level/generic issue.

If I may make a suggestion/request: If you have some spare time, could you maybe load the MHO-C122 build into your LYWSD03MMC and see if any LCD segments come up?
In my opinion this should be the case.

pvvx, 2024-07-28:
Error found. When copying from BLE, the variable name "lcd_i2c_addr" was not changed to "i2c_address_lcd".
ZigBee files for MHO-C122 have been replaced.

FaBjE, 2024-07-28:
Great work @pvvx !

After flashing the latest binary it didn't work for me.
I did some analyzing, changed some code, sw bricked device, erased all flash, switched back to BT firmware (LCD is still working) and back to zigbee.
Now it is working!

[image]

I think the erase all flash in combination with the new SW did the trick. (Maybe the display was disabled after my tinkering to change the settings)
I will monitor it's behavior in the coming days.

FaBjE, 2024-08-05:
After a week it is still working fine 👍🏻 

I have one feature/functionality suggestion.
At the moment the "bluetooth bond `(|)` = Segment: 3.6" is always displayed.
The bit is set on the reboot-screen, but never cleared.

Maybe it is an idea to use: "`(|)` = Segment: 3.6" for the zigbee network state. 
and "`BLE` = Segment 4.7" for the bluetooth state.

For (also for [BZdevice](https://github.com/pvvx/BZdevice)) you would be able to see which one is connected.

LCD's that don't have the symbol can just use the `BLE` icon for both.

closed 2024-12-15 by FaBjE