pvvx/ZigbeeTLc #113: Update for TH03Z is targeting wrong device

tr4nt0r, 2024-08-17:
The firmware `1141-d3a3-01993001-TH03Zz_v0122.zigbee` for the TH03Z is shown as update for the wrong device in Home Assistant. I don't have a TH03Z , but I have a [Tuya door/window sensor with illumination sensor](https://de.aliexpress.com/item/1005004990682942.html).  ZHA identifies it as a TS0601 by _TZE200_pay2byax and have to use a custom quirk to be able to use it in Home Assistant. 

I assume this firmware would brick this device if I tried to update it via OTA.

[image]

[image]

pvvx, 2024-08-17:
Tuya has assigned the same OTA firmware IDs for TH03Z and other devices.
Manufacturer Code | Image Type
-- | --
1141 | d3a3

yfaykya, 2025-01-06:
I accidently updated a _TZ3000_xr3htd96 TS0201 and it became a TH03Z-z and the values for humidity and temp are now wrong. Is there way to force downgrade? (using ZHA)
Thanks!

[image]

yfaykya, 2025-01-06:
Another sensor I did not upgrade :  

[image]

pvvx, 2025-01-06:
Using Zigbee OTA, flash [BLE version](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota). 
Then in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html), use the desired firmware.