pvvx/ZigbeeTLc #114: ZDO error: bind failed with status 'TABLE_FULL' (140)

bernard-dandrea, 2024-08-18:
Hi

when I try in igbee2MQTT to update a setting in Reporting, eg Min Rep Interval for Temperature/measuredValue, I receive the following error

2024-08-18 10:22:28z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c138c3f42bb9c6/1 msTemperatureMeasurement from '0x00124b0029de1d21/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))' 

[image]

Here is my My z2m version

[image]

Here is device configuration

I think I am at the latest level for z2m and firmware

Would anyone has an idea about the problem ?

Thanks for your support

pvvx, 2024-08-18:
The binding of the report transmission of this cluster has not been completed. The reason given is that the binding table is full.
Possible reasons are that many bindings have been assigned to static clusters, but there is not enough memory to save the necessary ones.
So you've assigned a bunch of unnecessary reports on data that doesn't change or doesn't have alerting features. But the device doesn’t have enough memory for the ones you need.

bernard-dandrea, 2024-08-19:
Many thanks for your answer

The installation of the device is out of the box 

[image]

[image]

The reporting table is standard and I cannot change or unbind a reporting

Could it be that the Maximum number for report table is too short and need to be increased ?

juha63, 2024-08-19:
Looks like this error only appears in Zigbee2MQTT v 1.39.1 with external converter zigbeetlc_v0121.js. No errors without the js.

jjelcangues, 2024-08-28:
I have the same error with a zigbee air remote control module and a texas instrument coordinator, in another ha instance that I use for tests  and a silicon coordinator device bind is working perfectly.

@pvvx when you say thatt the device doesn’t have enough memory for the ones you need are you refering to the z2m coordinator? add a z2m router and balance the devices between the router and coordinator would works?

pvvx, 2024-08-28:
Someone creates duplicate bindings.
For example, ZHA also creates extra bindings - for example with the "UserInterface" cluster.
The table in the device has 8 bindings, but 4 is enough.

alessandroberna, 2024-09-04:
> Looks like this error only appears in Zigbee2MQTT v 1.39.1 with external converter zigbeetlc_v0121.js. No errors without the js.

Not really, facing the same issue with Zigbee2MQTT v1.39.1 without the external converter. 
The LYWSD03MMC is only bound to 4 clusters, as per default settings (genPollCtrl, PowerCfg, Humidity and Temperature). 
[image]
[image]
nonetheless i still get `z2m: Request 'zigbee2mqtt/bridge/request/device/configure_reporting' failed with error: 'Bind 0xa4c138dcf273f049/1 msTemperatureMeasurement from '0x00124b0026b73d42/1' failed (ZDO error: bind failed with status 'TABLE_FULL' (140))'` whenever i attempt to edit a reporting setting. 

z2m info:
[image]

bernard-dandrea, 2024-09-05:
I just can confirm that binding works well with version 1.39.0 of zigbee2mqtt and give TABLE FULL error with 1.39.1 and 1.40.0 as well

cbu99, 2024-12-17:
Solution for me was to [hard reset device](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#action-of-the-button) and rebind it. This seemed to clear the table on the device. No errors now.