pvvx/ZigbeeTLc #115: LYWSD03MMC HW:B1.5 doens't work with Z03MMC_v0122.bin

Dh0mp5eur, 2024-08-19:
Hi, 
I have some LYWSD03MMC under HA/Z2M.

Z2M suggested to upgrade the firmware to Z03MMC_v0122.bin.
After the upgrade some devices don't work.
All devices in HW B1.4 work correctly.
But those in HW B1.5 no longer display anything and no longer communicate.

Hardware:
CHMLmini_TLW_V1.5
202102
HW: B1.5

I have the same problem if I flash the firmware with the usb flasher.
I rollback to Z03MMC_v0121.bin to return to a normal state.

Thanks for your support

pvvx, 2024-08-19:
Updated 21 hours ago with file accepted automatically to ZHA.

[image]

HW Information: HW:B1.5
[image]

[image]

[image]

Dh0mp5eur, 2024-08-20:
Hi,
I don't understand, after another OTA update failed.
I flashed my device via USB and now it works.

Sorry for the inconvenience.
Thanks

closed 2024-08-20 by pvvx