pvvx/ZigbeeTLc #116: How to re-flash the firmware?

luvxinh, 2024-08-20:
Hi,

I reckon this might not be the question but hopefully you can shed some light on this for me. Thank you and advance.

So I have several LYWSD03MMC devices. Some were flashed with your firmware, which works great. There are 2 however, I did flash it with some other firmware that I can't recall now. There's a battery problem with those 2 devices: they consumes battery very quickly. 

You can see that they are the only devices that now have less than 50% battery
[image]

So now I want to flash a new firmware to those 2 devices. However, I can't seem to find those devices in the connect table.

[image]

Here's the info of the device:

[image]

Can you help me with this, please?

BTW, only those 2 devices have this logbook
[image]

pvvx, 2024-08-20:
[TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) works only with BLE devices.
Zigbee OTA is used to reflash Zigbee devices.
This is a standard function in ZHA.

To reflash Zigbee to BLE, you need to point ZHA to a special file ["1141-020a-99993001-ATC_v46.zigbee"](https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota)...

luvxinh, 2024-08-20:
Thanks a bunch. I thought to use TelinkMiFlasher for the reflash...

Can you please elaborate on "point ZHA to a special file"? what do you actually mean? Is there a function in ZHA to flash that firmware? I figure it has something to do with "Manage Zigbee device". Maybe I'll need to "write attributes" somewhere?
[image]

I understand that Zigbee has higher power consumption but am willing to take the tradeoff. The thing with my devices is that they consume power too quickly. The other devices (also zigbee, running your firmware) do not. That was why I asked.

Sorry for the questions, but as I'm a novice user, I'd appreciate your time and effort.

Thank you

closed 2024-12-05 by pvvx