pvvx/ZigbeeTLc #118: MHO-C122 not supported in Zigbee2MQTT

wimb0, 2024-09-02:
MHO-C122 seems to be missing in the zigbeetlc_v0121.js converter file.
If I understand the converter file correctly, it should show up as a LYWSD03MMC in Z2M?

I flashed one with the Zigbee firmware, and Z2m shows it as not supported:

[image]

robvanoostenrijk, 2024-09-03:
Your MHO-C122 should be natively supported in Zigbee2MQTT:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/9e4c840b5466dfd0891039ddf66056fbc5e9bc9e/src/devices/zigbeetlc.ts#L121

pvvx, 2024-09-03:
Most likely https://github.com/Koenkk does not support `manufacturerName: 'MiaoMiaoCe'`
https://www.miaomiaoce.com/

robvanoostenrijk, 2024-09-04:
@pvvx Looks like you are correct. That is the issue here. Is this a firmware issue or MiaMiaoCe is the correct name?

@wimb0, once @pvvx confirms I'll raise a pull request against the converter in Zigbee2MQTT to rectify the manufacturer name to what the firmware is reporting.

robvanoostenrijk, 2024-09-04:
We'll wait for pvvx to confirm if this is a firmware typo or not. If not, I'll change Zigbee2MQTT upstream.

His comment indicates it should be `MiaoMiaoCe`, as per the official website. In that case the firmware manufacturer is wrongly set (letter 'o' missing).

juha63, 2024-09-04:
sensorEpCfg.c:

```
#define ZCL_BASIC_MFG_NAME     {9,'M','i','a','M','i','a','o','C','e'} // MiaoMiaoCe
```

This has never changed since the first commit and has always been "MiaMiaoCe". Same with MHO_C401N.

wimb0, 2024-09-04:
Maybe fix both?
Correct name in new firmwares.
Both manufactureres in Z2M converters.

pvvx, 2024-09-04:
Ver 0.1.2.3: Added the ability to change the "manufacturer name". Changing the structure of saving the model and the manufacturer name. Corrected manufacturer name "MiaoMiaoCe" for MHO-C122 and MHO-C401N.

wimb0, 2024-09-04:
Thanks for the quick fix!
Will the new firmware be available as OTA in Z2M?

pvvx, 2024-09-04:
Unfortunately I do not support or use Z2M.

juha63, 2024-09-04:
@wimb0: You should be able to do a local OTA update in Z2M using the bin file 1141-020b-01233001-ZMHOC122.zigbee

wimb0, 2024-09-04:
I've updated via local OTA.
My MHO-C122 is working fine now!

wimb0, 2024-09-09:
Since this issue is resolved, I'll close it.

closed 2024-09-09 by wimb0