pvvx/ZigbeeTLc #120: Flashing _TZ3000_fllyghyj

jagheterfredrik, 2024-09-12:
I have a TS0201 _TZ3000_fllyghyj. It looks similar-ish to https://pvvx.github.io/TS0201_TZ3000/ but with a different PCB layout. It has a ZTU module and a 8305 sensor, same GPIO setup as https://pvvx.github.io/TS0201_TZ3000/.

The ISP header is different, starting from square pin 1:

- VCC
- GND
- SWS
- RST
- B7
- B1

Reading the image_type_id from the OTA cluster returns 65535. I bricked one device by flashing it over USB-COM with TS0201 images. Have you seen this device, how can I figure out how to get it supported?

Thanks for all your hard work on these cheap temp/hum sensors 👍

jagheterfredrik, 2024-09-12:
Revived it by setting atime 0 and connecting+flashing "simultaneously" using the TS0201_v48.bin and can see it in TelinkFlasher, however temp and hum is zero.

jagheterfredrik, 2024-09-12:
Downgrading to v4.7 yielded temp and hum correctly.

Sensor None id: 05835959, I2C address: 0x80, Kt: 16500, Kh: 10000, Zt: -4000, Zh: 0

jagheterfredrik, 2024-09-12:
Upgraded to v48 over BLE and again, sensor does not work. Not sure what changed between the versions.

jagheterfredrik, 2024-09-12:
Finally managed to get everything sorted. I flashed your full flash on two devices, ending up with colliding IEEEs when on zigbee firmware. Had to flash a full flash from another device type to get it sorted (I guess I could've changed the IEEEs on 0xD8200 but didn't want to bother with CRC).

closed 2024-09-12 by jagheterfredrik

pvvx, 2024-09-13:
> Upgraded to v48 over BLE and again, sensor does not work. Not sure what changed between the versions.

beta 4.8 - Fixed initialization/reading of CHT8305