pvvx/ZigbeeTLc #122: Are TLSR825x also compatible with OpenThread?

UnaiUribarri-TomTom, 2024-09-16:
Since Zigbee and Thread are both based on 802.15.4, maybe it is possible to have a simple thread implementation that just broadcasts an UDP packet every X seconds.

I am just curious if Thread is more power efficient than Zigbee.

pvvx, 2024-09-17:
The most energy efficient is BLE.

UnaiUribarri-TomTom, 2024-09-17:
Thread theoretically consumes less power than Zigbee because the wait states are shorter. Maybe it takes twice longer transmitting a packet, but wait periods in RX mode are 10x shorter because it does not need to wait for  notification acks and parent node maximum poll interval is also longer.

I already have a solid thread network with six border routers and great redundancy that came free with my wifi routers, TVs and speakers, but I do not have any BLE mesh network at home and my devices are too far away from my home assistant server for direct connection.

pvvx, 2024-09-18:
But there is no SDK from Telink for "Thread". Also there is no complete documentation for the RF part of the TLSR825x chips.
Plus there is no documentation and nuances on working with your "Thread" network.