pvvx/ZigbeeTLc #124: porting firmware for TS0201 _TZ3000_bguser20

KipK, 2024-09-26:
Hi 

I'm trying to adapt the build for my TS0201, this version is identified as _TZ3000_bguser20.

[image]

I've already flashed your build for TS0201, it runs ok but button is not working and led not flashing.

I've checked with a multimeter, button is wired to PB5 instead of PC0.
So I've changed the values in board_ts0201_tz3000.h like this:

```
#define BUTTON1             GPIO_PB5
#define PB5_FUNC			AS_GPIO
#define PB5_OUTPUT_ENABLE	0
#define PB5_INPUT_ENABLE	1
#define	PULL_WAKEUP_SRC_PB5	PM_PIN_PULLUP_10K
```

I haven't changed the led yet , I couldn't find the correct pin so will try with default one first. 

When compiling the OTA fw, I got those warnings:

```
create OTA image
 
./bin/1141-0211-01243001-z03mmc.zigbee was created with ZCL OTA Header.
 
Invoking: TC32 Create Extended Listing
Finished building: build/z03mmc.lst
 
TLSR825x MemInfo version 29.10.23
Warning: Undefined symbol 'ss_apsmeSwitchKeyReq'!
Warning: Undefined symbol 'ss_apsmeTransportKeyReq'!
Warning: Undefined symbol 'tl_zbNwkBeaconPayloadUpdate'!
```

I prefer to ask before bricking my device, does this error seems normal ? Can I still flash my build without risking bricking it ? 

Thanks

KipK, 2024-09-26:
Seeing the traecs on my pic,, it looks like the led is on PD2, I'll have to confirm that.

KipK, 2024-09-26:
I couldn't wait to test, so I've flashed it with OTA. Everything seems all right. Button and leds are now working as expected with the define i've changed above.

dynasticorpheus, 2024-10-23:
@KipK Would be great if you could share some hits to prep an OTA for my TS0201 _TZ3000_v1w2k9dd which is Zigbee out of the box. Bit unsure how to create something that is recognized by ZHA when using below config

`zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: advanced
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: /config/zigpy_ota

zha_toolkit:
`

pvvx, 2024-11-30:
> I'm trying to adapt the build for my TS0201, this version is identified as _TZ3000_bguser20.

[This version](https://github.com/pvvx/ZigbeeTLc/issues/129#issuecomment-2505164467) also uses "TS0201_TZ3000_bguser20" https://pvvx.github.io/TS0201_TZ3000_TH03/

Tuya = mess

KipK, 2024-11-30:
> Tuya = mess
sure that tuya is hell of a mess :)

> [This version](https://github.com/pvvx/ZigbeeTLc/issues/129#issuecomment-2505164467) also uses "TS0201_TZ3000_bguser20" https://pvvx.github.io/TS0201_TZ3000_TH03/
> 

Is the pinout of this one the same as the one I've posted above ? 

```
#define LED_ON 1
#define LED_OFF 0
#define GPIO_LED GPIO_PB4
#define PB4_FUNC AS_GPIO
#define PB4_OUTPUT_ENABLE 1
#define PB4_INPUT_ENABLE 1
#define PB4_DATA_OUT LED_OFF
```

pvvx, 2024-11-30:
> Is the pinout of this one the same as the one I've posted above ?

No. TH03 uses different pin connections for LED (GPIO_PC0) and KEY (GPIO_PD4).

KipK, 2024-11-30:
damn tuya mess ;)