pvvx/ZigbeeTLc #125: How to re-flash LYWSD03MMC to LYWSD03MMC-z?

nurtext, 2024-09-26:
Hi,

I have lots of devices flashed with the original firmware from devbis (https://github.com/devbis/z03mmc) and now want to re-flash all the devices to your pvvx-firmware. How to accomplish this using zigbee2mqtt and do I need to to back to BLE first before flashing to Zigbee again?

Appreciate your help, thanks.

Cheers,
Cedric

pvvx, 2024-09-26:
Files for Zigbee OTA.
To switch to ZigbeeTlc:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin -> README.md:
https://github.com/pvvx/ZigbeeTLc/tree/master/bin/devbis -> readme.txt

https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota -> README.md:

_From [devbis Zigbee](https://github.com/devbis/z03mmc) firmware to BLE:

[db15-0203-99993001-ATC_vxx.zigbee new version devbis](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/zigbee_ota/db15-0203-99993001-ATC_v48.zigbee)

[1141-0203-99993001-ATC_vxx.zigbee old version devbis](https://github.com/pvvx/ATC_MiThermometer/raw/refs/heads/master/zigbee_ota/1141-0203-99993001-ATC_v48.zigbee)_

nurtext, 2024-09-27:
Thanks, worked very well!

closed 2024-09-27 by nurtext

radzio, 2025-04-14:
> Thanks, worked very well!

@nurtext  can you share what exactly you did 🙏 ?