pvvx/ZigbeeTLc #127: 03MMC: v0.1.2.3: reported temperature values are differ from display

rpallai, 2024-10-18:
After updating to [Z03MMC_v0123.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/Z03MMC_v0123.bin), I noticed that the display and the reported temperature values differ consistently. The former updates correctly, while the latter follows with significant delays. Interestingly, this does not apply to the RH value, which always updates instantly over Zigbee when the display changes. Reverting back to v0.1.2.2, I do not experience this issue, and the value reported over Zigbee always matches the value shown on the display.

rpallai, 2024-10-18:
Before 10:45, v0.1.2.3 was running, and after that, I downgraded to v0.1.2.2. Nothing else changed. I didn’t modify the configurations after flashing.

[image]

pvvx, 2024-10-18:
It was your Zigbee coordinator who did not agree with the report settings. And the second time, after reflashing, the agreement was made.
[image]

rpallai, 2024-10-20:
@pvvx Thanks for your reply. I have an idea of what happened. When I first flashed the firmware, I was running an out-of-date Z2M that didn’t support _-z_ suffix in device names, so I believe it didn’t configure the device’s report settings. Later, I upgraded Z2M to the latest version, but I didn’t re-pair the device to the network, so it remained unconfigured. When I flashed it second time, the updated Z2M (which supported the firmware) was already running, allowing it to properly configure the report settings, and everything worked as expected.

What do you think, could this have been the case?