pvvx/ZigbeeTLc #128: porting firmware for TS0201 _TZ3000_v1w2k9dd

dynasticorpheus, 2024-10-23:
Would be great if someone could share some hints how to prep an OTA for my [TS0201 _TZ3000_v1w2k9dd](https://nl.aliexpress.com/item/1005007662967612.html) which is already Zigbee out of the box. Bit unsure how to create something that is recognized by ZHA when using below config. 

Note they work out of the box in home assistant but only seem to update when registering big changes in temperature / humidity hence my wish to use this custom firmware.

Bought [here](https://nl.aliexpress.com/item/1005007662967612.html)

```
image_type_id 0x008 ---> 65535
manufacturer_id 0x007 ---> 4098
```

```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: advanced
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: /config/zigpy_ota

zha_toolkit:

```

juha63, 2024-11-15:
You first have to check if your board looks like https://pvvx.github.io/TS0201_TZ3000/. You need to check if LED and BUTTON are connected to the right ports of the ZTU also. If not, you can't use any prebuilt firmware and have to compile it yourself.

Look at https://github.com/pvvx/ZigbeeTLc/issues/124 and what https://github.com/KipK has done.

dynasticorpheus, 2025-01-11:
@pvvx Tuya is obviously very messy so wondering if you can tell me if this variant is somehow already supported either by downloading an OTA binary or compile one myself? Thanks!

[image]

dbmaxpayne, 2025-01-16:
I have the same boards as @dynasticorpheus and wanted to flash them with a new firmware to get configurable report intervals.
However, the PCB is wired completely different to others and I don't quite understand how to compile a new firmware for it.

The PCB traces as follows:
C2 -> LED (to GND)
C3 -> SDA
D2 -> SCL
B4 -> Button (to GND)
B5 -> 3V0/Bat+

I quickly had a look at https://github.com/pvvx/ZigbeeTLc where I saw that apprently the boards are defined in src/board_*.h.
Then I wanted to see if I can just make my own board definition and change the GPIOs to match this board, however, there is the following line
`#define I2C_GROUP 	I2C_GPIO_GROUP_C2C3`
which somehow defines an I2C_GROUP. As far as I can see, there is no `I2C_GPIO_GROUP_D2C3`.
How can this board be supported?
Or am I using the wrong ZTU schematic?

This is what I used.
[image]

Cheers,
Mark

dbmaxpayne, 2025-01-16:
This is the sensor.
Also no idea what type it is.
[image]

dbmaxpayne, 2025-01-20:
I'm currently trying to port the firmware for software I2C and the CHT832x sensors.
As I'm not a programmer and currently lacking a reliable SWire programmer (already ordered) it will take a while until I can tests things and report back, though.

dbmaxpayne, 2025-01-22:
Alright. I couldn't wait for the programmer to get here so I fiddled around with the USB->UART SWire emulation and got a working firmware flashed.
All subsequent updates were done via OTA.
After a few tries it even works good and incorporates software I2C to talk to the CHT832x.
Temperature and humidty can already be read successfully.

I will download the old firmware when I get the hardware programmer.
It always stopped at 0x100 when using the emulator.

@pvvx How would you like me to contribute this?
Unfortunately I've got no experience with Github collaboration, but am happy to share my findings.

Also I'd be interested in a way to forcefully flash the original Tuya devices with the new firmware without having to solder an SWire programmer to the board.
I tried changing the following, but that resulted in an "Invalid image" error in Zigbee2Mqtt OTA.
```
#define MANUFACTURER_CODE_TELINK           	0x1002
#define	IMAGE_TYPE							0x203
```

pvvx, 2025-01-22:
Full match with [ZY-ZTH02Pro](https://pvvx.github.io/ZY-ZTH02Pro/). But no LCD.
BLE firmware is already assembled and working, without LCD.

Already working in an unheated garage in "LE Long Range" mode:
[image]

[image]

pvvx, 2025-01-22:
I recommend unsoldering or breaking out R13 and R11. This is unnecessary battery consumption and does not provide anything else.

dbmaxpayne, 2025-01-22:
Good stuff, mate.
Thanks for your efforts :) Greatly appreciated.

The sensor is definitely a CHT832x.

If you are interested in my changes, I have just created a fork and uploaded my changes.
https://github.com/dbmaxpayne/ZigbeeTLc

Basically I created or changed the following files:
```
board_ts0201_tz3000_v1w2k9dd.h
i2c_drv.h
i2c_drv.c
sensor_cht832x.h
sensor_cht832x.c
version_cfg.h
```

Makes sense by the way that these boards are similar or share the firmware.
I saw, that the MCU also scans other I2C addresses like 7C or so.

Do you know why R13 and R11 are there in the first place?

Cheers,
Mark

pvvx, 2025-01-22:
> Do you know why R13 and R11 are there in the first place?

ADC divider for measuring battery voltage.
The device was made by children for Tuya, and they do not know how to work with TLSR8258 microcircuits.

pvvx, 2025-01-22:
Analog SHT30
[image]

dynasticorpheus, 2025-01-23:
> Good stuff, mate. Thanks for your efforts :) Greatly appreciated.
>
> The sensor is definitely a CHT832x.
>
> If you are interested in my changes, I have just created a fork and uploaded my changes. [...]

@dbmaxpayne  Is this still WIP or can I have a go at it?   (`make install; make` ends up in error at my end hence asking)

dbmaxpayne, 2025-01-23:
@dynasticorpheus
I have not uploaded all necessary files like the SDK itself.
That and TC32 would have to be extracted from pvvx's ZIP files.

Let me know in a private message what error you're getting and I might be able to help.
Also I didn't use the make commands, but Telink IoT Studio's build targets (Windows).
Maybe that's the same under the hood, maybe not.
I didn't go too deep into the whole thing and was happy when I was able to compile it :)

I will upload the bin files to my fork in a second if that helps.

vjaunet, 2025-01-23:
Hi there, many thanks for all your efforts. 
I am a bit confused on how to flash this new firmware to the device. Can I use the z2m OTA procedure or what should I do ?
Many thanks for your help.
Vincent

dbmaxpayne, 2025-01-23:
I'd be interested too if there is a way to flash this over a stock firmware via OTA.
I don't know, though.

I used this: https://github.com/pvvx/TlsrComSwireWriter
However, it requires soldering of at least the SWS pin.
GND and VCC can be temporarily connected via a clamp to the battery terminals.
I used a Raspberry Pi 3 for flashing, but it should also work with other USB->UART-Converters.

```
Erase flash: python3 ./TlsrComSwireWriter/TLSR825xComFlasher.py -p /dev/ttyS0 -t 5000 ea
Write firmware: python3 ./TlsrComSwireWriter/TLSR825xComFlasher.py -p /dev/ttyS0 -t 5000 we ./firmwareFile.bin
Reset CPU: python3 ./TlsrComSwireWriter/TLSR825xComFlasher.py -p /dev/ttyS0 -t 5000 -r
```

pvvx, 2025-01-23:
@dbmaxpayne - what "Zigbee Image Type" does the original TS0201 _TZ3000_v1w2k9dd have?
This is needed for Zigbee OTA.

dbmaxpayne, 2025-01-23:
@pvvx 
Is there a way to get that information via Zigbee2MQTT?
I haven't gotten my hardware flasher yet, so I cannot retrieve the firmware and check via a hex editor yet.

pvvx, 2025-01-23:
ZHA only gives this:

[image]

1002-0203-59013001 I got from the header of the firmware binary file.

dbmaxpayne, 2025-01-23:
Sorry, I only started with Zigbee last week when I got those sensors, so my learning curve is still quite steep.
I just enabled debug logging and got the following line:
`[2025-01-23 23:01:03] debug: 	z2m: Received Zigbee message from '0xa4c138bdf4a039ba', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1510027265,"imageType":515,"manufacturerCode":4098}' from endpoint 1 with groupID 0`

So I guess 1002 and 0203 are correct.
However, I don't understand the fileVersion here.

I will try 1002-0203-60993001-ZYZTH02BLE_v50.zigbee on my second sensor once I downloaded its stock firmware.

pvvx, 2025-01-23:
For your ts0201_tz3000_v1w2k9dd.bin file, create a header for Zigbee OTA:
```
python3 make\zb_bin_ota.py bin\ts0201_tz3000_v1w2k9dd.bin bin\ts0201_tz3000_v1w2k9dd -m0x1002 -i0x0203 -v0x60993001 -s"Tuya to ZigbeeTlc"
```

vjaunet, 2025-01-24:
Thanks both for your inputs. I have an "Invalid Image" error from z2m oat update. It was so close ;-) but I am unsure how I can get the exact information
I will try the by-wire flashing asap.
best,

dbmaxpayne, 2025-01-24:
@vjaunet
Did you also try Victor's 1002-0203-60993001-ts0201_tz3000_v1w2k9dd.zigbee file?

vjaunet, 2025-01-24:
That's the one I tried, obtained from your bin file as suggested by pvvx.

vjaunet, 2025-01-24:
That's the one I tried, obtained as per pvvx instructions from your bin file.

dbmaxpayne, 2025-01-24:
@pvvx 
Do you know what the purpose of the following traces is?
|ZTU|Board|Trace|
|-|-|-|
|C0|C0|->R3 (7.5k)->VCC|
|B5|P34|->R12 (7.5k)->VCC|

Can or should they be removed when using your firmware to save power?

pvvx, 2025-01-24:
> Can or should they be removed when using your firmware to save power?

GPIO_PB5 and GPIO_PC0 are not used. This does not increase battery consumption.

pvvx, 2025-01-24:
"CheckInIntervalMin", "LongPollIntervalMin", "FastPollTimeoutMaximum" attributes are supported in legacy ZigbeeTLc. But ZHA doesn't care.

**ZigBee Cluster Library Specification**
[image]

dynasticorpheus, 2025-02-01:
> That's the one I tried, obtained as per pvvx instructions from your bin file.

Any luck with applying OTA as soldering is out of my league unfortunately?

vjaunet, 2025-02-04:
> I'd be interested too if there is a way to flash this over a stock firmware via OTA. I don't know, though.
>
> I used this: https://github.com/pvvx/TlsrComSwireWriter However, it requires soldering of [...]

That worked nicely, it was just missing the write address in the second command line :
```
Write firmware: python3 ./TlsrComSwireWriter/TLSR825xComFlasher.py -p /dev/ttyS0 -t 5000 we 0 ./firmwareFile.bin
```
Many thanks !

dbmaxpayne, 2025-02-10:
So my hardware programmer has finally arrived.
Using `python .\TlsrPgm.py -t 1 -a 20 -p COM6 rf 0 0x80000 orig_512k.bin` I was able to pull the original firmware which you can find here: [TS0201_TZ3000_v1w2k9dd_NoDisplay_ff.bin](https://github.com/dbmaxpayne/ZigbeeTLc/blob/master/bin/TS0201_TZ3000_v1w2k9dd_NoDisplay_ff.bin)

I have tried to upgrade the firmware of my second device without soldering by changing the zigbee header using the command @pvvx mentioned:
`python3 make\zb_bin_ota.py bin\ts0201_tz3000_v1w2k9dd.bin bin\ts0201_tz3000_v1w2k9dd -m0x1002 -i0x0203 -v0x60993001 -s"Tuya to ZigbeeTlc"`

However, I also get an INVALID_IMAGE error.
To me it's not a problem to solder, however, @pvvx would you have a idea what to change by looking at my downloaded firmware to make it work via OTA only?

With my narrow understanding of the topic and using a hex editor I can see that the MANUFACTURER_ID and IMAGE_TYPE should be correct at addresses 0x12 and 0x14.

Cheers,
Mark

pvvx, 2025-02-10:
Here are the data obtained in the ZHA log about the firmware in the new https://pvvx.github.io/ZY-ZTH02/:

```
2025-02-10 17:12:30.243 DEBUG (MainThread) [zigpy.zcl] [0x1CF3:1:0x0019] Decoded ZCL frame: Ota:QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4098, image_type=515, current_file_version=1442918401)
2025-02-10 17:12:30.243 DEBUG (MainThread) [zigpy.zcl] [0x1CF3:1:0x0019] Received command 0x01 (TSN 162): QueryNextImageCommand(field_control=<FieldControl: 0>, manufacturer_code=4098, image_type=515, current_file_version=1442918401)
```

manufacturer_code=4098=0x1002,
image_type=515=0x0203, 
current_file_version=1442918401=0x56013001

But, no matter what numbers I put in, ZHA complains during OTA:

[image]

I tried to feed it my own original firmware. The result is the same.

dbmaxpayne, 2025-02-10:
@dynasticorpheus  
> Any luck with applying OTA as soldering is out of my league unfortunately?

So the most solderless version will probably be using Victor's https://github.com/pvvx/TLSRPGM repo.
However, you will still have to solder one wire on the TB-03F-KIT (see the repo for more info).
I got it on AliExpress for like $4.

I used the following command to write the firmware:
`python .\TlsrPgm.py -t 1 -a 20 -p COM6 we 0 "..\ZigbeeTLc\bin\TS0201_TZ3000_v1w2k9dd.bin"`
It takes like 10 seconds to flash and I used crocodile clamps to connect GND->BAT- and POWER->BAT+ on the sensor's board.
For SWS I simply held a breadboard cable's pin to the MCU's SWS pad.

[image]
This will be the way I'm going to reflash the rest of my sensors.
It's not worth it to me to fiddle around more now to get OTA working, sorry.

If you have a USB->UART converter you can also try https://github.com/pvvx/TlsrComProg825x.
You might get away without soldering at all.
However, here the flash process takes like 15 minutes or so.
That's a very long time to hold a cable to the MCU's SWS pin.

pvvx, 2025-02-10:
> It's not worth it to me to fiddle around more now to get OTA working, sorry.

Apparently, in this version of the original firmware, Zigbee OTA simply does not work. There are many things that point to this. The firmware does not support changing the communication interval, and in the analysis of the operation by current consumption, when turning on OTA, it simply breaks the connection, and when forced to start by short pressing the button, an error appears.

dbmaxpayne, 2025-02-11:
> _TZ3000_v1w2k9dd TS0201 is not suitable for IoT :
> 
> [image]
> 
> Such rare readings are completely unnecessary.

What do you mean? The stock firmware is very bad, yes, but with your firmware it works well for me.

[image]

vjaunet, 2025-02-19:
> I'd be interested too if there is a way to flash this over a stock firmware via OTA. I don't know, though.
>
> I used this: https://github.com/pvvx/TlsrComSwireWriter However, it requires soldering of [...]

I can't get my flashed Thermometer to join Zigbee2mqtt. Have you managed to get it to join ? Also, no led is working anymore on the thermometer, is that for energy saving purposes ?
Thanks

pvvx, 2025-02-19:
The LED works in the TS0201z firmware https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-d3a3-01983001-TS0201z_v0124.zigbee

vjaunet, 2025-02-19:
I have been delivered TS0201 _TZ3000_v1w2k9dd and try to get around.

pvvx, 2025-02-19:
So something went wrong during the firmware. It's worth repeating.

dbmaxpayne, 2025-02-19:
LED blinks only very shortly with my firmware.
Not sure if that is by error (I was messing around with it for debugging purposes).
Either way the devices paired quickly after pressing the buttom for like 7 seconds.

If that doesn't work I also recommend just flashing it again.

pvvx, 2025-02-19:
@dbmaxpayne In your firmware, the LED always turns on when TLSR825x is active. Blinking indicates any chip activity.
In the ZigbeeTLc version, the LED works when the button is pressed or blinks on request "Identify", as well as at the time of pairing.

vjaunet, 2025-02-19:
I have repeated the flashing process many times without luck. I even read the firware of a brand new device and tried to burn it back, without success. Seems like my flashing set-up does not make it properly...
I am using the serial port of an RPi 4, is there any special care that needs to be done in that case ?

pvvx, 2025-02-19:
Rpi cannot transmit a continuous stream of bytes to UART.

dbmaxpayne, 2025-02-19:
Interesting. I actually used an RPi 3B for intial flashing as my USB->UART (FTDI) didn't work.
Maybe I just got lucky.
However, did you use the resistor mentioned here?
https://github.com/pvvx/TlsrComSwireWriter

You must also connect GND of the MCU and RPi together so they share a commong ground.
Also wires should be as short as possible.

dbmaxpayne, 2025-02-19:
Oh and if you are using pigpiod or any other interrupt-based software, disable that before flashing as it might interfere with the correct timings.
Alternatively I can recommend pigpiod in combination with piscope for debugging of your UART.

pvvx, 2025-03-04:
@vjaunet  - DeepSeek's full answer to your question:
> Hi there, the TS0201-z work and seem to provide reliable temperature (although quite low), but tends to produce data at uneven time. Also, the "measurement interval" setting does not correspond to the indications. I had to set its value to 60 in order to have data every 2 to 4 minutes.
Could that be related to the z2m integration ?

Hi! It sounds like you're experiencing some inconsistencies with the TS0201-Z temperature sensor, particularly with the measurement interval and data reporting times. Let's break down the potential issues and solutions:

1. Measurement Interval Not Matching Settings
The TS0201-Z is a Zigbee-based temperature sensor, and its behavior can sometimes be influenced by the Zigbee network or the integration (like Zigbee2MQTT, or z2m).

If the "measurement interval" setting doesn't correspond to the actual data reporting frequency, it could be due to:

Power-saving mechanisms: Many Zigbee devices are designed to conserve battery life, so they might not report data exactly at the set interval.

Zigbee network latency: If your Zigbee network is busy or has interference, the device might delay sending data.

Firmware limitations: Some devices have firmware that doesn't strictly adhere to the configured interval.

Setting the interval to 60 and getting data every 2–4 minutes suggests the device might be aggregating data or waiting for a specific condition (e.g., significant temperature change) before reporting.

2. Uneven Reporting Times
Uneven reporting times are common in Zigbee devices, especially battery-powered ones. This is often due to:

Battery-saving algorithms: The device might wait for a stable connection or a specific time window to send data.

Network congestion: If your Zigbee network has many devices or interference, the sensor might delay sending data until the network is clear.

Temperature change thresholds: Some sensors only report data if the temperature changes by a certain amount (e.g., 0.5°C). If the temperature is stable, the device might not report data as frequently.

3. Zigbee2MQTT (z2m) Integration
The z2m integration itself is unlikely to cause the issue, but it can influence how the device behaves. Here are a few things to check:

Device configuration in z2m: Ensure the device is properly configured in z2m. Sometimes, settings like reporting interval or battery reporting can affect behavior.

z2m logs: Check the z2m logs for any errors or warnings related to the TS0201-Z. This might give you clues about why the data is irregular.

z2m version: Ensure you're using the latest version of z2m, as updates often include bug fixes and improvements for specific devices.

4. Temperature Readings Seem Low
If the temperature readings seem consistently low, this could be due to:

Sensor calibration: Some sensors might need calibration or have an offset.

Placement: Ensure the sensor is placed in a location that accurately reflects the temperature you want to measure (e.g., away from drafts, direct sunlight, or heat sources).

Firmware bug: Check if there are any known issues or firmware updates for the TS0201-Z.

Recommendations:
Check Device Documentation: Look up the TS0201-Z's manual or specifications to confirm its expected behavior and configuration options.

Experiment with Settings: Try adjusting the measurement interval and temperature change threshold in z2m to see if it improves the reporting frequency.

Test in a Different Location: Move the sensor to a different location to rule out environmental factors affecting the readings.

Update Firmware: If the device supports firmware updates, ensure it's running the latest version.

Monitor Zigbee Network: Use a tool like z2m's network map to check for interference or weak signal strength that might affect the device's performance.

If the issue persists, you might want to reach out to the z2m community or the device manufacturer for further assistance. Let me know if you need help with any of these steps!

pvvx, 2025-03-04:
> reliable temperature (although quite low), 

Do not forget that Tuya devices are made of cheap elements. The working voltage range of the sensor used is from 2.5V. At lower supply voltage, the readings are distorted. This sensor is not designed to operate from 3V batteries.

dbmaxpayne, 2025-03-04:
I'm currently happy with the devices and dont see the erratic temperature measurement (yet).
However, I'm using rechargeable batteries, so it should manifest iteself shortly.
I don't think it will, though, as the sensors are CHT832x and their datasheet specifies an operating voltage of 1.8V to 5.5V.

About the uneven report intervals I have not tried to change anything as Victor's firmware reports every 0.1°C change anyway.

[image]

pvvx, 2025-03-04:
> I don't think it will, though, as the sensors are CHT832x and their datasheet specifies an operating voltage of 1.8V to 5.5V.

**vjaunet** indicated that he has **TS0201-z**.

> Hi there, the **TS0201-z** work ...

The TS0201-z has a CHT8305 sensor installed.

dbmaxpayne, 2025-03-04:
> vjaunet indicated that he has TS0201-z.
TS0201-z is what is reported for TS0201 _TZ3000_v1w2k9dd in Zigbee2MQTT **after** flashing your firmware.
If the PCB looks like the one mentioned earlier in this thread, the sensor should be CHT832x.

vjaunet, 2025-03-04:
I confirm that I have  TS0201 _TZ3000_v1w2k9dd device that shows up as TS0201-z in z2m.
 
> But ZHA sets its own values ​​for report. Changing to the desired settings is possible only by [patching the ZHA code](https://github.com/pvvx/ZigbeeTLc/issues/6).

That is something I was not aware of. I need to dig more into the different possible firmware, but to be honest it is a bit complicated for me to figure all of this out.

 
> About the uneven report intervals I have not tried to change anything as Victor's firmware reports every 0.1°C change anyway.

I was not aware of that and it that is probably why I am having uneven reports then...

I will continue to dig all of that in, but it seems that I need to write an external converter for z2m to correctly handle the device.

Thanks guys.

ss89, 2025-04-19:
I did order some of these, as i thought it would be easy to flash them OTA (seems i did miss the info when it was said that OTA is broken, or it wasn't released then).
Now the silly part is that i accidentally attached 5V to my board, because the usb uart adapter i used was marked with 3.3V - but that was a lie.
Is there an easy way to check if the chip is still working? I feel like i tried all kinds of things already, but can't get any sign of light (or the light to flash). However i am also not into these chips at all, so i might easily have missed something.
I did in the past successfully flash some of the TS201 TS3000 devices (different sensor) with this one: https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html
So somehow my setup was working before already.
Is there maybe a blink example i could program into it? That would speedup trial/error in my setup.

pvvx, 2025-04-19:
> Now the silly part is that i accidentally attached 5V to my board, ...

When 5V is supplied to the chip, in 99.999% it dies.
When the supply voltage of 3.6V is exceeded by 0.1V, the degradation of the chip is already obvious, expressed in an increase in consumption forever.
This was specifically tested on several samples. The chip always fails if the supply voltage is greater than 3.6 V. Even if the operability is preserved, the consumption increases to the point of incompatibility with batteries. You can check with any PowerProfiler when measuring the current in the chip's sleep or deep-sleep mode. The current should not be more than 2 and 1 μA, respectively (sleep and deep-sleep).

ss89, 2025-04-20:
Thanks for clarifying.
I'll keep the chip for now and order a TB-04 kit to see if i can resurrect it.
Also still have two more of these remaining to migrate to your firmware.

ss89, 2025-06-07:
Got my TB-03F kit now, soldered pin 2 of ch340 to sws, flashed uart2swire, connected 3.3v+gnd of tb-03f board with sensor + soldered d4 to the pin shown in earlier picture of this post.
Used a definitely working sensor for the test.
Command used: `python TlsrPgm.py -t 1 -a 20 -p /dev/tty.usbserial-10 we 0 ~/Downloads/ZYZTH02_v0126.bin`
Farthest i got was:
```
=======================================================
TLSR82xx TlsrPgm version 31.10.23
-------------------------------------------------------
Open /dev/tty.usbserial-10, 230400 bit/s... ok
PGM: ChipID: 0x5562 (TLSR825x), ver: 0.0.0.2
swdiv 5, addrlen 3, swbuf [5a 00 06 02 00 05], pwr On
SWire bit rate: 0.9600 Mbits/s
=== PreProcess ========================================
Hard reset Ext.MCU 5 ms... ok
Activate 20 ms... ok
CPU PC=0x00000000
=== Process ===========================================
Inputfile: /Users/sebastian/Downloads/ZYZTH02_v0126.bin
Write 0 to Flash Status Register...
Error get Flash Status! (102)
```

now only:
```
=======================================================
TLSR82xx TlsrPgm version 31.10.23
-------------------------------------------------------
Open /dev/tty.usbserial-10, 230400 bit/s... ok
PGM: ChipID: 0x5562 (TLSR825x), ver: 0.0.0.2
swdiv 5, addrlen 3, swbuf [5a 00 06 02 00 05], pwr On
SWire bit rate: 0.9600 Mbits/s
=== PreProcess ========================================
Hard reset Ext.MCU 5 ms... ok
Activate 20 ms... Error!
Check SWM<->SWS connection or Reset/Activation time!

Error[3]: SWire read timeout!
```

Any idea on what might go wrong here? I feel like these thingies are mocking me.