pvvx/ZigbeeTLc #129: TH03 firmware

vkravets, 2024-11-14:
Is any chance to upload custom firmare analogue of TH03Z with LCD?

https://www.aliexpress.com/item/1005007309529179.html?spm=a2g0o.order_list.order_list_main.11.a23b1802WLv9gX#nav-specification

vkravets, 2024-11-14:
the display will work in this case?

pvvx, 2024-11-14:
The required firmware and the option of how to solder the jumpers have not yet been made.

vkravets, 2024-11-14:
I'm ready to be a tester of it ) I can desoldering/soldering, I don't have soldering fan, but I think I can find how to desoldering the Tuya MCU

pvvx, 2024-11-14:
I have the following names in the firmware of the ZTH05 thermometer from Tuya:
TS0601_TZE204_f8mujzmz
TS0601_TZE204_upagmta9

vkravets, 2024-11-14:
Let you know when recheck with my device

vkravets, 2024-11-14:
[image]
[image]
[image]
[image]

I don't see this mcu unit

It's more similar to th05 but ZigBee version

pvvx, 2024-11-15:
> I don't see any tuya mcu unit

Yes, it is completely different from ZTH05.
But the main chip is unknown.

vkravets, 2024-11-15:
Zigbee2MQQT detects it as following device:
[image]

vkravets, 2024-11-15:
How we can detect or debug which chip is used? Is it possible to take off the metalic holder at top?

juha63, 2024-11-15:
Looks like the metal (rf) cap is soldered to the pcb. True? Maybe it's also fixed with latches through hole? Then you have to take your hot air gun and desolder the whole pcb first and then the cap. Of course there will be a good chance that you'll destroy the "chip". It depends on your soldering skills.

pvvx, 2024-11-15:
Mac/IEEE address from Telink. Perhaps there TLSR8258.

vkravets, 2024-11-15:
I'm already ask my friend to desoldering metal cup, so in any case give you info what inside )

vkravets, 2024-11-15:
In description of the to the issue I post the link to ali also )

pvvx, 2024-11-15:
By the link https://www.aliexpress.com/item/1005007309529179.html
(from the first message) is not delivered to my region.

vkravets, 2024-11-15:
Hmm, in this case will buy LYWSD03MMC for sure... )

About the desoldering metal cup... bad news there is no any mark for the chip )

[image]
[image]
[image]

vkravets, 2024-11-15:
> TLSR8258

I can try to use Zigbee OTA to flash custom firmware? Above picture can help to understand if it's TLSR8258?

pvvx, 2024-11-15:
TLSR8258F512ET32 ?
https://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/

vkravets, 2024-11-15:
As I know BLE is not here ( only zigbee... or BLE can be not powered during init the chip?

pvvx, 2024-11-15:
Quartz at 24.0 MHz, like all TLSR825x.
The conductors on the module are not visible in the photo. But the general arrangement of the elements roughly corresponds to TLSR825x.

All TLSR825x have the same crystal. The only difference is in the plastic case and what pins are made from the crystal.
Support all types of BLE PHY - 2M, 1M, 500k, 125k and Zigbee - 250k. All have the same RAM size - 64 KB, but can have different Flash. Different number on the plastic = marketing.

vkravets, 2024-11-15:
Where I can take custom firmware and how to flash?

pvvx, 2024-11-15:
For the TH03Z variant, the firmware needs to be created. Find out which pins go where, how the LCD segments are routed. Take the source code and write... :)

vkravets, 2024-11-15:
I can also invistigate, I have only one question if something goes wrong how to flash via UART... since I'm not sure where the uart pins to flash... May be by photos you already find them? )

vkravets, 2024-11-15:
[image]

pvvx, 2024-11-15:
First, you need to read the existing Tuya firmware.

The simplest option:
https://github.com/pvvx/TlsrComSwireWriter
https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py

---

[Datasheet ](https://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/) DS_TLSR8258-E_Datasheet for Telink BLE IEEE802.15.4 Multi-Standard Wireless SoC TLSR8258.pdf :

[image]

[image]

vkravets, 2024-11-15:
Ok, I have USB-Serial TTL can I use it also? But is not clear how to use RST?

vkravets, 2024-11-15:
I found all neeed pins
As expected P2 section have all needed pins (numeration from the text P2)
1. VCC+ 3.3V
2. SWS
3. GND

The reset pin from the chip (RESETB)
Near the module's antenna from the buttom the second pin

[image]

RESETB it's the same which need to use as RST?

I have the USB-TTL with reset pin thus I'm going to use following schema of connection

```
TX -> R 1k -> SWS
RX         -> SWS
RST        -> RST
GND        -> GND
```

Everything is right?

pvvx, 2024-11-16:
If this is so new to you, how will you create the firmware?

Pictures of wires won't do. A map of the segments on the LCD glass is required. Example -https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd_mho_c122.c#L19-L41
Next, you need to rework the code to handle the output of data to the display.

vkravets, 2024-11-16:
I'm able to read the firmware 

```
$ TLSR825xComFlasher.py --port /dev/tty.wchusbserial53220378511 -d -t 500  rf 0 0x10000 1.bin
=======================================================
TLSR825x Flasher version 00.00.05
-------------------------------------------------------
Open /dev/tty.wchusbserial53220378511, 921600 baud...
Reset module (RTS low)...
Activate (500 ms)...
Debug: default swdiv for 24 MHz = 52 (0x34)
Debug (read data):
000023: 3b 3b fa
Debug (read data):
000024: 3b 3b bb
Debug (read data):
000025: 3f 3b a3
Debug (read data):
000026: 3f 3f bc
Debug (read data):
000027: 3f 3f 84
Debug (read data):
000028: 77 c7 bf
Debug (read data):
000029: 77 c7 47
Debug (read data):
00002a: 77 47 74
Debug (read data):
00002b: 77 47 44
Debug (read data):
00002c: 77 47 77
Debug (read data):
00002d: f7 87 87
Debug (read data):
00002e: f7 87 f4
Debug (read data):
00002f: f7 87 84
Debug (read data):
000030: ff f0 ff
Debug (read data):
000031: ef f8 8f
Debug (read data):
000032: ef 88 0f ef
Debug (read data):
000033: ef 08 ef 08 ff
Debug (read data):
000034: ef 08 0f ef ff
Debug (read data):
000035: ef 08 0f 0f ff
Debug (read data):
000036: ef 08 0f e8 ff
Debug (read data):
000037: ef 08 0f 08 ff
Debug (read data):
000038: ef 08 e8 ef ff
Debug (read data):
000039: cf 08 e8 0f ff
Debug (read data):
00003a: df 08 e8 d8 ff
Debug (read data):
00003b: df 08 d8 08 ff
Debug (read data):
00003c: df 18 18 df ff
Debug (read data):
00003d: df 10 10 1f ff
Debug (read data):
00003e: df 10 10 d0 ff
Debug (read data):
00003f: df 10 10 10 ff
Debug (read data):
000040: 1f df df df ff
Debug (read data):
000041: 1f df df 1f ff
Debug (read data):
000042: 1f df df d0 ff
Debug (read data):
000043: 1f 9f 9f 10 ff
Debug (read data):
000044: 3f 9f 1f bf ff
Debug (read data):
000045: 3f bf 1f 1f ff
Debug (read data):
000046: 3f bf 3f b0 ff
Debug (read data):
000047: 3f bf 3f 30 ff
Debug (read data):
000048: 3f bf b0 bf ff
Debug (read data):
000049: 3f bf b0 3f ff
Debug (read data):
00004a: 3f bf a0 a0 ff
Debug (read data):
00004b: 3f 3f a0 20 ff
Debug (read data):
00004c: 3f 3f 20 3f ff
Debug (read data):
00004d: 3f 3f 20 3f ff
Debug (read data):
00004e: 3f 3f 60 20 ff
Debug (read data):
00004f: 7f 3f 20 20 ff
Debug (read data):
000050: 7f 7f 7f 7f ff
Debug (read data):
000051: 7e 7f 7f 7e ff
Debug (read data):
000052: 7e 7e 7f 60 ff
Debug (read data):
000053: 7e 7e 7f 60 ff
Debug (read data):
000054: 7e 7f 7f 7e ff
Debug (read data):
000055: 7e 7f 7e 7f ff
Debug (read data):
000056: 7e 7e 7f 60 fe
Debug (read data):
000057: 7e 7e 7e 40 fe
Debug (read data):
000058: 7e 7e 40 7f fe
Debug (read data):
000059: fe 7e c0 7e fe
Debug (read data):
00005a: fe fe 40 c0 fe
Debug (read data):
00005b: fe fe c0 c0 fe
Debug (read data):
00005c: fe fe c0 fe fe
Debug (read data):
00005d: fe fe c0 fe fe
Debug (read data):
00005e: fe fe c0 c0 fe
Debug (read data):
00005f: fe fe c0 c0 fe
Debug (read data):
000060: fe c0 fe fe fe
Debug (read data):
000061: fe c0 fe fe fe
Debug (read data):
000062: fe c0 fe c0 fe
Debug (read data):
000063: fe 80 fe fe c0
Debug (read data):
000064: fe 80 fe 80 fe
Debug (read data):
000065: fe 80 80 fe fe 80 80 fe
Debug (read data):
000066: fe 80 80 fe fe 80 80 fe fe
Debug (check data):
0ccc66: 80 fe 80 80 fe fe 80 80 fe fe
bit mask: 0xc0
UART-SWS 92160 baud. SW-CLK ~47.0 MHz(?)
Outfile: 1.bin
Read Flash from 0x000000 to 0x010000...
-------------------------------------------------------
Worked Time: 286.621 sec

Done!
```

[original-firmware-th03-v1.1.bin.zip](https://github.com/user-attachments/files/17784490/original-firmware-th03-v1.1.bin.zip)

vkravets, 2024-11-16:
[image]

Using following connection:
TX -> R 1k -> SWS
RX         -> SWS
RST        -> VCC+
GND        -> GND

pvvx, 2024-11-16:
> I'm able to read the firmware

TLSR8258 use 1 MB or 512 KB Flash.

```
$ TLSR825xComFlasher.py --port /dev/tty.wchusbserial53220378511 -d -t 500  rf 0 0x100000 fullflash.bin
```

vkravets, 2024-11-16:
Ahh, it's SRAM 64k ) So tiny and 512k... 
Yep it's 512k, if choose to dump 1M after 512k take the same data from 0 offset.
[original-firmware-th03-v1.1_512k.zip](https://github.com/user-attachments/files/17786715/original-firmware-th03-v1.1_512k.zip)

vkravets, 2024-11-20:
I have small progress for it 
- default firmware with small changed config (without display) successfully have been upload and run
- Zigbee paired and populate all info (bat, hum, temp)

Working on the display, since it's has the same display and controller as TH05 will try to take some info from it

[image]
[image]

pvvx, 2024-11-21:
https://pvvx.github.io/TH-05/
Diver LCD: [BL55028](https://pvvx.github.io/TH-05/BL55028_V1.3_en.pdf),

vkravets, 2024-11-21:
Ok, I've already use this code. If it works only via i2c, in this case I'm going to find where the mistake. Thanks for documentation

pvvx, 2024-11-21:
The [LCD driver is the same as LYWSD03MMC B1.9](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd_lywsd03mmc.c#L59).
First, the BLE firmware is made. It has an interface for specifying the bitmap of the segment layout in the LCD glass.

vkravets, 2024-11-22:
> The [LCD driver is the same as LYWSD03MMC B1.9](https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd_lywsd03mmc.c#L59). First, the BLE firmware is made. It has an interface for specifying the bitmap of the segment layout in the LCD glass.

I've prepared BLE firmware with th following configuration:
```
#define DEVICE_LYWSD03MMC	3	// LCD display LYWSD03MMC
// HW_VER_LYWSD03MMC_B14 = 0
// HW_VER_LYWSD03MMC_B19 = 3
// HW_VER_LYWSD03MMC_B16 = 4
// HW_VER_LYWSD03MMC_B17 = 5
// HW_VER_LYWSD03MMC_B15 = 10

#define DEV_SERVICES ( SERVICE_OTA \
		| SERVICE_OTA_EXT \
		| SERVICE_PINCODE \
		| SERVICE_BINDKEY \
		| SERVICE_HISTORY \
		| SERVICE_SCREEN \
		| SERVICE_THS \
		| SERVICE_KEY \
		| SERVICE_LED \
)

#define USE_SENSOR_AHT20_30		1

#define GPIO_KEY2			GPIO_PD4	// key "Connect", input, pcb mark "P8"
#define PD4_INPUT_ENABLE	1
#define PD4_DATA_OUT		0
#define PD4_OUTPUT_ENABLE	0
#define PD4_FUNC			AS_GPIO
#define PULL_WAKEUP_SRC_PD4 PM_PIN_PULLUP_1M

#define GPIO_LED			GPIO_PC0 // (LED E)
#define PC0_INPUT_ENABLE	1
#define PC0_DATA_OUT		1
#define PC0_OUTPUT_ENABLE	0
#define PC0_FUNC			AS_GPIO
#define PULL_WAKEUP_SRC_PC0	PM_PIN_PULLDOWN_100K

#define PULL_WAKEUP_SRC_PB7	PM_PIN_PULLUP_1M // UART TX (B1.6)
```

And device up and I able to connect via BLE

I'm getting H&T data, battery but nothing with screen, it's don't turn on (
Detect as B1.6, according to code, nothing had been detected on i2c and fall back to 1.6 

[image]

Any suggestions/ideas what I can done with LCD to init it?

vkravets, 2024-11-22:
I've notice you can custom command which can be send to device, I'm interesting to I2C scan... How I can do this?

pvvx, 2024-11-22:
Enter command "03"
View log and Chrome console....
https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#control-function-id-when-connected

https://github.com/pvvx/ATC_MiThermometer/blob/master/src/cmd_parser.h#L6
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/cmd_parser.c#L270

vkravets, 2024-11-24:
> Have some additional info:
> 
> * LCD driver is  BL55028
> * LCD the same as TH05 (BLE version)
> * LCD driver connected to SoC via UART
>   
>   * PB1/TX -> BL55028 SCI (Pin 3)
>   * PB7/RX -> BL55028 SDA (Pin 2)

My bad, I miss the R between driver, thus It's connected to I2C bus as needed with R2 which the same capacity as R4 for tha TH sensor. It's seems need to pullup some pin to wakeup it before I2C init

Sorry for it...  My bad (

pvvx, 2024-11-27:
https://pvvx.github.io/TS0201_TZ3000_TH03/

pvvx, 2024-11-28:
BLE firmware "TH03_v48.bin" added.
TH03 power consumption is twice as much as LYWSD03MMC (!) The reason is the LCD driver chip and T&H sensor.
Another emulation of a thermometer from Tuya. Buy and throw away.

vkravets, 2024-11-28:
Thanks, for your work and quick response. 
I'm agree with such power consumption there is no sence to continue work at ZB firmware.

Close. 

In any case who dissagree can take work from BLE and migrate here

closed 2024-11-28 by vkravets

pvvx, 2024-11-29:
Revert to original Tuya Zigbee firmware:

[image]

pvvx, 2024-11-30:
[image]

nsandau, 2025-04-14:
@pvvx 
How can I flash the Zigbee -> BLE firmware OTA? 
I am using Zigbee2MQTT but the device gets detected as a WSD500A and it ignores my local index with the firmware. 

Also, thank you so much for the huge amount of work that you clearly put into supporting these devices!