pvvx/ZigbeeTLc #131: TS0201_TZ3210_ncw88jfq

kszaboa, 2024-11-16:
Hi pvxx,

any chance for a FW for this tuya zigbee thermometer? 
It can be connected to ZHA but do not update values unless you connect it first to a tuya hub. Then it updates every 5 min. Same procedure on every battery change :(

[image]
[image]
[image]
[image]

pvvx, 2024-11-16:
I ordered, I'll see when it's delivered.

[image]

juha63, 2024-11-22:
I found some informations here: https://www.elektroda.com/rtvforum/topic4042463.html

pvvx, 2024-11-22:
> I found some informations here: https://www.elektroda.com/rtvforum/topic4042463.html

This is about the WiFi version.

pvvx, 2024-12-05:
https://pvvx.github.io/LKTMZL02/

[image]

The AHT20 sensor is terrible:

[image]

kszaboa, 2024-12-14:
mine is not shown as LKTMZL02
[image]

pvvx, 2024-12-14:
LKTMZL02-z is a [ZigbeeTLc firmware](https://github.com/pvvx/ZigbeeTLc/tree/master/bin).

kszaboa, 2024-12-15:
[image]

mine is not so bad, TS0201_TZ3000_fllyghyj (orange) vs TS0201_TZ3210_ncw88jfq (blue)
the new one shows even higher humidity levels, but the difference is only 1-1,5%

pvvx, 2024-12-15:
TS0201_TZ3000 usually has a CHT8305 sensor. Humidity readings are usually 10% too low.

stonie08, 2025-01-25:
What's the status on this sensor? Should it work with ZigbeeTLc? I have this exact sensor with the same firmware version 0x00000083. The sensor is set to Zigbee by default. Which firmware version are we supposed to flash in which order?

I just barely recovered from a bricked sensor and don't know where I might have gone wrong. I get a bit confused by the documentation and the sheer number of files there are. So outline what I did and maybe to leave some pointers for others who might stumble upon this post:
[The page for the sensor](https://pvvx.github.io/LKTMZL02/) links at the bottom to the pvvx/ATC_MiThermometer repo that seems to be for the BLE variant of the sensor and also holds the [LKTMZL02_v50](https://github.com/pvvx/ATC_MiThermometer/blob/master/LKTMZL02_v50.bin). This seems to be the wrong one in my case?

The ZigbeeTLc repo has the bin folder that holds the [1141-021f-01243001-LKTMZL02Z.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-021f-01243001-LKTMZL02Z.zigbee) and the [1141-d3a3-01993001-LKTMZL02z_v0124.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-d3a3-01993001-LKTMZL02z_v0124.zigbee) but also the [LKTMZL02Z_v0124.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/LKTMZL02Z_v0124.bin). The two former are explained a bit in the readme but the last one isn't. What is that one for?

Based on the readme in that bin folder I went with the 1141-d3a3-01993001-LKTMZL02z_v0124.zigbee which seems to be for that initial flash?

Getting ZHA to do it was another challenge because there were a number of ways to do this in the issues. In the end [based on the documentation](https://github.com/zigpy/zigpy/wiki/OTA-Configuration) I added the following to my configuration.yaml, created a `/homeassistant/zigbee_ota` folder and put the [1141-d3a3-01993001-LKTMZL02z_v0124.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-d3a3-01993001-LKTMZL02z_v0124.zigbee) file in the folder:
```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: advanced  # !!! Please read the message below before enabling OTA updates from raw OTA files !!!
          warning: "I understand I can *destroy* my devices by enabling OTA updates from files. Some OTA updates can be mistakenly applied to the wrong device, breaking it. I am consciously using this at my own risk."
          path: ./zigbee_ota
```

After doing some more trickery to get ZHA to realize there's an update available I started the update. It ran to 100% but then stopped and the sensor wouldn't respond anymore. I waited a bit longer, but the display wasn't updating anymore, so I removed the battery but the display stayed off. 
I then tried to get to flash any firmware with the [USBCOMFlashTx](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) tool and luckily the "[Original FullFlash.bin](https://pvvx.github.io/LKTMZL02/bin/LKTMZL02_ff.zip)" restored my device. It should be noted that if one's USB2serial device doesn't have an RTS pin connecting the three others is enough.

pvvx, 2025-01-25:
> also the [LKTMZL02Z_v0124.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/LKTMZL02Z_v0124.bin). The two former are explained a bit in the readme but the last one isn't. What is that one for?

For flashing with a programmer or for creating a *.zigbee OTA file by adding a header. With *.bin or *.zigbee it is possible to make BLE OTA in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

pvvx, 2025-01-25:
> I waited a bit longer, but the display wasn't updating anymore, so I removed the battery but the display stayed off.

After a while, you need to put the battery back in.
It takes time for the internal capacitor in this device to discharge.
It also has a reset button... The old settings written by the original version in Flash are not suitable.

pvvx, 2025-01-25:
Where is everyone rushing to?

The files for LKTMZL02Z_v0124 are signed - v0.1.2.4 beta2.
This means that this version has not yet been tested for all options and may freeze, etc. You should not use beta versions without the ability to restore via a programmer.

The beta version may change after some time. But you won't be able to do Zigbee OTA, since the number is the same.
You'll need to do a forced update via Zigbee OTA. And that's not available in ZHA.
You will have to create a new header for the *.zigbee file, replacing the version number with a higher one.

stonie08, 2025-01-26:
> For flashing with a programmer or for creating a *.zigbee OTA file by adding a header. With *.bin or *.zigbee it is possible to make BLE OTA in [TelinkMiFlasher.html](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)

Okay I see. Now I flashed this one with a programmer and it indeed worked. Nice! Thank you

> After a while, you need to put the battery back in.

Poorly phrased on my part. Of course I put the battery back in, the display _on the device_ stayed off. It also didn't react to the reset button. Pressing/holding the reset button also didn't cause any reaction of the LED. So it definitely seemed like it was bricked. 

> You should not use beta versions without the ability to restore via a programmer.

Hardly a way to avoid because the latest release is 1.2.2 and doesn't yet have a firmware for this sensor. I wanted to test how well it works with the custom firmware before buying more of them and because the original fw one has some issues with the humidity in ZHA.

> But you won't be able to do Zigbee OTA, since the number is the same.

I guess I can just update with a programmer once again.

pvvx, 2025-01-26:
> Hardly a way to avoid because the latest release is 1.2.2 

Before LKTMZL02Z_V0124 there was a tested version of ZL02_V0123.
The current LKTMZL02Z_V0124 (beta2) has already been tested for the main errors in LKTMZL02.

pvvx, 2025-05-26:
> Will this custom FW work directly with Zigbee2MQTT (instead of ZHA), or will it require external custom components?

For Zigbee2MQTT, all new devices require additional scripts to be created. And wait until the Zigbee2MQTT author includes these scripts in the update.

Bjk8kds, 2025-05-27:
> For Zigbee2MQTT, all new devices require additional scripts to be created. And wait until the Zigbee2MQTT author includes these scripts in the update.

Oh okay, thank you so much..

Btw. Mine doesn't show the firmware version (default) so I can't update OTA manually, I've tried using ZHA but the firmware version doesn't show either.
Is there a problem with mine? Do I have to use brute force (I mean use USB TTL)?
<details>
  <summary>image</summary>

[image]
</details>

mrmaximas, 2025-09-03:
@pvvx hi! I bought visually identical sensors here https://aliexpress.ru/item/1005006963419948.html but they seem to be classic Zigbee, not TUYA. 

`{"id":149,"type":"EndDevice","ieeeAddr":"0xa4c138362e09af72","nwkAddr":28802,"manufId":4417,"manufName":"TELINK","powerSource":"Battery","modelId":"TLSR82xx","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,32,1026,1029,1],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"TELINK","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b002a1aa3c4","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1756813462154,"checkinInterval":3600}`

I planned to use them in my native TUYA, but the seller seems to have deceived me.
Perhaps you still have the TUYA firmware for these devices?

mrmaximas, 2025-09-03:
@pvvx thanks! I only have this two usb-ttl adapters (most likely they are the same):

[image]
[image]

Is there any chance to flash the native TUYA firmware `LKTMZL02_ff.bin` through? 
Maybe there is a way to flash it over the air using z2m?

pvvx, 2025-09-03:
At address 0xFF000 replace the underlined bytes with whatever you like. 

[image]

Or change all 8 bytes to 0xff

Or trim the firmware file to size 0xFF000 (1044480 bytes)

pvvx, 2025-09-03:
>  I only have this two usb-ttl adapters (most likely they are the same):

These adapters do not contain a 3.3V stabilizer. When connected to USB, 5.2V is often supplied, and this adapter has two diodes, on which 1.2V drops. As a result, the chip in the thermometer will receive 4V and will be damaged.

mrmaximas, 2025-09-03:
@pvvx I also have this module, but it's not something I can flash, is it?

[image]
[image]

[image]

pvvx, 2025-10-07:
It is advisable to connect PD3 -> RST and PD4 -> SWS.
And set the duration of the reset signal, for example "-t200".

`python3 TlsrPgm.py -pCOM9 -t200 -a500 -s i`

Without a connection to RST, it is advisable to first stop the CPU and then read the firmware...

```
python3 TlsrPgm.py -pCOM9 -z11 -a-70 -s i
python3 TlsrPgm.py -pCOM9 rf 0 0x100000 ff.bin
```

mrmaximas, 2025-10-07:
@pvvx what is the right command to flash original tiuya fw from https://pvvx.github.io/LKTMZL02/ ?

mrmaximas, 2025-10-07:
@pvvx I flashed the original TUYA firmware and the trimmed to 0xff000. My goal was to flash the firmware from TUYA, but TUYA does not accept it as unauthorised. I opened a dispute on AliExpress. The chinese have deceived me twice already. I flashed your own firmware, and it works perfectly in z2m. Thank you very much for your help! 🙏🏻