pvvx/ZigbeeTLc #132: Add more details about image type gathering in README.md

ss89, 2024-11-17:
I have bought one of the TS0201 mentioned [here](https://pvvx.github.io/TS0201_TZ3000/) from: [aliexpress](https://de.aliexpress.com/item/1005005352101380.html?gatewayAdapt=glo2deu)

The pages [Original OTA ver 70](https://pvvx.github.io/TS0201_TZ3000/1141-d3a3-00463001-ts2001.zigbee) link references this file: `1141-d3a3-00463001-ts2001.zigbee`, so image type must be `d3a3` (assuming they did not change the internals of the same article from when pvvx received it from their order vs. when I received it from my order.

However without knowing this page, I wouldn't know the image type and the README.md doesn't specify on how to get it either.
I'd assume that i could get it from the ota firmware flasher if it was a bluetooth device.
If i'm not completely wrong, then I can not use the [ota firmware flasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) page to connect to it, because it has a zigbee firmware on it already, so no OTA update from bluetooth enabled device possible.

At least during resetting it (making it blink multiple times) it did not show up under a similar mac in the device connect view.
There were lots of other devices, however I'd rather not want to try out ~20 devices until I find the correct one.

What should be possible via zigpy/HA/ZHA is the OTA via zigbee.
I have added this to my configuration.yaml for HA:
```yaml
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m
          url: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index.json
```
Shouldn't I get a firmware update suggestion with this?

My device Signature in HA looks like this:
```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0003",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_xr3htd96",
  "model": "TS0201",
  "class": "zigpy.device.Device"
}
```

Any help is very appreciated.

ss89, 2024-11-17:
I do see that I as a user in this case have to check which device I have.

My main problem is: I don't see where I should find the "Image Type" myself, as it is neither printed on the device nor is it part of ZHA device information. I would now rely on the information that you are hosting here already (https://pvvx.github.io/TS0201_TZ3000/) and try to update the device with the most recent version for `1141-d3a3-*`.
However it would be nice if it was mentioned somewhere how I do find both of the values.

I do see that you do not take any responsibility etc. etc. etc.
I just hoped that you could give me a little push in the right direction - and update the readme with more detail on how to get the image type.

pvvx, 2024-11-17:
> My main problem is: I don't see where I should find the "Image Type" myself, as it is neither printed on the device nor is it part of ZHA device information. 

This is a question for ZHA support.
"Manufacturer Code", "Image Type", "File Version", "Stack Version" - these are standard data received from the device in Zigbee 3.0.
Tuya devices do not support identification by this data. Tuya has its own Zigbee standard.
As a result, completely different Tuya devices have the same version and firmware type identifiers.

ss89, 2024-11-17:
> "Manufacturer Code", "Image Type", "File Version", "Stack Version" - these are standard data received from the device in Zigbee 3.0.

Fair point. THIS is the information I needed. So I should be able to get this information from ZHA and if I do not get it somewhere, then this is an issue of ZHA not being verbose enough. Correct?

pvvx, 2024-11-17:
> Correct?

Yes.

---

If you automatically receive an update for any Tuya device, it is recommended to double-check everything several times.
Do not expect that this update will work for your Tuya device.

pvvx, 2024-11-17:
There are no settings or information about Zigbee devices in ZHA. All settings are set as the creators of ZHA want, by setting constants in the ZHA program code itself. The user cannot change anything. In Z2M, for any Zigbee device, you need to describe special scripts for setting up and working with the program version variants of this device. For an uninitiated user who does not know the Zigbee protocols and does not know how to program, this is not suitable at all.