pvvx/ZigbeeTLc #137: LYWSD03MMC b1.5 with TSLR8251 chip

anight, 2024-11-29:
Hello,
Just bought couple of them on aliexpress
I'm wondering if 32k RAM of 8251 chip would be enough to run firmware Z03MMC_v0123 ?
Thanks!

pvvx, 2024-11-30:
Who told you that there is only 32 KB of RAM?
All TLSR825x have the same die and all have 64 KB of RAM. The flash die is built into the package for 512 KB or 1 MB.
The differences between the chips in this series are in the number of pins brought out to the plastic case.
Everything else is marketing.

closed 2024-11-30 by pvvx

anight, 2024-11-30:
@pvvx Well, that's what datasheets say and that's confusing.
I totally trust with your experience with the chips, it looks like you may be right - I just flashed my device with zigbee fw and it works just fine!
Thank you for the explanations and great project!