pvvx/ZigbeeTLc #138: Displaying arbitrary data on the LCD

davystrong, 2024-12-02:
Hello!

I have been using the [BLE custom firmware](https://github.com/pvvx/ATC_MiThermometer) for a while. Specifically, I use it in active mode and display the outside temperature while reporting the inside temperature to Home Assistant.

However, bluetooth doesn't work particularly well because it's quite far away from my server. I would like to use Zigbee instead as I have other Zigbee devices to use as a network, but [I asked about that previously](https://github.com/pvvx/ATC_MiThermometer/issues/545) and was told that updating the display isn't supported on the Zigbee firmware. Is there any specific reason for this? Is there any way I can implement this?

If it's relevant, my device is a C401. I think it's the new version (C401N).

pvvx, 2024-12-02:
Zigbee does not have standard functions for transmitting data to the screen. ZHA and Z2M do not support such functions.

pvvx, 2025-02-18:
Zigbee is not a priority for me, because I use BLE. I have too many devices and Zigbee will overload, require several coordinators... Therefore, we will have to wait a long time for the addition of some functionality to Zigbee...