pvvx/ZigbeeTLc #139: Another ZTH01 but CHT8310 sensor

ferrets6, 2024-12-06:
Hi,
I have this ZHT01 with sensor CHT8310.

[image]
[image]
<br><br>

I read this issue [Another ZTH01](https://github.com/pvvx/ZigbeeTLc/issues/108) and followed your [comment](https://github.com/pvvx/ZigbeeTLc/issues/108#issuecomment-2378020296) to flash this firmware [ZTH01Z_v0123.bin](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/ZTH01Z_v0123.bin) via UART

And now I have this on Z2M
[image]

Well... it's not that cold in my home 😞 

I tried with and without converter [zigbeetlc_v0121.js](https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/zigbeetlc_v0121.js), no changes.

Maybe the sensor CHT8310 it's not compatible? Or should be something else?

pvvx, 2024-12-07:
CHT8310 is not supported in TLSR825x firmware yet.
I have not had any such thermometer with TLSR825x and CHT8310.
It is supported in [PHY62xx](https://github.com/pvvx/THB2).

ferrets6, 2024-12-10:
I'm trying to understand how to make an adapter for CHT8310, but I'm afraid it's too difficult...
CHT8310 Advanced Datasheet has 12 registers vs CHT8305 which has only 6.

I think we can use only the top 4, with **_0x02 Config_** of CHT8305 split in two different registers in CHT8310 **_0x02 Status_** and _**0x03 Config**_
[image]

Also, the temperature and humidity formulas are quite different, from these on CHT8305
[image]
[image]

to these on CHT8310
[image]
[image]

I don't know if it's the right way...

pvvx, 2024-12-10:
If I find a thermometer with the TLSR825X chip and the CHT8310 (8215) sensor, then I will add support to ZigbeTlc...
[An example of working with the CHT8310 (8215) sensor](https://github.com/pvvx/THB2/blob/master/bthome_phy6222/source/sensors.c#L76) in the project [THB2](https://github.com/pvvx/THB2).