pvvx/ZigbeeTLc #142: LYWSD03MMC cannot reconnect to repeater

mmib, 2024-12-18:
Have LYWSD03MMC with 04 Sep firmware. It works fine until repeater got down (e.g. electricity outage). After this device do not automatically reconnect to repeater. Do reconnect after reboot (remove and insert battery). Other devices  (soil moisture sensors) connected to the same repeater works fine with no reconnect issue.

pvvx, 2024-12-18:
https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#action-of-the-button

mmib, 2024-12-19:
not sure how it can help? Maybe I explained the problem a bit unclear. Binding was ok and device is working and it do reconnect after battery jerking. However it should reconnect to network automatically without any manual work like other devices do.

pvvx, 2024-12-20:
This behavior is not observed with a normal Zigbee coordinator.

1. In ZigBee 3.0, a maximum time is set before a device is excluded from the network if there are no responses from it. This is not a function of the device, but a function of the coordinator and repeaters.

2. After the ZigBee network drops, the thermometer searches for the network every 3 minutes. This is done to minimize consumption when there is no ZigBee network. But still, in this mode, the battery consumption is much higher.

3. The Zigbee network is not designed to have the coordinator and/or routers disconnected. Install a standalone router. Otherwise, 90% of your devices will periodically disappear from the unstable ZigBee network.

mmib, 2024-12-20:
Okay, thank you. All other devices do not have such issue, that's why I thought something could be tweaked/fixed in firmware.

pvvx, 2024-12-20:
When testing the coordinator shutdown for more than a day, the Zigbee Xiaomi aqara sensors always fell off. Others did not.