pvvx/ZigbeeTLc #143: Zigbee support for (old) MHO-C401 and MJWSD05MMC still planned?

NuclearKitty, 2024-12-20:
Thinking in 2024 I wouldnt get any old sensors, I recently bought 5 MHO-C401 sensors (and a MJWSD05MMC, thinking "in develpment" means "beta fw available" - my bad). 

Turns out, 3 of my 401s were the old version (date on package 03.2024, date on PCB 05.2020).

I wanted to ask if support for these sensors in Zigbee is still planned? I really dont want to use them with BLE and add another protocol to my already convoluted enough home setup.

AfromD, 2025-03-05:
I don't get the point of your criticism. For your use case you are happy with BT. Fair enough. But you are commenting in a repository that is about Zigbee. It is literally called "ZigbeeTLC". Having a Zigbee wall thermostat does make sense, because there are several heating systems that operate on Zigbee. None on BT as I know of. So it is not for your garden shed, but you are trying to achieve different things.

pvvx, 2025-03-05:
I don't have Zigbee heating systems. And I don't see them for sale.
And the ZigbeeTLc project is supported only by me and I described to you the reasons why I think that MJWSD05MMC does not need to be reflashed in Zigbee.

AfromD, 2025-03-05:
Tuya, Bosch, Danfoss, Hive and Sonoff do zigbee heating. Right now, battery wall thermostats are very few available, and they all have displays that need backlighting/interaction to be read. This device has buttons, a temp sensor, a screen that is always on, and a radio that can do zigbee. And it costs a fraction of available devices. So it would be a pretty good fit and a pretty cool conversion if it would be a thermostat. At least what I thought.

pvvx, 2025-03-05:
To support the full functionality of MJWSD05MMC in Zigbee, additional Zigbee clusters and their maintenance code must be added to the FW. ZHA and Z2M do not have the functionality to work with these standard Zigbee 3.0 additions.

These additions require the creation of separate code for this project. They do not fit into the general code, as it requires too many differences from typical thermometers with a screen.

And let me remind you once again:
> Any new version of the program requires testing on dozens of devices. I do not have that much time for such tests. For these reasons, adding support for new devices is limited.

pvvx, 2025-03-05:
> Tuya, Bosch, Danfoss, Hive and Sonoff do zigbee heating. 

Tuya is a Chinese brand for selling devices that use inexpensive components from junk. They are manufactured by small companies. The main goal of Tuya is to recycle defective and expired components from China. Otherwise, recycling such components requires funds... Tuya uses a different Zigbee protocol from the Zigbee 3.0 standard.

Sonoff is a similar company that makes cheap devices out of junk.