pvvx/ZigbeeTLc #144: MHO-C401N Detected as old MHO-C401

csutcliff, 2024-12-29:
Hi,

I have a batch of 8 devices purchased together from the same supplier. Some of these are properly detected as MHO-C401N and are working perfectly.

A number of them are detected as the old MHO-C401:

```
01:14:46: Searching for devices
01:14:50: Connecting to: MHO-C401
01:15:07: Hardware Revision String: V_14
01:15:07: Software Revision String: 0016
01:15:07: Firmware Revision String: 1.0.0_0016
01:15:07: Detected Mi device
01:15:07: Connected
01:15:07: Found old version of SW:14 MHO-C401
```

However when flashed with the Zigbee or BLE Custom firmware for the MHO-C401 as detected they have no screen output. The device remains working via BLE/ZigBee just the screen remains at the state from when the flash finished.

If I manually flash the firmware for MHO-C401N to these devices the screen begins to work again (along with everything else).

This leads me to believe they are actually MHO-C401N devices, and are being wrongly detected as the old MHO-C401.

I've attached a picture of the board inside the device if it helps:

[image]

pvvx, 2024-12-29:
https://github.com/pvvx/ATC_MiThermometer/issues/586#issuecomment-2558312694

csutcliff, 2024-12-29:
These seem to contradict that, they are version 14 but are N models

pvvx, 2024-12-29:
There are no other options for determining the current HW version.
Suggest your software version of the definition of HW.

Frank1616, 2025-01-11:
@csutcliff 
I have exactly the same problem. I bought two devices via AliExpress. I tried to flash the Zigbee firmware with the result that the display froze. But I can connect the devices with Home Assistant via Zigbee (ZHA and Z2M both work)
The devices are recognized as Zigbee model MHO-C401-z. (without “N”)
The circuit board is exactly the same as in your photo. And also the log entries during the flash process were identical.

You wrote: "_If I manually flash the firmware for MHO-C401N to these devices the screen begins to work again (along with everything else)_."
How did you get it work to flash the correct zigbee firmware? (for MHO-C401N  I guess)

csutcliff, 2025-01-11:
@Frank1616 

First you have to flash the wrong BLE firmware (401) via Zigbee OTA I used Z2M. see: https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

The screen still won't work but you can then use the BLE flasher that you used in the first place. Don't press the flash zigbee button but manually browse for the correct 401N firmware that you've downloaded and flash that.

Frank1616, 2025-01-11:
fast reply - thx!
to be sure - to go back to BLE I have to use the firmware for MHO-C401 - not MHO-C401N ?

csutcliff, 2025-01-11:
yes, because the zigbee firmware you flashed is for non-N the zigbee OTA will only allow you to flash matching firmware, hence you have to go back to BLE and manual flash from there.

Frank1616, 2025-01-11:
o.k. this was quite hard stuff for me. But I think I get it worked.
Finally I was able to flash the Firmware ZMHOC401N_v0123r2.bin and  the device is still alive. And the display works again!

I did a pairing with Z2M. But it is still recognized as Zigbee Model MHO-C401-z (without N) ????
And there is still the need for an external converter...

I tried before this one:
https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/zigbeetlc_v0121.js
but there is not really support for Zigbee Model MHO-C401-z.

it worked halfway as I added the entry
{modelID: 'MHO-C401-z', manufacturerName: 'MiaoMiaoCe', model: 'MHO-C401-z'},

@csutcliff 
Do you use the sensor also with Z2M? Do you have a suggestion for the right entry in the external converter?

pvvx, 2025-01-11:
> yes, because the zigbee firmware you flashed is for non-N the zigbee OTA will only allow you to flash matching firmware, hence you have to go back to BLE and manual flash the correct "N" firmware from there.

Any firmware for Zigbee can be signed with the required IDs to work with Zigbee OTA.
There are already many scripts on the Internet for this purpose.

You can create a Zigbee OTA header for BLE firmware like this:
python3 utils\zb_bin_ota.py MHO_C401_v49.bin ZMHO_C401_v49 -m0x1141 -i0x0208 -v0x99993001 -s"MHO-C401Nz to MHO-C401b"

Frank1616, 2025-01-11:
The flash sequence was:
1. Original BLE firmware -> Zigbee-non-N (Display frozen)
2. then back -> BLE-non-N-firmware "1141-0201-99993001-MHO_C401_v49.zigbee"
3. now I thoght followed the change with  BLE-N-custom firmware "MHO_C401N_v49.bin" (after that the display works again)
4. and finaly the Zigbee-N-firmware  ZMHOC401N_v0123r2.bin

but the result in Z2M is still Zigbee Model "MHO-C401-z"

this is the log of the last flashing step:

> 17:04:28: Reconnect
> 17:04:34: Hardware Revision String: 2022
> 17:04:34: Software Revision String: V4.9
> 17:04:34: Firmware Revision String: github.com/pvvx
> 17:04:34: Detected custom Firmware
> 17:04:34: Hardware Version: MHO-C401N 2022, Software Version: 4.9
> 17:04:34: Custom config HEX string: 55071000002808a93163085a
> 17:04:45: Send Ext data: 22D20463005802D7 ok
> 17:04:45: ExtShow: big_number: 1234, small_number: 99, ext.vtime: 600 s, flg: D7
> 17:04:49: Send Ext data: 22D20463005802D7 ok
> 17:04:49: ExtShow: big_number: 1234, small_number: 99, ext.vtime: 600 s, flg: D7
> 17:06:12: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZMHOC401N_v0123r2.bin'...
> 17:06:12: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/ZMHOC401N_v0123r2.bin
> 17:06:12: File size: 128628 bytes
> 17:06:12: Count: 8040
> 17:06:23: Start DFU
> 17:07:26: Update done after 62.046 seconds
> 17:07:30: Disconnected.

pvvx, 2025-01-11:
```
python3 zb_bin_ota.py ZMHOC401N_v0123r2.bin ZMHOC401toMHOC401N -m0x1141 -i0x0201 -v0x99993001 -s"MHO-C401z to MHO-C401Nz"
python3 zb_bin_ota.py ZMHOC401_v0123r2.bin ZMHOC401NtoMHOC401 -m0x1141 -i0x0208 -v0x99993001 -s"MHO-C401Nz to MHO-C401z"
```

[zb_bin_ota.py](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/make/zb_bin_ota.py)
https://github.com/pvvx/ZigbeeTLc/tree/master/bin

[Zigbee Image Type](https://github.com/pvvx/pvvx.github.io?tab=readme-ov-file#id-numbers-of-alternative-firmware)

Frank1616, 2025-01-11:
@pvvx 
Many thanks for your support!
I greatly appreciate your work.
And do my best to follow your tips.

csutcliff, 2025-01-12:
I think I'm using the built in support, no external converter.

Perhaps it has remembered the wrong firmware device?

Try some combination of remove & re-pair and/or re-iterview and reconfigure?

Frank1616, 2025-01-12:
@csutcliff 
You are right, there is a problem with recognition within Z2M. When I pair the new flashed temp sensors with Home Assistant ZHA (secondary test installation) the devices are shown correct as 'MHO-C401N-z'.
Now I will try get it work with Z2M...

Frank1616, 2025-01-12:
Finally everthing works as desired. The right Zigbee firmware 'MHO-C401N-z' in combination with Zigbee2MQTT without external converters. (an update of the coordinator software seemed to be helpful)

@csutcliff  many thanks for beginning this thread and your assistance. Otherwise these both temp. sensors would be crap for me.

gabrielambuehl, 2025-01-16:
Ran across the same issue, just wanted to say thanks to pvvx for outlaying the fix so clearly (and the work on the firmware, of course).

pgaspard2177, 2025-01-23:
> The flash sequence was:
>
> 1. Original BLE firmware -> Zigbee-non-N (Display frozen)
> 2. then back -> BLE-non-N-firmware "1141-0201-99993001-MHO_C401_v49.zigbee"
> 3. now I thoght followed the change [...]

Hello, how did you do "2. then back -> BLE-non-N-firmware "1141-0201-99993001-MHO_C401_v49.zigbee" ?
I've tried everything with my "MHO-C401-z" device in z2m : configuration.yaml modification, custom index.json, local file for ota
This device is still "unsupported" in z2m frontend, doesn't appear in "OTA" when i click on "Check all", and is unable to be flashed... 
In logs, I don't see anything related to this device
Thank you

gabrielambuehl, 2025-01-23:
I cross-flashed mine to the N firmware directly through Zigbee (so no detour through BLE). After pairing it again it still identifies as MHO-C401-z but is otherwise behaving completely normally. It shows up under OTA but obviously already has the latest firmware so no OTA available.

Did you force remove and pair it again?

pgaspard2177, 2025-01-23:
> I cross-flashed mine to the N firmware directly through Zigbee (so no detour through BLE). After pairing it again it still identifies as MHO-C401-z but is otherwise behaving completely normally. It [...]

Yes, I see this device in z2m as a "MHO-C401-z", it's "unsupported" but it works with updates once in an hour only, and I can't change that.
I've flashed another MHO-C401 months ago and this one is seen in z2m as a LYWSD03MMC-z model and MHO-C401N-z zigbee model, and exposes more things like comfort display, comfort temperature/humidity.
I don't understand these differences between the 2 devices. Did I accidentally flashed MHO-C401 with a LYWSD03MMC firmware ? But I don't think it's possible

mmirannalti, 2025-01-24:
> I cross-flashed mine to the N firmware directly through Zigbee (so no detour through BLE). After pairing it again it still identifies as MHO-C401-z but is otherwise behaving completely normally. It [...]

Thanks for sharing!
I will try the same route.

gabrielambuehl, 2025-01-24:
Mine is showing up as MHO-C401-z and  LYWSD03MMC-z respectively but reacts normally to adjustments on reporting etc.

`
$cat /config/zigbee2mqtt/my_index.json
[
    {
        "url": "mho.ota",
        "force": true
    }
]
`
the OTA was prepared as outlined by pvvx above. Github does not allow me to add it.

frank-f, 2025-02-16:
I bought a few C401s from AliExpress recently. They are all C401N, but the web flashing tool detects them as C401. Consequently it then selects the wrong firmware.

Here's my log for that (I manually selected the correct firmware for this one):
```
12:11:08: Searching for devices
12:11:14: Connecting to: MHO-C401
12:11:17: Hardware Revision String: V_14
12:11:17: Software Revision String: 0016
12:11:17: Firmware Revision String: 1.0.0_0016
12:11:17: Detected Mi device
12:11:17: Connected
12:11:17: Found old version of SW:14 MHO-C401
12:11:20: Activating now, please wait...
12:11:24: Registration successful
12:11:24: Send Login, please wait...
12:11:24: Received device infos are correct
12:11:24: Login successful
12:12:55: File: ZMHOC401N_v0124.bin
12:12:55: File size: 129652 bytes
12:12:55: Count: 8104
12:12:56: Start DFU
```