pvvx/ZigbeeTLc #145: TS0202_TZ3000

claviger-pc, 2024-12-31:
Is there firmware available for testing? I have a newer version of this device (dated 2024-08) and I would be happy to test. It does not have uart pads on the back, but I can just solder straight to the module.

pvvx, 2024-12-31:
Zigbee does not provide testing capabilities.
UART operation requires constant operation of the chip.
And what is there to "test" via UART?

claviger-pc, 2024-12-31:
By testing, I mean flashing the module and seeing if it works with the newer revision. UART I would only use for flashing, not for testing.

pvvx, 2024-12-31:
The chip has a great hardware single-wire debug interface in the form of Telink SWire at 2Mbit/s. Through it, there is access to all SoC hardware. Analog of JTAG.

claviger-pc, 2024-12-31:
yes, I have used that on other devices for the one-wire flashing. Do you have firmware for this TS0202_TZ3000? or any source code?

pvvx, 2024-12-31:
If you have a device, then you can read its firmware. The chip has no protection against reading via SWS.

claviger-pc, 2024-12-31:
I'm sorry maybe I was not clear:

Do you have custom firmware or are you developing custom firmware?

pvvx, 2024-12-31:
For TS0202 (_TZ3000_lf56vpxj) I did not make firmware. They work more or less normally.