pvvx/ZigbeeTLc #146: MHO-C401 Screen Background Not Refreshed

Mortim-zz, 2025-01-05:
Hi pvvx, thanks a lot again for all your work.
I tried the zigbee version for the now supported MHO-C401 (with no N), and everything works apart from the screen background slowly turning gray.
I compared the screen refresh, and the zigbee version of the firmware does not refresh the background, only the segments are changing. On the non-zigbee-version the whole screen is turning black and then white.
When removing and reinserting the battery, the screen is also refreshed correctly once on the zigbee version.
This is a device which came with the newer firmware (from pvvx/ATC_MiThermometer#586)

[image]

pvvx, 2025-01-05:
Apparently, the manufacturers found some outdated E-Ink somewhere...
I'll take a look soon and fix the E-Ink regeneration enablement.

pvvx, 2025-01-05:
fixed V0.1.2.3 rev2

Mortim-zz, 2025-01-06:
I'm not sure how to install this OTA, as it has the same version as the installed one, so it doesn't turn up in z2m.
Also, there seems to be a small error with the file name in index.json it should be
`https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-0201-01233001-ZMHOC401.zigbee` not
`https://github.com/pvvx/ZigbeeTLc/raw/master/bin/1141-0208-01233001-ZMHOC401.zigbee` (0201 instead of 0208)

Mortim-zz, 2025-01-06:
Nevermind, I switched to the bluetooth version and back.
Seems to work fine, thank you!

yamar8, 2025-01-07:
Hi @Mortim-zz 
Can you please help me with go back to Bluetooth version?
I've tried without success.
The problem is that my screen is frozen after I moved to the ZigBee version.

Thanks.

Mortim-zz, 2025-01-07:
@yamar8, you can have a look at https://github.com/pvvx/ATC_MiThermometer/issues/586#issuecomment-2558316249 and https://github.com/pvvx/ATC_MiThermometer/issues/586#issuecomment-2564336972