pvvx/ZigbeeTLc #147: ZTH05 firmware

Ctrl-F4, 2025-01-11:
I have this Tuya ZigBee [ZTH05 ](https://pvvx.github.io/TS0601_TZE204/)wich i want to flash with ZigbeeTLc firmware. 
For BLE there is ZTH05_v49.bin firmware.
Can you provide the ZigbeeTLc firmware also? 
Thanks for considering my request.

pvvx, 2025-01-19:
[ZTH05](https://pvvx.github.io/TS0601_TZE204/) requires modification - removal of excess microcircuit and installation of conductors.

And as of today, it has only been [tested with the AHT30 sensor](https://github.com/pvvx/ATC_MiThermometer/issues/589#issuecomment-2563769349).

tech1977, 2025-02-21:
Hi all,
I have a Tuya ZigBee ZTH05 sensor connected with HA Zigbee2MQTT and I would like to flash with ZigbeeTLc,
after the hardware modification can I do the webupdate? how is the sensor detected that is not Bluetooth?
Thanks

juha63, 2025-02-21:
@tech1977 
There is no Zigbee FW yet.

tech1977, 2025-02-21:
> [@tech1977](https://github.com/tech1977) There is no Zigbee FW yet.

sorry but this https://pvvx.github.io/TS0601_TZE204/ is not for flashing Custom firmware ZigbeeTLc on Tuya ZigBee ZTH05

juha63, 2025-02-21:
[image]

tech1977, 2025-02-21:
so with my Tuya ZigBee ZTH05:

Zigbee Model TS0601
Zigbee Manufacturer _TZE204_upagmta9

I also make the hardware modification, can't I put ZigbeeTLc firmware? are there any future developments?

tech1977, 2025-02-21:
> Device on stock firmware (Tuya) won't work with these modifications.
> 
> You may try [this firmware](https://github.com/juha63/ZigbeeTLc/blob/zth05/bin/ZTH05Z_v0121_beta.bin) I built month ago. It is working for me but is very experimental. Do it on your own risk and without any support and **ONLY if your device has ATH20/30 sensor**. It is based on V1.2.1 of pvvx's work.
> 
> For me, these devices are not interesting anymore. Too much work to desolder the chips and adding extra connections etc. pp. And you can get better devices on Ali which will work without hardware modifications at the same price.

I understand perfectly that it is not worth it;
do you recommend buying these LYWSD03MMC sensors?
or what?

juha63, 2025-02-21:
If you as me, you should buy some of [these](https://github.com/pvvx/ATC_MiThermometer) that come with BT and won't need soldering. Look [here](https://github.com/pvvx/ZigbeeTLc) if there is a Zigbee version available. Then you can flash it with https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

If you ask Victor, I think he will recommend those from above with sensor SHTxx which should be more precise. And he won't recommend Zigbee. ;-)

I like [this](https://pvvx.github.io/MHO_C401/) personally, but it's more expensive.

tech1977, 2025-02-21:
LYWSD03MMC can you remove the battery and power it directly with 3.3v dc?
I have to take about 10 to put them on the wall where I can have a power supply.
thanks for your help!!

juha63, 2025-02-21:
Yes!

Ctrl-F4, 2025-03-31:
The new ZBdevice firmware  (ZTH05_zb0004.bin) seems to work well. So thanks for supporting this device!

pvvx, 2025-03-31:
> The new ZBdevice firmware (ZTH05_zb0004.bin) seems to work well. So thanks for supporting this device!

This is a "trial" firmware.
Created to be able to test the operation with Zigbee, but having the ability to use BLE and Zigbee OTA for those whose Zigbee coordinator does not allow Zigbee OTA. Which is relevant for closed gateways of different brands.
If everything works correctly with your Zigbee coordinator, then it is advisable to make Zigbee or BLE OTA on a clean Zigbee firmware. It consumes a little less battery ...

1141-0221-01253001-ZTH05.zigbee -> https://github.com/pvvx/ZigbeeTLc/tree/master/bin

Ctrl-F4, 2025-04-01:
Ok, thanks for clarifing this, just flashed ZTH05_v0125.bin via TelinkOTA, working excellent.

alik-ce, 2025-04-28:
Hello, I tested the latest version ZigBee firmware (v0127) in ZTH05 with S8310 sensor and the display shows EE. Is the S8310 sensor supported?

pvvx, 2025-04-29:
[ZTH05](https://pvvx.github.io/TS0601_TZE204/) is supported only with hardware modification.

The firmware includes support for the following sensors:
```
#define USE_SENSOR_CHT8305		1
#define USE_SENSOR_CHT8215		1
#define USE_SENSOR_AHT20_30		1
#define USE_SENSOR_SHT4X		1
#define USE_SENSOR_SHTC3		1
#define USE_SENSOR_SHT30		1
```

alik-ce, 2025-04-29:
Thank you for the information. Hardware modification was done before installing fw, removing ic and two resistors. Debug output is automatically available on the SWS pin?

pvvx, 2025-04-29:
SWS is always available when the chip starts. It is not possible to disable it.

alik-ce, 2025-04-29:
GPIO_PA7 SWS, debug TX - is used for programming, but according to the source code also for debugging. Do I understand correctly?

pvvx, 2025-04-29:
Debugging is disabled in the posted firmware.