pvvx/ZigbeeTLc #148: LYWSD03MMC LCD shows all segments when Display is OFF

mrmaximas, 2025-01-12:
I switch off the display via z2m ui, instead of going down, it shows all segments. 
When the display is switched on, everything works fine.
Zigbee Model LYWSD03MMC-z
Zigbee Manufacturer Xiaomi
Description Temp & RH Monitor Lite (pvxx/ZigbeeTLc)
Support statusSupported
IEEE Address 0xa4c138296a39c852
Network address 0x6B8F / 27535
Firmware build date 20240904
Firmware version 3001-0123
Manufacturer [Xiaomi](https://www.zigbee2mqtt.io/supported-devices/#v=Xiaomi)
Model [LYWSD03MMC-z](https://www.zigbee2mqtt.io/devices/LYWSD03MMC-z.html#xiaomi-lywsd03mmc-z)
[image]

pvvx, 2025-01-12:
Not all versions allow you to turn off the display.
And those require turning the battery on and off.

mrmaximas, 2025-01-12:
> Not all HW versions allow you to turn off the display.

Devbis version turn off display w/o problem.

pvvx, 2025-01-12:
Same as ZigbeeTLc. Just outputs a blank image.

mrmaximas, 2025-01-12:
Devbis version outputs a blank image, this version outputs all segments. How to fix it?

pvvx, 2025-01-12:
For what purpose is it necessary to correct the indication that on this version of HW the display chip does not turn off?

mrmaximas, 2025-01-12:
I want the display to show nothing. 
Am I right in understanding that this will reduce the power consumption a bit?

mrmaximas, 2025-01-12:
Ok, thanks!

closed 2025-01-12 by mrmaximas