pvvx/ZigbeeTLc #151: Revert back to Bluetooth / BLE using ZHA

ezcGman, 2025-01-22:
Hey there,

I'm trying to roll back a device to BLE since an hour and can't get it to work. I found multiple issues here but they're all outdated or talk about how to do it via Zigbee2MQTT. I also found this one, but it's outdated: https://github.com/pvvx/ZigbeeTLc/issues/7

The config in HA has changed in 2024.2.0 (https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates) and now needs to look like this: https://github.com/zigpy/zigpy/wiki/OTA-Configuration

I can't however figure out which provider I need to use or how to just flash a single file... I tried creating an index.json file, but that also didn't work.

This is my config and my index.json file:
```
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: zigpy_local
          index_file: /config/zigpy_ota/index.json
```
```
{
    "firmwares": [
        {
            "fileSize": 83154,
            "manufacturerCode": 4417,
            "imageType": 522,
            "sha512": "77bb1700e7871f01170492944964f5c00eb52191c41ea366ae54da563e805ac4295086cb723815412de91003699e07e65f8c738aef6ce2370e482079a6ae3fae",
            "path": "/config/zigpy_ota/db15-0203-99993001-ATC_v49.zigbee",
        }
    ]
}
```
Took this file as a reference: https://github.com/zigpy/zigpy/blob/0f42cab8f7058cc68044678e4e6e46e59d1747b9/tests/ota/files/local_index.json#L4

But this does nothing, not even a log line in `homeassistant.log`. I also tried issuing this `image_notify` command, but also did nothing...

Any hints on how to flash that firmware and go back to BLE would be much appreciated!

Greetings,

Andy!

If somebody has a nice

kuerb, 2025-02-08:
All the different info is quite confusing, but I just reverted back to BLE by linking the JSON directly.
```
zha:
  zigpy_config:
    ota:
      enabled: true
      # Download the index over HTTPS
      z2m_remote_index: https://raw.githubusercontent.com/pvvx/ATC_MiThermometer/refs/heads/master/zigbee_ota/index_v50.json
```

pvvx, 2025-02-11:
These problems are related to the unfriendly Zigbee OTA interface in ZHA or Z2M. I can't influence it.

closed 2025-02-11 by pvvx

hapklaar, 2025-05-14:
> All the different info is quite confusing, but I just reverted back to BLE by linking the JSON directly.
>
> ```
> zha:
>   zigpy_config:
>     ota:
>       enabled: true
>       # Download the index over [...]

But this is for Z2M instead of ZHA right?

dashernln, 2025-05-22:
> I flashed two of my thermometers to zigbee, but only one is detected as having a fw update available to flash back to BLE, not sure why that is...

doesn`t i miss something? placed configuration in configuration.yaml, restarted HA and wait, while there is something blinks in zigbee (tiomeouts for fw update checks or something else)?

pvvx, 2025-05-22:
The thermometer makes a request for an update every 15 minutes.
The availability of updates described in the "yaml" settings is checked by ZHA only after Home Assistant is restarted and not immediately.