pvvx/ZigbeeTLc #153: Way to adjust IEEE?

ss89, 2025-01-25:
I have had issues with these modules from the very beginning: https://pvvx.github.io/TS0201_TZ3000/

First I only ordered one and it was my first zigbee device, so lots of learnings on my side had to be done.
Ofcourse i softbricked it with the very first update, even though i tried to prepare as good as I could.
I did then unbrick it by using the com flasher (https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) with 921600 baud and this firmware: https://pvvx.github.io/TS0201_TZ3000/ts0201_orig_ff.zip
Everything went back to "reset-to-factory" and also i was a bit surprised that the IEEE seemed to have changed (but wasn't entirely sure that time).
I didn't bother, as that wasn't an issue for me then.

Now that I bought two more of these (sorry for not seeing the "do not buy" message on your page) I did have troubles with the original firmware that was on and again decided to migrate them to your firmware.
This time my zigbee coordinator freaked out and crashed during update (lucky me?).
So I again unbricked those (2!) and the first one went fine without any IEEE change.
The second one however did change its IEEE to one that was already in the network (the one of my first module).

**TLDR**
Can I somehow adjust the IEEE of my existing device that is on your firmware, so that ZHA can properly manage the factory reset device and then upgrade it to your firmware?

pvvx, 2025-01-25:
If you wrote the original FullFlash.bin to different devices, then IEEE will be the same for them. IEEE is stored in Flash. It can be erased during a full Flash clearing - then a new IEEE will be assigned at startup based on a random number, except for the first 3 bytes. The first 3 bytes indicate the chip manufacturer number.

ss89, 2025-01-25:
> If you wrote the original FullFlash.bin to different devices, then IEEE will be the same for them. IEEE is stored in Flash. It can be erased during a full Flash clearing - then a new IEEE will be assigned at startup based on a random number, except for the first 3 bytes. The first 3 bytes indicate the chip manufacturer number.
> 
> The Zigbee 3.0 standard does not provide IEEE change functions. It is possible to change the MAC of the device, also known as IEEE, in the BLE firmware.

I did now do this:
[image]

It seems the device is completely empty now. If i get you right, then flashing the FullFlash.bin again will restore it but with the same IEEE.
What's the process to only erase the part, so that after reboot it would re-generate the IEEE?

pvvx, 2025-01-25:
> If i get you right, then flashing the FullFlash.bin again will restore it but with the same IEEE.

Yes.
Erase and Flash: ZTS0201Z3000_v0124.bin

ss89, 2025-01-25:
Thanks, that has worked. Very nice!

closed 2025-01-25 by ss89