pvvx/ZigbeeTLc #154: Where does the genPollCtrl binding come from ?

bernard-dandrea, 2025-01-26:
[image]

I have looked in zigbeetlc.js and cannot find bindings only for 'msTemperatureMeasurement', 'msRelativeHumidity', 'genPowerCfg'

[image]

with the devbis firmware, there is no binding on genPollCrl

I have the feeling that the TABLE FULL error when you want to update the existing reporting is directly linked to the binding of genPollCtrl 

when you manage to remove that binding (or with devbis version), you have no issue to update the reporting or add ones

thanks for your answer

pvvx, 2025-01-26:
Judging by the name "FULL TABLE" your software makes the device create more than 8 bindings (reports). I don't understand what the point of this is if only four are used.

pvvx, 2025-01-26:
> binding on genPollCrl

Why does your program "binding on genPollCrl" at all?

bernard-dandrea, 2025-01-26:
Many thanks for your fast answer 

Just want to clarify that I don't want to add any reporting but modify the existing one. After a clean installation of the device with telink utility and integration in Zigbee2mqtt, it is not possible to update the reporting of temperature due the TABLE FILL error.

I point out that disabling genPollCrl in binding make the updates in reporting work so I suspect that corresponding entries full the binding table

So, I was wondering what has generated the genPollCrl entries in reporting as I cannot find anything in z2m configuration file zigbeetlc.js (I notice that there was a binding in [zigbeetlc_v0121.js](https://github.com/pvvx/ZigbeeTLc/blob/master/z2m/zigbeetlc_v0121.js but I guess that file is not be used anymore).

I don't understand your explanations concerning Poll_Control story as I am not aware of the initial problem with ZHA / Z2M and what needed to be done to cope with it

pvvx, 2025-01-26:
> I don't understand your explanations concerning Poll_Control story 

https://github.com/pvvx/ZigbeeTLc/issues/128#issuecomment-2612901784

robvanoostenrijk, 2025-01-28:
@bernard-dandrea Remove any custom extensions you have installed for ZigbeeTLc, Zigbee2MQTT supports these devices natively. The custom extensions were never updated, only the native Zigbee2MQTT support was.

After restarting Zigbee2MQTT without the custom extension, remove/unpair your ZigbeeTLc device and repair it.
This should resolve the TABLE_FULL errors you are seeing.

@pvvx Maybe remove the custom extension code from the repository, as support there in Z2M out of the box.

pvvx, 2025-01-28:
> @pvvx Maybe remove the custom extension code from the repository, as support there in Z2M out of the box.

ok

closed 2025-11-07 by pvvx