pvvx/ZigbeeTLc #155: Request for Support of Xiaomi Linptech Seat Pressure Sensor PS1BB (TLSR8250)

hectork2, 2025-01-27:
Hi,

I hope you are doing well. Based on the review conducted in the following article: [Xiaomi Linptech Seat Pressure Sensor PS1BB Review and ESPHome Setup](https://smarthomescene.com/reviews/xiaomi-linptech-seat-pressure-sensor-ps1bb-review-and-esphome-setup/), it has been identified that the chip used in the Xiaomi Linptech Seat Pressure Sensor PS1BB is of type **TLSR8250**.

I was wondering if there are any plans to add support for this device in the future? It would be great to have this sensor supported.

Thank you for your hard work and contributions to this project.

Best regards,  
Nader

pvvx, 2025-03-04:
I don't have such a sensor.

Psytoshgen, 2025-03-04:
> I don't have such a sensor.

Well, I can test, or help, or something… 

Можно на русском? Датчик не стабильно держит сеть, поэтому готов к реверс инжинирингу. Все равно валяется без дела. Может есть базовый код, где я смогу изменить нужную часть? Или может нужны какие-то данные, чтобы удаленно сделать прошивку?

pvvx, 2025-03-04:
Я даже не знаю какой там сенсор. Типа кнопки?

Psytoshgen, 2025-03-05:
> Я даже не знаю какой там сенсор. Типа кнопки? И что нужно получить в Zigbee?

Да, там лента, в которой по всей длине сенсоры нажатия. В корпусе одна кнопка. Приеду с работы, подробно отфотографирую внутреннюю часть и ленту.

Psytoshgen, 2025-03-05:
Вот пациент, внутри ленты контактные пластины и два провода идущие к контроллеру, через usb-c порт, задействовано всего 2 контакта, поэтому это работает точно как кнопка, адресации в пластинах нет, есть только нажато или нет.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

pvvx, 2025-03-05:
Какая глупость это устройство...
Купите Zigbee кнопку и припаяйте к выводам кнопки ленту.
Это будет самым простым, быстрым и дешевым решением.

Psytoshgen, 2025-03-06:
Справедливо, я бы даже не кнопку а геркон предпочел, они дешевле сейчас, почему-то. Устройство бралось в надежде сделать из него датчик сна. Но как писал выше, стабильность коннекта у него отвратительная. Поскольку исправлений от производителя приходится ждать годами, а иногда ждать вообще не приходится, решил попытать счастья здесь, тем более у вас есть и комбинированные прошивки. Если вариантов перешить нет, выпаяю ленту и переделаю геркон, их у меня есть запас

hectork2, 2025-03-06:
> Какая глупость это устройство... Купите Zigbee кнопку или какой датчик открытия двери/окна и припаяйте к выводам кнопки ленту. Это будет самым простым, быстрым и дешевым решением.
> 
> Пример Zigbee кнопки - https://pvvx.github.io/TS0041_TZ3000_fa9mlvja/ Работает нормально с родной прошивкой. Смысла переписывать FW не имеет.
> 
> И там ещё какой-то чип:
> 
> [image]
> 
> У меня уже нет возможностей поддерживать десятки разных устройств. Для теста каждой новой версии требуется проверка более 60 вариантов прошивок и устройств на все комбинации и тесты со внешним ПО. Это требует уже более месяца бесплатной работы… По этому, поддержку в ZigbeeTLc ограничил только популярными моделями термометров. Другие варианты, если они были нужны мне, иногда тоже выкладываются... Но не более.
> 
> Так же, для беспроводных датчиков в “Умном доме” больше всего подходит BLE, чем Zigbee. Zigbee у меня используется для исполнительных устройств, а не для датчиков. Так как проблем с Zigbee значительно больше и надежность у Zigbee хуже, плюс ограничения в количестве устройств. Самая низкая надежность у WiFi. WiFi устройства вообще не являются "беспроводными". Zigbee - почти аналогично, т.к. требуют роутеров, которые работают от электросети. И при частой передаче показаний Zigbee не могут работать от батареек. Исполнительные устройства тоже работают от внешнего питания - именно по этому им походит Zigbee. И у Zigbee вторая беда - один канал связи и если он шумный, или пересекается с WiFi, то потребление батарейных устройств растет. Плюс, если при отключении внешнего питания отключаются роутеры и координатор, то сеть Zigbee падает - будут стихийно отваливаться некоторые устройства. Т.е. обязательна необходима установка автономных Zigbee роутеров с питанием от больших АКБ. Всего этого лишены BLE устройства. И дальность связи у них в режиме "LE Long Range" больше чем у Zigbee и WiFi. Это по итогам опыта работы с “Умным домом” за период более 4 лет.
> 
> PS: Есть такой закон, проверенный за период моего “творчества“ в народных тематиках за более 40 лет: “Народ всегда выбирает худшее”. Это и относится к Zigbee.

Thank you for your response. I understand that you currently don't have much time to support new devices.
However, I'm still confused about one thing. You said that this device is so simple that it only has an on/off button that can be connected to any other switch. If that's the case, how can the pressure be controlled? Is there a way to adjust the pressure level using this device?

[image]

pvvx, 2025-03-06:
> If that's the case, how can the pressure be controlled? Is there a way to adjust the pressure level using this device?

Based on the appearance of the sensor-tape in the photo above, it is impossible to determine what this sensor is.

pvvx, 2025-03-06:
> По стабильности с вами не соглашусь пожалуй, в моем случае то, что нельзя просто сосканить из беспроводного выигрывает, учитывая что я все же в квартире живу а не в собственном доме, не очень хотелось бы светить открытыми BLE данными вокруг. 

В BLE cуществует шифрация маяков и никто не узнает что там передано.

> Так же количество соединений на BLE приёмник оставляет желать лучшего. Выход конечно есть - провода, но тогда придется ими обмотаться. Поэтому зигби видится более менее хорошим и стабильным вариантом

Количество обслуживаемых BLE маяков ограничено приемом до 300 штук в секунду. Другого ограничения нет. В общем их может быть хоть несколько тысяч. Тогда будут чаше приниматься те, которые ближе.

pvvx, 2025-03-06:
> Устройство бралось в надежде сделать из него датчик сна. 

Для этого ныне есть дешевые модули Радаров для определения присутствия и движения. При хорошей настройке будет показывать биение сердца и ваше местоположение с точностью в десяток см.

pvvx, 2025-03-07:
> Всего два провода от ленты не даст много вариантов развития ситуации.

Он может быть и емкостным.

hectork2, 2025-03-08:
> Какая глупость это устройство... Купите Zigbee кнопку или какой датчик открытия двери/окна и припаяйте к выводам кнопки ленту. Это будет самым простым, быстрым и дешевым решением.
>
> Пример Zigbee [...]

I completely agree with you that Zigbee has its drawbacks, especially regarding stability, battery consumption for low-power devices, and interference issues. However, until a straightforward, comprehensive, and user-friendly solution—similar to Zigbee2MQTT—is available for Bluetooth Mesh, it remains challenging to recommend Bluetooth Mesh for typical home users, primarily due to its limited ecosystem and support.

I fully acknowledge the strong points you mentioned about Bluetooth Mesh, especially its fully distributed architecture and independence from central routers. But as long as Bluetooth Mesh doesn't match Zigbee's ease of setup and integration, I think Zigbee continues to be the better choice for most home applications.

pvvx, 2025-03-08:
> But as long as Bluetooth Mesh doesn't match Zigbee's ease of setup and integration, I think Zigbee continues to be the better choice for most home applications.

But Zigbee requires more resources from any SoC. And the program implementation is more complex

To create firmware for this device, you need to use different Zigbee clusters than for thermometers. This is not suitable for generalizing the code in ZigbeeTLc, as it is easier to implement as a separate project.