pvvx/ZigbeeTLc #158: Devices never finish handshake

Lel85, 2025-02-05:
I had the issue that after the first registration i was unable to bind them to any router apart from the coordinator. One LYWSD03MMC eventually showed as rematched, however never updated. the other, after trying to rebind, was unable to ever complete the handshake.
I tried now to rebind the unresponsive one to reflash it with the bluetooth firmware via the OTA ZHA updater. unfortunately that one is now caught in the same loop of never finishing the handshake.

I attached the Coordinator Logfiles. The Device in question has the following MAC address.
a4:c1:38:fe:00:49:e2:51

[home-assistant_zha_2025-02-05T17-37-51.973Z.log](https://github.com/user-attachments/files/18676464/home-assistant_zha_2025-02-05T17-37-51.973Z.log)

I will now try the UART way of flashing as they are not accessible anymore via Zigbee

pvvx, 2025-02-05:
Did you reset the settings by shorting the contacts marked "reset" and "ground" before you did the binding?

Lel85, 2025-02-05:
Initially? i cant remember. I think so at least with one of the two. i managed to add them both and they have been updating, i could just not move them to connect via a router in in the home network.

than i tried rebinding them by bridging reset and Ground until the display shows "---"
Device is than found and the hand shake starts but the "configuration" step never ends. 
One device still shows up as the LYWSD03MMC by Xiaomi, the other device shows no model, just IEEE and network

frank-f, 2025-02-16:
How close to the coordinator (not a router) was the device during the interview? Some of my devices were giving me trouble too and I had to get as close as one meter, while others paired just fine from five meters away.

Lel85, 2025-02-16:
I had them like a meter next to the coordinator.

There were some evolvements afterwards. Apart from these two thermometers not working. I had a movement sensor which became unresponsive.  impossible to pair it again. new devices? impossible to pair. so it seems the routing database of the coordinator become somehow corrupted.

So i removed everything, deleted ZHA and switched to Z2M. everything works now.

Not sure if it is really due to those devices or not. but maybe to use ZHA with caution with the firmware

closed 2025-11-07 by pvvx