pvvx/ZigbeeTLc #161: Support for new device ZG-227Z

pohlond, 2025-02-09:
Hello, I have successfully added support for the new device ZG-227Z. It is a different variant of the TH03Z sensor. The temperature/humidity sensor is AHT20 and is connected via software I2C. MCU is TLSR8253. 

// GPIO_PA7 - SWS
// GPIO_PB1 - TX
// GPIO_PC2 - LED
// GPIO_PB7 - RX
// GPIO_PD4 - KEY
// GPIO_PC3 - SDA
// GPIO_PC4 - SCL

Patch file:
[ZG_227Z.zip](https://github.com/user-attachments/files/18724731/ZG_227Z.zip)

Images:
[image]
[image]

pvvx, 2025-02-10:
https://pvvx.github.io/ZG-103Z/

pohlond, 2025-02-10:
So the PCB serves as a template for multiple types of devices, nice...

pvvx, 2025-02-10:
1. Original name in z2m/ZHA - TS0601_TZE200_???? ?
2. Original Full-Flash ?
3. Zigbee OTA ID: 1286-0203-10013001 ?

pohlond, 2025-02-10:
1. Original name: TS0601_TZE200_a8sdabtg
2. Tuya bin: [zg-227z_orig.bin.zip](https://github.com/user-attachments/files/18730373/zg-227z_orig.bin.zip)
3. Zigbee OTA ID: 1286-0203-10013001

pvvx, 2025-02-10:
https://pvvx.github.io/ZG-227Z/ ?

pvvx, 2025-02-10:
ZigbeeTlc version has  been added. https://github.com/pvvx/ZigbeeTLc

closed 2025-02-11 by pvvx

NonPIayerCharacter, 2025-02-11:
Same firmware works on ZG-227ZS v1.2
But instead of TLSR8253 it has TLSR8258. It announced itself with the same name.
I tried to port it, but couldn't make the sensor work.
Unfortunately, the backup i took is most likely corrupted somehow, because when i tried to flash it back it didn't work. So i set it aside and forgot about it until now.

Backup: [ff.zip](https://github.com/user-attachments/files/18759619/ff.zip)

[image]

dkbiz83, 2025-04-03:
> > Same firmware works on ZG-227ZS v1.2
> > But instead of TLSR8253 it has TLSR8258.
> 
> It doesn't matter. The main thing is that the port pins match.

Hello Viktor,

I have been looking for TS0201, ZTH01 on Aliexpress to implement your [mod with two external MY18B20 sensors](https://github.com/pvvx/ATC_MiThermometer/issues/571), however, I got ZG-227ZS v1.2 as pictured above. I was trying to find TS0201, ZTH01 again, but it looks like Chinese change the hardware too fast and there are no options left with the same TS0201/ZTH01 hardware. Is there a chance you could help add MY18B20 support to ZG-227ZS v1.2? I would really appreciate that!

SNMetamorph, 2026-01-28:
Recently got these ZG-227Z V3.5 sensors. It still uses TLSR8253 controller and AHT20 temperature sensor. How to check is it supported by this custom firmware? How to flash device with such firmware? What equipment I need for this?

[image]

[image]

[image]

filimonic, 2026-01-29:
> [@filimonic](https://github.com/filimonic) thank you for such detailed guide! I'll try to flash some of my sensors in few days, and as far as I understood I should use [bin/ZG227Z_v0133.bin](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/ZG227Z_v0133.bin) firmware in USBCOMFlashTx. So if this will wrong firmware, should I be able to write another firmware there?

With wires you can write it or any other fw. IDK anything about ZB firmwares if it is OK or not. 

BTW, seems you can OTA ZB device firmware with Z2M\ZHM. No guides here from me, sorry

SNMetamorph, 2026-01-29:
> > [@filimonic](https://github.com/filimonic) I tried, did everything right, but after restarting sensor, original firmware starts. Electrical contact is indeed good, I used soldering. What's interesting is that seems like firmware writing process are starting successfully, LED does not light up as with original firmware, but slightly blinking during data transmission. What could be wrong? Maybe I should do "Erase All Flash" before trying to write custom firmware?
> 
> Try disconnecting\connecting while flashing both SWS and 3V3 then.

Tried to flash in same way, but with different USB-UART, now device does not starts up at all, I'll try to flash original firmware from older versions then.

SNMetamorph, 2026-01-30:
> > By the way, could +5V UART logic level damage TLSR chip? Or this is "choked" enough to not be able to do any damage?
> 
> It depends on several factors. The TSLR825x's internal supply voltage is 1.2V. The onboard flash crystal is rated at 1.8V. When the chip is in sleep mode, the current consumption is only a few µA. For example, if you supply 5V from a UART output, currents up to a few tens of mA are typical. Any pin of the chip, through their built-in power protection diode, charges the capacitors in the supply. The chip then wakes up and connects its DC-DC supply to a charged capacitor of over 4V…
> 
> The CPU operating current is only 3-5 mA. This is less than the output of any signal from the USB-UART. As a result, the chip is powered through a protective diode on pin. If the operating current were higher, the voltage from the USB-UART TX output would drop. But the current is low, and the chip operates at an inflated supply voltage. The consequences have already been described – something burns out there...

Thank you for explanation. Luckily, both of USB-UARTs that I tried were having +3.3V UART logic level. So, it's unlikely that something were damaged, but if it were, it would just drain battery entirely in couple of months?

SNMetamorph, 2026-02-01:
> If you didn't supply more than 3.6V to the device, then what's the question?

I have concerns about fact that during flashing procedure chip was powered through SWS pin (GND connected, VCC not connected; unfortunately, haven't seen proper guide before that), could this lead to damaging TSLR chip even despite fact that TX pin does not hold more than +3.3 Volts?

pvvx, 2026-03-25:
> А если я не вижу этого датчика ZG-227Z в меню OTA 

Может не быть кластера OTA. Тогда только программатором.
Так-же бывают устройства, что кластер OTA есть, но не работает должным образом.
В основном такое бывает в вариантах с чипами на 512 КБ Flash (маркировка TLSR8xxxF512xxx)