pvvx/ZigbeeTLc #163: LYWSD03MMC, FW: 0.1.2.4, Zero Values of Humidity and Temperature

timonych, 2025-02-11:
Hello!

I've upgraded 15th modules of LYWSD03MMC and some of them shows Zero values of Humidity and Temperature. but battery shows correct.
Wrong data persist on device and via zigbee.

[image]

With FW 0.1.2.3 all modules were worked correct.
Have successfully downgraded via ATC firmware and flashed with 0.1.2.3 FW.

I can do come experiments with module if it necessary.

timonych, 2025-02-11:
> What is the HW version of Xiaomi LYWSD03MMC?

Output from ATC:
````
20:50:41: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
```

pvvx, 2025-02-11:
HW B1.4 - ok

[image]

[image]

HW B1.5 - ok

pvvx, 2025-02-11:
> > What is the HW version of Xiaomi LYWSD03MMC?
> 
> Output from ATC Web UI:
> 
> ```
> Status: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
> Vbat: 2964 mV , Temp: 24.16°C, Humi: 44.06%, ID: 13, Switch: Open, Counter: 0, TRG Out: 0, flg: 0x05
> ```

What type of T&H sensor?

timonych, 2025-02-11:
Have checked HW via ZHA:
```
1. HW: 1.5, FW: 0.1.2.3 - Correct Values
2. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
3. HW: 2.0, FW: 0.1.2.4 - Incorrect Values
4. HW: 2.0, FW: 0.1.2.4 - Incorrect Values
5. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
6. HW: 1.6, FW: 0.1.2.4 - Incorrect Values
7. HW: 1.5, FW: 0.1.2.4 - Correct Values
8. HW: 1.5, FW: 0.1.2.4 - Correct Values
9. HW: 1.5, FW: 0.1.2.3 - Correct Values
10. HW: 1.5, FW: 0.1.2.4 - Correct Values
11. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
12. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
13. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
14. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
15. HW: 2.0, FW: 0.1.2.3 - Correct Values (rollback from 0.1.2.4 via proxy FW ATC)
```
But ATC shows HW 1.7 instead of ZHA HW 2.0

> What type of T&H sensor?

Can I fetch this data via remote or should disassemble all parts?

pvvx, 2025-02-11:
Found a bug in the program with the SHT4x sensor.
All firmware has been updated.

pvvx, 2025-02-11:
> But ATC shows HW 1.7 instead of ZHA HW 2.0

https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#chipset-lywsd03mmc-hwb14

|HW | LCD I2C   addr | SHTxxx   I2C addr | Note |
|-- | -- | -- | -- |
|B1.4 | 0x3C | 0x70   (SHTC3) |  |
|B1.5 | UART | 0x70   (SHTC3) |  |
|B1.6 | UART | 0x44   (SHT4x) |  |
|B1.7 | 0x3C | 0x44   (SHT4x) | Test   original string HW |
|B1.9 | 0x3E | 0x44   (SHT4x) |  |
|B2.0 | 0x3C | 0x44   (SHT4x) | Test   original string HW |

Version 1.7 or 2.0 is determined at first run by reading the HW line written in Flash.
Display matrices or controllers are different for all versions, except B1.7 = B2.0.

arnlyamr, 2025-02-11:
> Found a bug in the program with the SHT4x sensor. All firmware has been updated.

[image]
[image]

Hi, I got a similar bug, how can I reflash it or can I throw the sensor in the trash?)

pvvx, 2025-02-11:
Use Zigbee OTA files: https://github.com/pvvx/ZigbeeTLc/tree/master/bin/fix

arnlyamr, 2025-02-11:
> Use Zigbee OTA files: https://github.com/pvvx/ZigbeeTLc/tree/master/bin/fix

Please point me to the link to find out what ATC FW is :) Can I do this on a Mac?

pvvx, 2025-02-11:
> Please point me to the link to find out what ATC FW is :) Can I do this on a Mac?

Use Zigbee OTA with file [1141-020a-12343001-Z03MMC_v0124_fix.zigbee](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/fix/1141-020a-12343001-Z03MMC_v0124_fix.zigbee)

arnlyamr, 2025-02-11:
> Use Zigbee OTA with file [1141-020a-12343001-Z03MMC_v0124_fix.zigbee](https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/fix/1141-020a-12343001-Z03MMC_v0124_fix.zigbee)

I can't figure out how to flash and return to BLE from ZigbeeTLc. In the flasher https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html the sensor is no longer displayed. Do I need to install some software, in addition to the flasher with git?
￼
￼

pvvx, 2025-02-11:
> I can't figure out how to flash and return to BLE from ZigbeeTLc. In the flasher https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html the sensor is no longer displayed. Do I need to install some software, in addition to the flasher with git? ￼ ￼

? Why all this?

Use **Zigbee OTA**

timonych, 2025-02-11:
> Found a bug in the program with the SHT4x sensor. All firmware has been updated.

Install latest https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-020a-01243001-Z03MMC.zigbee
Can confirm, bug was resolved.

[image]

[image]

Big thanks!

jackyaz, 2025-02-11:
I flashed a device with 0.124 before adding to ZHA and cannot get it to pair. Do I have any recovery options?

arnlyamr, 2025-02-11:
> > I can't figure out how to flash and return to BLE from ZigbeeTLc. In the flasher https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html the sensor is no longer displayed. Do I need to install some software, in addition to the flasher with git? ￼ ￼
> 
> ? Why all this?

Because I have Yandex Station Midi as a hub, I did not find the ability to upload the firmware. Thanks for your help.

pvvx, 2025-02-11:
> I flashed a device with 0.124 before adding to ZHA and cannot get it to pair. Do I have any recovery options?

How does this relate to this topic?

jackyaz, 2025-02-11:
The device is stuck like this and unreachable by MQTT

https://private-user-images.githubusercontent.com/61744758/412137819-336b58c3-ed0e-49c6-adb8-b662c656e532.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzMTQyMzIsIm5iZiI6MTczOTMxMzkzMiwicGF0aCI6Ii82MTc0NDc1OC80MTIxMzc4MTktMzM2YjU4YzMtZWQwZS00OWM2LWFkYjgtYjY2MmM2NTZlNTMyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjExVDIyNDUzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY1NWNiMmQ4NjAxY2FmN2JlNDNmNzRlMzIxMTE0Y2MxNzQ3YTdiMDUzMDE5ZTZmY2ZiZGNmYTZlNGI3NDQ2YTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nVtsozyTtbJY1i-8dYDsJCU_gB5sSbtm19Z5Drbsy5s

pvvx, 2025-02-11:
> The device is stuck like this and unreachable by **MQTT**

What does **MQTT** have to do with **ZHA** and **pair**?

> I flashed a device with 0.124 before adding to **ZHA** and **cannot get it to pair**. Do I have any recovery options?

robvanoostenrijk, 2025-02-12:
I can confirm that the updated `01243001` firmware (commit b419d96) fixed the issue of invalid readings on CGDK2.

Ended up doing a downgrade and re-upgrade in Z2M, since you cannot update the same version number.

Downgrade command:
`mosquitto_pub -t 'zigbee2mqtt/bridge/request/device/ota_update/update/downgrade' -m '{ "id": "Device Name" }'`

Nico83500, 2025-02-15:
> I can confirm that the updated `01243001` firmware (commit [b419d96](https://github.com/pvvx/ZigbeeTLc/commit/b419d968a4b50fc94470daf765e87eb6788d0d2b)) fixed the issue of invalid readings on [...]

How do you do to upgrade in Z2M ? I have 0123001 firmware and OTA doesn't find any update. Thank you.

robvanoostenrijk, 2025-02-16:
> How do you do to upgrade in Z2M ? I have 0123001 firmware and OTA doesn't find any update. Thank you.

You will need a custom OTA json:

- Create `OTA/index.json` in zigbee2mqtt home directory
- Contents (for THZ03 and CGDK2, which I use)
```
[
        {
                "url": "OTA/1141-0211-01243001-ZTS0201Z3000.zigbee"
        },
        {
                "url": "OTA/1141-0206-01243001-ZCGDK2.zigbee"
        }
]
```
- Ensure the files `1141-0211-01243001-ZTS0201Z3000.zigbee` and `1141-0206-01243001-ZCGDK2.zigbee` are also in the `OTA` folder.

Zigbee2MQTT will now detect the update available. In case of downgrade, put the older version files in the OTA folder and update the JSON accordingly. Then use the MQTT downgrade command I posted earlier to downgrade. You cannot downgrade via the web interface.

Nico83500, 2025-02-16:
> You will need a custom OTA json:
>
> * Create `OTA/index.json` in zigbee2mqtt home directory
> * Contents (for THZ03 and CGDK2, which I use)
>
> ```
> [
>         {
>                 "url": [...]

Thank you it works !

closed 2025-11-07 by pvvx