pvvx/ZigbeeTLc #164: MHO C401(N) bricked after flashing MHO C401 (old) firmware

michaelkanis, 2025-02-11:
Hi,

I just flashed the v0.1.2.4 firmware to one of two MHO C401 devices and now the screen is frozen and it doesn't show up neither as Bluetooth device nor can my Zigbee coordinator find it.

It seems I may actually have the C401N devices, because on a second device from the same batch I successfully installed the BLE firmware v5.0 for C401N and everything works as expected .

Is there any way I can unbrick the device? Obviously I can't flash it OTA. I guess I can flash it using some USB to Serial device but I can't find any guide on how to do that. Could you please give me a hint?

pvvx, 2025-02-11:
Zigbee probably works, but does not display on the screen.

michaelkanis, 2025-02-11:
Wow thanks for the lightning fast response! What do you mean by "does not display on the screen"? Would there still be a chance to flash back the BLE firmware through Zigbee? I am trying to find the device with Home Assistant (ZHA) using a Conbee II coordinator.

pvvx, 2025-02-11:
> I am trying to find the device with Home Assistant (ZHA) using a Conbee II coordinator.

On the back of the MHO C401, press and hold the button for 10 seconds....

michaelkanis, 2025-02-11:
I see. I already did that (and tried again now) but unfortunately it doesn't work.

mogim0gi, 2025-03-13:
I think i made the same mistake.. @michaelkanis  - did you manage to solve it for yourself ?

mogim0gi, 2025-03-13:
@pvvx  - The display is fully functional, holding button on the back changes Celsius to Farenheit, but its no longer detectable by the web flashing tool. Any tips ? is USB serial flasher the only option ?

pvvx, 2025-03-13:
Web flashing tool can only work with BLE.
No operating system or web browser supports Zigbee.
To work with Zigbee, you need special software and a Zigbee coordinator.

pvvx, 2025-03-13:
Zigbee OTA -> BLE files:
https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota

Jumperm, 2025-06-30:
> Zigbee probably works, but does not display on the screen.
> 
> > I can flash it using some USB to Serial device but I can't find any guide on how to do that. Could you please give me a hint?
> 
> https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version

Hi,
is the “RST pin TLSR825x or +Vbat” pin necessary to connect to the FTDI?
BR