pvvx/ZigbeeTLc #165: MHO-C401(N) - display stuck and display not refreshing properly

alexsperling, 2025-02-15:
Hi @pvvx ,
first of all, thank you for your amazing work!

I bought some thermometers with e-ink display on aliexpress. First i bought 7 of them to use them in my smart home. After flashing them all with the zigbee firmware I realised one of them had issues with refreshing the display properly (right one in the pictures). the other 6 worked fine.

then i bought another one from aliexpress (left one in the pictures) because i thought that the display (of the right sensor) is a faulty one.
after receiving it today i immediately connected it throught your web-flasher-tool (https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)
and flashed the zigbee software. unfortunately the display on that unit is now stuck and wont refresh anymore.

both sensors can connect to my home assistant via z2m and i can read all of the values properly. do you have any clues why the displays wont function properly? any help would be highly appreciated!

thank you in advance!

[image]
[image]

frank-f, 2025-02-16:
You have one C401N and one C401. They require different firmware to drive the displays. The webflasher tool detected all of mine wrong too, trying to flash the C401 firmware to my C401N devices resulted in exactly the same situation you're in - everything works, but the displays don't update.

alexsperling, 2025-02-17:
fixed my problem, thank you @frank-f 
i used this guide to flash back to BLE and than again to zigbee.

closed 2025-02-17 by alexsperling