pvvx/ZigbeeTLc #170: MJWSD06MMC device supported

RykovDC, 2025-03-07:
When will the MJWSD06MMC device be supported?

pvvx, 2025-03-07:
Once you provide the chipset information, I will consider supporting MJWSD06MMC.

RykovDC, 2025-03-08:
Then do not close issues, as I will buy the sensor, I will send you a photo of the board.

pvvx, 2025-03-08:
[image]

RykovDC, 2025-03-13:
А будет кастомная Zigbee прошивка для MJWSD05MMC?

pvvx, 2025-03-27:
https://pvvx.github.io/MJWSD06MMC

RykovDC, 2025-03-27:
То есть чипсет подходит?

pvvx, 2025-03-27:
> То есть чипсет подходит?

Да. Будет время - слеплю что-нибудь...

pvvx, 2025-03-28:
> А будет кастомная Zigbee прошивка для MJWSD05MMC? Кастомную для BLE я увидел, но интересует еще и Zigbee.

Прошивка через OTA пока невозможна. Аналог как у Xiaomi LYWSD03MMC с новыми версиями 2.1.1_xxx оф. прошивок.
Логин с ключами от MIHome проходит, но требуется подпись прошивки (сертификатами).
Соответственно необходимо найти и скачать файл оф. OTA с подписью с сайта/cloud mi и пробовать аналогичный хак, как у LYWSD03MMC... Иначе только путем разбора корпуса и пайки проводков к программатору...

SLboat, 2025-03-28:
[image]
inside pic

pvvx, 2025-03-29:
BLE прошивка для MJWSD06MMC готова...

[image]
Average 14.46 uA (default settings)

chertvl, 2025-03-29:
Виктор, спасибо за Вашу работу. Очень буду ждать ОТА для этой модели. Пара градусников уже ждут что бы быть интегрироваными в НА через штатные компоненты, да что бы батарейка экономнее жралась. :)
Буду следить за веткой.

pvvx, 2025-03-30:
> Battery voltage seems off, 3.3V shows 2.8V,

The voltage supplied to the chip under a load of 3..5 mA is measured. The voltage of a discharged CRxxxx battery does not drop below 3V without load at +25C.
The internal power supply of the chips is connected through a reverse polarity protection transistor, and there is a voltage drop across it.

SLboat, 2025-03-30:
> > Battery voltage seems off, 3.3V shows 2.8V,
> 
> The voltage supplied to the chip under a load of 3..5 mA is measured. The voltage of a discharged CRxxxx battery does not drop below 3V without load at +25C. The internal power supply of the chips is connected through a reverse polarity protection transistor, and there is a voltage drop across it.
> 
> Opening - is the state of the reed switch if it is soldered to the GPIO_PB7 contacts (board label "R") and GND. The function is configured in TelinkMiFlasher.html

Then the voltage is fed through the 65XY LDO chip and stepped down to 2.8V for use together, so maybe that makes sense. The chip uses the stabilized voltage.

[image]

After measuring, the V test point voltage on the battery is 2.810V, which seems to be as expected.
This caused the measured voltage to look unreliable.

pvvx, 2025-03-30:
A bad LCD controller chip is selected. 
The power supply of the LCD controller according to the documentation is up to 2.5 V...
The current consumption during operation is more than 7 μA. When maximum power saving is enabled.
The first versions of Xiaomi LYWSD03MMC have only 3 μA.

[image]
This voltage divider for the battery is an unnecessary energy consumer (to 3 µA). And it is better to break it.

SLboat, 2025-03-30:
> A bad LCD controller chip is selected. The power supply of the LCD controller according to the documentation is up to 2.5 V... The current consumption during operation is more than 7 μA. When maximum power saving is enabled. The first versions of Xiaomi LYWSD03MMC have only 3 μA.
> 
> [image] This voltage divider for the battery is an unnecessary energy consumer (to 3 µA). And it is better to break it.
> 
> The total current in sleep mode of all components is more than 10 µA. (LYWSD03MMC - 5..6 µA)

This looks like a cost-cutting option, and the complexity of the entire case has been reduced, with a lot of snaps and elimination of screws and other elements, they've squeezed the cost down to the bone.

The battery divider circuit seems to be available to measure the correct voltage? You mentioned the VBAT connection to the PC4. looks like we can get the correct voltage:)

pvvx, 2025-03-30:
The capacitors in the power supply are installed with a capacity several times smaller than recommended by the BLE chip manufacturers.
You shouldn't expect the battery in such a hack job from Xiaomi to work for a long time.

SLboat, 2025-03-30:
I agree, I'm afraid this is the only way to go, if we need battery data.
Of course it's possible to swap it out for a bigger resistor, but Xiaomi doesn't seem to care about that or use a flexible gpio to drive on-demand measurements.
Most people probably won't be able to break this resistor (lack of tools) and the ultimate compromise may be that it won't change the basic facts. Xiaomi doesn't seem to care about 3uA and it uses a larger battery to hide this.

pvvx, 2025-03-30:
[image]

I'm not sure it's an LDO. It's probably just a MOSFET (?)

SLboat, 2025-03-30:
> [image]
> 
> I'm not sure it's an LDO. It's probably just a cheapest MOSFET (?)
> LDO with low intrinsic currents are expensive...
> 
> 
> > Of course it's possible to swap it out for a bigger resistor,
> 
> Removing one resistor going to GND and reducing the second completely solves the problem.
> But Xiaomi wants to drain batteries. They probably have an agreement with battery manufacturers :)
> 

Yes it's a ldo, should be 2.8V, probably made in China xc6206[image]

In the official publicity, it offers two modes to support the battery, the standard mode and the long-life mode, with a life of 2 and 3 years.

pvvx, 2025-03-30:
> In the official publicity, it offers two modes to support the battery, the standard mode and the long-life mode, with a life of 2 and 3 years.

It's just an advertisement. It has nothing to do with reality.

SLboat, 2025-03-30:
Yes, the official firmware update seems to be very lazy and the long life mode slows everything down.

Both Ldo and resistor seem to sacrifice extra current and it's hard to believe that the same design can achieve 2 years of battery life (official firmware).

There also seems to be a new mechanism where the official firmware saves an offline data every hour.

[image]

[image]

pvvx, 2025-03-30:
What's the point of all this if:
1. Set the beacon interval to 5 seconds, and measurements to 20 seconds.
2. Throw away unnecessary components
3. Solder a capacitor

It turns out to be up to 10 μA of average current consumption.

And if you calculate the battery (550 mAh), as Xiaomi does, you get:

550/0.01=55000 h
55000/24/365=6.28 years!

SLboat, 2025-03-30:
> What's the point of all this if:
> 
> 1. Set the beacon interval to 5 seconds, and measurements to 20 seconds (= default settings).
> 2. Throw away unnecessary components
> 3. Solder a capacitor
> 
> It turns out to be up to 10 μA of average current consumption.
> 
> And if you calculate the battery (550 mAh), as Xiaomi does, you get:
> 
> 550/0.01=55000 h 55000/24/365=6.28 years!
> 
> Xiaomi thinks so: 550/[0.027](https://github.com/pvvx/ZigbeeTLc/issues/170#issuecomment-2758586704)/24/365=2.3 years
> 
> [In reality](https://github.com/user-attachments/files/18102754/swra349.pdf), given Xiaomi's botched design, only up to 40% of the battery will be available -> Total: 1 year

2 years seems like a reasonable goal for Xiaomi, maybe they aren't interested in inventing devices that will last 6 years :)

pvvx, 2025-03-31:
> 2 years seems like a reasonable goal for Xiaomi, maybe they aren't interested in inventing devices that will last 6 years :)

Fake 2 years?
Where will 2 years come from with a pulse current of 15 mA?

SLboat, 2025-03-31:
> > 2 years seems like a reasonable goal for Xiaomi, maybe they aren't interested in inventing devices that will last 6 years :)
> 
> Fake 2 years? Where will 2 years come from with a pulse current of 15 mA?

> > 2 years seems like a reasonable goal for Xiaomi, maybe they aren't interested in inventing devices that will last 6 years :)
> 
> Fake 2 years? Where will 2 years come from with a pulse current of 15 mA?
> 
> According to the measurements of the original version, it turns out that the battery will work until its internal resistance increases to 54 Ohms. Next, a threshold is reached in which the voltage drops below the operating voltage for the chip. According to all calculations, without taking into account the battery self-discharge and other factors, this will amount to 1..1.5 years as a maximum.

[image]
[image]

- Turn on the long range mode
- ble Connect to the gateway

Flat 19uA (general environment), 550mah, probably very barely close to 3.3 years, which is probably what Xiaomi is basing it on.

pvvx, 2025-03-31:
Alternatively, if the circuit and firmware are changed, the limit is about 3 years.

SLboat, 2025-03-31:
When turning off the battery long range mode, the leveling current drops instead to only 16ua (long range leveling 18~19uA), but that all seems to depend on the ambient temperature and humidity not changing much.

But ble is a really small draw compared to zigbee.

[image]

SLboat, 2025-03-31:
[image]
Sleep current 11ua :)
Like you said, there is some waste.

pvvx, 2025-03-31:
The pulse load for the battery is not calculated this way.

SLboat, 2025-03-31:
> The pulse load for the battery is not calculated this way.
> 
> Окончание срока службы батареи наступает, когда при импульсном токе (RF-TX) 15 мА напряжение батареи достигает минимального рабочего напряжения для микросхем. The battery reaches the end of its service life when, with a pulse current (RF-TX) of 15 mA, the battery voltage reaches the minimum operating voltage for the chips.
> 
> In this case, this will happen after 40% of the battery capacity has been consumed.

When trying to start the thermometer with a 2.864V CR2032 coin cell battery, the low battery icon appears and the entire thermometer does not work.
[image]

pvvx, 2025-03-31:
Practice and all user reports for 4 years indicate that CR2xxx batteries give only up to 40% at pulse currents up to 10 mA. This is provided that the circuit does not have a pulse-smoothing capacitor of tens of microfarads.

SLboat, 2025-03-31:
[image]

This is the MJ6_5.1 firmware released by His Excellency two days ago, and the measured leveling current is 13ua, and the startup time with ble icon is 26uA.
This seems to give an expected time of more than 4 years (according to Xiaomi's algorithm), but based on your actual experience, 3 years might be conservatively able to be carried out, counting discounts.

The main codebase doesn't seem to have the source code for this thermometer yet, only the bin file, hope to see the new source code soon :)

Even without circuit modifications, this is a very long-lived thermometer, which has a lot of potential. In the Chinese market it sells for 34 yuan (roughly $5), a bit cheaper than the big three, 3 is probably a bad design, too big, and the time and date functions are silly.

pvvx, 2025-03-31:
In the alternative firmware, the parameters are set to default and are set to get 3 years of operation.

screekworkshop, 2025-03-31:
Thank you very much, the only thing less than ideal at the moment is the abnormal battery voltage, 3 years is enough to forget about the battery thing, which is wonderful.

pvvx, 2025-03-31:
> at the moment is the abnormal battery voltage

But these are real readings, and not what is somewhere before LDO.

screekworkshop, 2025-03-31:
> > 2 years seems like a reasonable goal for Xiaomi, maybe they aren't interested in inventing devices that will last 6 years :)
> 
> Fake 2 years? Where will 2 years come from with a pulse current of 15 mA?
> 
> According to the measurements of the original version, it turns out that the battery will work until its internal resistance increases to 54 Ohms. Next, a threshold is reached in which the voltage drops below the operating voltage for the chip. According to all calculations, without taking into account the battery self-discharge and other factors, this will amount to 1..1.5 years as a maximum.
> 
> [@SLboat](https://github.com/SLboat) - Haven't you read https://github.com/user-attachments/files/18102754/swra349.pdf ? All chip manufacturers have such publications. And all have a recommended formula for calculating the capacity of the capacitor required for installation in the chip power supply. But Xiaomi does not know this? Based on this factor alone, the battery life of the MJWSD06MMC will be up to 1.5 years.

Real world coin cell powered, and yes, this document seems to be saying that the reality of the situation is much harsher than the ideal.

An interesting find, powered by an old 2.86v coin cell battery:
- The official firmware shows a low battery icon (seems to detect voltage at startup) and doesn't work.
- The current pvvx firmware (5.1, day before yesterday), will boot up fine, the bluetooth icon is always displayed (not blinking), and no ble broadcast can be made. The screen is normal, but transmitting is impotent.

Xiaomi seems to have done something useful back to the voltage that can't transmit, and it still measures and displays in that sense, but it's no longer possible to transmit ble.

pvvx, 2025-03-31:
This is not true.
The thermometer simply went into deep sleep mode for a few minutes.
It did not have enough energy to refresh the screen.
When the voltage rises above 2.2 V, it will work, unlike Xiaomi.
The battery voltage drops at low negative temperatures. When the battery heats up again, everything will work.
The same applies when powered by a solar panel.
Xiaomi firmware will not allow the supercapacitor to be charged when powered by a solar panel.

pvvx, 2025-03-31:
New MJ6_v5.1.bin - LCD controller shutdown was forgotten.
Added: deep sleep for 3 minutes instead of 2.

pvvx, 2025-03-31:
Volt-ampere characteristic of a power supply circuit with LDO.
Firmware with transition to deep-sleep at low voltage.
The internal resistance of the power supply is 60 Ohm.
Measured with filter up to 1.6 kHz (therefore TX peaks are smaller).

[image]

pvvx, 2025-03-31:
Minimum voltage at which the chip detects low voltage (measurement step is set to 20 sec) and switches off.

[image]

Voltage drop with a capacitor is only 0.05V for a source with an internal resistance of 60 Ohm (discharged battery)

pvvx, 2025-03-31:
@SLboat - Due to the specifics of the auto-range switching circuit, you will not be able to correctly measure currents and other characteristics using the "PowerProfiler kit II" (NRF-PPK2).

pvvx, 2025-03-31:
Consumption current after disconnection, after registration "Low voltage":
[image]
Ldo disabled = jumper

SLboat, 2025-04-01:
> New MJ6_v5.1.bin - LCD controller shutdown was forgotten. Added: deep sleep for 3 minutes instead of 2.
> 
> The current code has not yet been rewritten to the new version, but the version for MJWSD06MMC has been somehow inserted and assembled - that's why it is not published yet Wait for version 5.2. Then it will make sense to check everything thoroughly...

It's hard to imagine the ongoing research you've done on them, they're very relevant, and of course that work has given this temperature sensor another popularity.

For 2.2v a critical setting is cool, it really is a pain in the ass when the voltage is very low, it may be depleted without a chance that it can get a charge.

Can see the code has a lot of patches being introduced by the way, I tested the new 5.1 firmware and attributes like battery and opening are not broadcasting but the battery percentage still seems to be there.

Looking at the official pcb, the voltage seems to go through a divider resistor into the test point corresponding to your marked vbat before going into the ldo, being able to use it for voltage detection seems interesting and necessary, hopefully we'll see changes like this in the future.

The MJ6 looks like it has a lot of potential, the SHTV4 for the SHTV3 and the CR2045 for the CR2032, and an extra button, that button might also make an event linkage for those who need it, which could be quite interesting. It would also be useful for the zigbeetlc firmware to make a reset setting.

pvvx, 2025-04-01:
> I don't understand the meaning of the sleep mechanism in the firmware design though, like entering after 2 minutes, what is that 2 minutes used for? Is it used to tell the user that it is currently ok?

At low voltage (<2.0 V), a transition to a deep sleep mode for 3 minutes occurs. After 3 minutes, the chip is awakened and checks the voltage for work (> 2.2 V). If the voltage exceeds 2.2 V, regular work begins. If not 2.2 V, it falls asleep again for 3 minutes.

SLboat, 2025-04-01:
Injecting 3V directly from the test point, the lcd display has high contrast residuals, the ldo seems to be in service of the display driver, perhaps for the right voltage, other than that reason the design is really puzzling.
 But at the same time it seems to be possible to input 5V directly from the battery connector, after the ldo it's bucked. Of course this may be pointless.

SLboat, 2025-04-01:
> > I don't understand the meaning of the sleep mechanism in the firmware design though, like entering after 2 minutes, what is that 2 minutes used for? Is it used to tell the user that it is currently ok?
> 
> At low voltage (<2.0 V), a transition to a deep sleep mode for 3 minutes occurs. After 3 minutes, the chip is awakened and checks the voltage for work (> 2.2 V). If the voltage exceeds 2.2 V, regular work begins. If not 2.2 V, it falls asleep again for 3 minutes.
> 
> When the regime of deep sleep is active, everything turns off to achieve minimal consumption. In order not to discharge the battery if the voltage drops due to low temperature. The LCD -display controller also switches to sleep mode, because it consumes more than other components.

Mode for supercapacitors, I see, for solar this is useful. It is possible to save on energy recovery units, which are costly. This means that it is also suitable for solar drives.

chertvl, 2025-04-01:
> and an extra button, that button might also make an event linkage for those who need it

@SLboat are you think its possible? i need it! ;O

pvvx, 2025-04-01:
> are you think its possible? i need it! 

Then, if you set a long BLE advertising interval, it will be difficult to connect to the device.

chertvl, 2025-04-02:
> > are you think its possible? i need it!
> 
> Then, if you set a long BLE advertising interval, it will be difficult to connect to the device.

Thats ok if its still possible. Connect and configure need just once :)

pvvx, 2025-04-02:
> On this little k2s device(with button), the manufacturer has designed a strange button pattern to be used as a keypad and to perform some other functions at the same time.

But there are many reports that users cannot reset the original firmware. They do not understand how to press the buttons :)
And there are thousands of such messages.

Are you suggesting to make a telegraph key from a button with the transmission of symbols in Morse code?

chertvl, 2025-04-04:
> For more advanced users, there is the option to solder a reed switch or a second button. It will transmit the Open/Close and/or counter status. Settings in TelinkMiFlasher.

As an advanced soldering iron user, I will try to throw a jumper on the dedicated pin for the opening sensor from the case button, since it is beautiful and convenient. And if something happens, I will switch to connection mode without the button. :) I did not want to do exactly this, I thought maybe there would be a software solution, but in general it will work out this way.

chertvl, 2025-04-04:
> throw a jumper on the dedicated pin for the opening sensor from the case button

Something like...

[image]

chertvl, 2025-04-06:
@pvvx 
Добрый день, Виктор. Уже пару дней пытаюсь понять, но не получается... Постараюсь быть кратким: термометр с MJ6_v5.1.bin не хочет корректно работать через btproxy с BTHome. 
В доме нюк со встроенным блютузом, и две esp32 с btproxy (active: false). Как только я отношу термометр в другую комнату, и в интерфейсе НА (во вкладке мониторинга рекламных данных) термометр закрепляется за внешним бт-приемником (любая из двух есп32) - пакеты попросту перестают доходить, сенсоры замирают на текущих значениях температуры/влажности/батарейки/rssi, хоть и не уходят в unavailable. Что бы сенсоры в компоненте обновились (единожды) - помогает нажатие перезапуска энтри этого градусника. Подскажите, пожалуйста, чего ему не хватает? :)  Чекбокс AdFlags ключен в настройках устройства.

pvvx, 2025-04-06:
> термометр с MJ6_v5.1.bin не хочет корректно работать через btproxy с BTHome.

Это необходимо спрашивать у тех, кто писал программы в "esp32 с btproxy".
Одновременна работа WiFi и BLE у ESP32 на одну антенну никуда не годится - там выходит до 90% пропусков приема BLE маяков. И до 100% потери когда происходит синхронизация шага передачи BLE маяка с Beacon WiFi (шаг 102.4 мс).
И тем более вы используете устаревший вариант ESP32. У ESP32 очень плохой уровень приема и BT с устаревшей версией 4.2.
У новых ESP32-C3 и т.д. c BLE всё значительно лучше.

chertvl, 2025-04-06:
> Спросите, какие форматы BLE рекламы поддерживает установленная программа на вашем ESP32.

Благодарю за развернутый ответ.
В целом, я знал, что прошивка никогда не гарантировала работу с btproxy проекта esphome, но была надежда, что все заведется :) Мне нравится такой принцип работы, пара десятков устройств годами работают без единого пропуска, правда у всех у них тип пакетов mi (весы, мифлоры, зубные щетки, елайт-диммеры, и конечно же полдюжины lywsd03mmc). Все оно у меня в XiaomiBLE всунуто.

В таком случае, попробую уточнить больше информации непосредственно у авторов btproxy,
в противном случае подожду поддержку mjwsd06mmc в XiaomiBLE от Эрнста, или поищу другие способы удаленного мониторинга термометра, который не добивает до мини-пк :)

chertvl, 2025-04-06:
> Уже работает в [custom-components/ble_monitor](https://github.com/custom-components/ble_monitor/issues/1449).

Да, видел, еще до того как Вы получили этот градусник - я уже искал способы :)
К сожалению, метод агрегации данных в BLE Monitor'е мне не симпатизирует, поэтому в свое время я от него отказался.

> **MJ6** тоже переключается на формат Mijia (Xiaomi)...

Само собой. Но пока еще не видны в Xiaomi BLE ;)

________________
Большое обновление
> термометр с MJ6_v5.1.bin не хочет корректно работать через btproxy с BTHome.

Я пересобрал свою esp32  с arduino на esp-idf, так как в этом фреймворке btproxy значительно меньше потребляет памяти.
А так же указал active: true.  И, к моему удивлению, это помогло.
Рядом стоящее с MJ6 устройство на esp32mini подцепило к себе термометр, и успешно его слушает уже несколько часов.
Кажется, это полностью решило мою проблему. 

[image]

pvvx, 2025-04-06:
Интеграция XIomi в HA практически не обновляется... И не знает новых ID влажности и температуры, примененных в оригинальной прошивке MJWSD06MMC. Аналогично и все другие не знали, пока мне пришли MJWSD06MMC и я не написал Эрнсту...

chertvl, 2025-04-06:
> Интеграция XIomi в HA практически не обновляется...

Не страшно. Я не против остаться на интеграции BTHome, к тому же они обе работают практически одинаково, и умеют в btproxy. Пусть все устройства будут в XiaomiBLE, а пара MJ6 в BTHome, одним компонентом больше одним меньше... Плюсом будет то, что с помощью перемычки я все же смог использовать корпусную кнопку на устройстве, чего нельзя было бы сделать в Xiaomi BLE. 
Так что теперь я полностью доволен продуктом :)

chertvl, 2025-04-06:
> В issue для интеграции "Xiaomi Gateway 3" [кто-то дал неверные данные](https://github.com/AlexxIT/XiaomiGateway3/issues/1572), но Gateway передает значения в MQTT уже с другими, перекодированными ID 1001, 1002, 1003, а не с теми что передает сам MJWSD06MMC.

Передам Алексею, предложу помощь.

pvvx, 2025-04-06:
# mibeacon5 spec
0x4801 = 18433 (Dec) - "temperature" float
0x4802 = 18434 (Dec) - "humidity" uint8
0x4803 = 18435 (Dec) - "battery" uint8

[1 more reply]

closed 2025-11-07 by pvvx