pvvx/ZigbeeTLc #173: Can't flash LYWSD03MMC

benbender, 2025-03-23:
I'm trying to flash a Zigbee-FW onto a `LYWSD03MMC`. I've flashed `ATC_v51.bin` via `TelinkMiFlasher` successfully as per Readme because the latest FWs are larger than 128kb.

After that, I've tried to flash `Z03MMC_zb0004.bin` but the the flasher hangs at `Clear Ext.OTA region`. The display shows three dashes and the bt-symbol. So it seems to be in OTA-mode. I waited for ~10 mins without any change. I'm getting the exact same result while trying to flash `Z03MMC_v0125.bin`. Am I missing something or did I encounter a bug?

````
18:41:13: Reconnect to: ATC_43CE91
18:41:16: Hardware Revision String: B1.6
18:41:16: Software Revision String: V5.1
18:41:16: Firmware Revision String: github.com/pvvx
18:41:16: Detected custom Firmware
18:41:16: Error on sending setting 55
18:41:16: Hardware Version: LYWSD03MMC B1.6, Software Version: 5.1
18:41:16: Custom config HEX string: 5587100a062804a9313104b4
18:41:28: Settings 01 was sent successfully
18:41:28: DevName: [ATC_43CE91]
18:41:31: Settings 10 was sent successfully
18:41:31: MAC: A4C13843CE91, RandMAC: C08D6643CE91
18:41:32: Get binkey from EEP...
18:41:32: Read bindkey: 06a91f9c6f7aa7f33fe74fca79dd2131
18:41:40: File: Z03MMC_zb0004.bin
18:41:40: File size: 165588 bytes
18:41:40: Count: 10350
18:41:43: File: Z03MMC_v0125.bin
18:41:43: File size: 131124 bytes
18:41:43: Count: 8196
18:41:45: Start DFU
18:41:46: Clear Ext.OTA region (129kib)...
````

pvvx, 2025-03-23:
Clearing a large amount of Flash during a BLE connection may cause instability. The procedure must be repeated. The next attempt will result in cleaning a smaller volume. Since part of it will already be cleaned.
Failures depend on the discharged battery and the quality of the connection with the external BT adapter.

benbender, 2025-03-23:
So this is (kind of) expected behavior? Therefor the correct way forward is to just wait? Or will this process "timeout" in the background and needs to be repeated?

pvvx, 2025-03-23:
The process of erasing the required area of ​​the Flash takes a couple of seconds.
Most likely, your battery is dead.

benbender, 2025-03-23:
So, changed the battery. Device sits ~5cm from my Macbook. Same result…

It also seems like the size of the region to be deleted won't change at all between retries.

In the Chromium Console-Log I noticed this – maybe a hint?

````
TelinkMiFlasher.html:1926 Status: Hardware Version: LYWSD03MMC B1.6, Software Version: 5.0
TelinkMiFlasher.html:1930 Hardware Version: LYWSD03MMC B1.6, Software Version: 5.0
TelinkMiFlasher.html:1930 Custom config HEX string: 5587100a062804a9313104b4
TelinkMiFlasher.html:1926 Send cmd (3303): Query 3 measurements
TelinkMiFlasher.html:1926 Send cmd ok
TelinkMiFlasher.html:1926 Custom (0x1f1f) Notifications id: 0x33 [03]
TelinkMiFlasher.html:1926 Vbat: 2754 mV , Temp: 23.21°C, Humi: 38.22%, ID: 6, Switch: Open, Counter: 0, TRG Out: 0, flg: 0x05
TelinkMiFlasher.html:1930 Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:1926 Status: Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:1 [Violation] 'click' handler took 1404ms
TelinkMiFlasher.html:1926 OTA region: 0x000000, size: 128 kbytes -> Ext.OTA busy
TelinkMiFlasher.html:1926 OTA region: 0x000000, size: 128 kbytes -> Ext.OTA busy
````

pvvx, 2025-03-23:
Vbat: 2754 mV - this indicates that the battery is discharged.

pvvx, 2025-03-23:
Checked it again for you. Full filled with Flash history entry and BZ OTA loading in explorer log:
```
...
Vbat: 2258 mV , Temp: 18.53°C, Humi: 44.46%, ID: 44806, Switch: Open, Counter: 0, TRG Out: 1, flg: 0x0F
Clear Ext.OTA region (162kib)...
Status: Clear Ext.OTA region (162kib)...
OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
OTA region: 0x041000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x041000...
Erase the Flash sector at: 0x041000...
OTA region: 0x042000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x042000...
Erase the Flash sector at: 0x042000...
OTA region: 0x043000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x043000...
Erase the Flash sector at: 0x043000...
OTA region: 0x044000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x044000...
Erase the Flash sector at: 0x044000...
OTA region: 0x045000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x045000...
Erase the Flash sector at: 0x045000...
OTA region: 0x046000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x046000...
Erase the Flash sector at: 0x046000...
OTA region: 0x047000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x047000...
Erase the Flash sector at: 0x047000...
OTA region: 0x048000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x048000...
Erase the Flash sector at: 0x048000...
OTA region: 0x049000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x049000...
Erase the Flash sector at: 0x049000...
OTA region: 0x04A000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04A000...
Erase the Flash sector at: 0x04A000...
OTA region: 0x04B000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04B000...
Erase the Flash sector at: 0x04B000...
OTA region: 0x04C000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04C000...
Erase the Flash sector at: 0x04C000...
OTA region: 0x04D000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04D000...
Erase the Flash sector at: 0x04D000...
OTA region: 0x04E000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04E000...
Erase the Flash sector at: 0x04E000...
OTA region: 0x04F000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x04F000...
Erase the Flash sector at: 0x04F000...
OTA region: 0x050000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x050000...
Erase the Flash sector at: 0x050000...
OTA region: 0x051000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x051000...
Erase the Flash sector at: 0x051000...
OTA region: 0x052000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x052000...
Erase the Flash sector at: 0x052000...
OTA region: 0x053000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x053000...
Erase the Flash sector at: 0x053000...
OTA region: 0x054000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x054000...
Erase the Flash sector at: 0x054000...
OTA region: 0x055000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x055000...
[... 56 lines cut ...]
Status: Erase the Flash sector at: 0x068000...
Erase the Flash sector at: 0x068000...
OTA region: 0x069000, size: 0 kbytes -> Ext.OTA event
Status: Erase the Flash sector at: 0x069000...
Erase the Flash sector at: 0x069000...
OTA region: 0x040000, size: 164 kbytes -> Ext.OTA ready
Status: Clean Flash OTA region: 0x040000, size: 164 kbytes. Go OTA...
OTA status after block 7 is  0
OTA status after block 15 is  0
OTA status after block 23 is  0
OTA status after block 31 is  0
OTA status after block 39 is  0
OTA status after block 47 is  0
OTA status after block 55 is  0
OTA status after block 63 is  0
OTA status after block 71 is  0
OTA status after block 79 is  0
OTA status after block 87 is  0
OTA status after block 95 is  0
OTA status after block 103 is  0
OTA status after block 111 is  0
OTA status after block 119 is  0
TA status after block 127 is  0
....
```

benbender, 2025-03-23:
Thanks for checking! I've switched browsers (from ungoogled-chromium to latest stock chrome) and tried my second brand new device. Same result. Battery at 3,1V and everything seems to be great. But again, I'm getting stuck at `20:10:24: Clear Ext.OTA region (162kib)...`

Just to be sure, I did the following:
- flashed `ATC_v50.bin` (because of the size-limit)
- reconnected after successful flash
- tried to flash `Z03MMC_zb0004.bin``
- got stuck

(Same result for two devices bought together)

pvvx, 2025-03-23:
There are errors, but they are not related to your problems. Some kind of failure in the github cache and an old firmware indexs file was loaded. I barely fixed it...

---

Checked again:
```
2:52:19: Hardware Revision String: B1.4
22:52:19: Software Revision String: V5.1
22:52:19: Firmware Revision String: github.com/pvvx
22:52:19: Detected custom Firmware
22:52:19: Hardware Version: LYWSD03MMC B1.4, Software Version: 5.1
22:52:19: Custom config HEX string: 55051000064004a9313100e1
22:52:27: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin_zb/Z03MMC_zb0004.bin'...
22:52:27: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin_zb/Z03MMC_zb0004.bin
22:52:27: File size: 165588 bytes
22:52:27: Count: 10350
22:52:32: Start DFU
22:52:34: Clear Ext.OTA region (162kib)...
22:52:45: Clean Flash OTA region: 0x040000, size: 164 kbytes. Go OTA...
22:53:38: Update done after 52.832 seconds
22:53:43: Disconnected.
```

benbender, 2025-03-23:
I did notice the older files and used local files. So no issue there. But yes, its updated but did not changed the result. I also tried to flash directly via github. Same…

Do you know if the difference in the OTA region may be a problem?

pvvx, 2025-03-23:
Zigbee and any file over 128 kilobytes is always OTA written to 0x40000.
The violation may be related to incorrect programming by the programmer of two firmware versions simultaneously in different regions.

benbender, 2025-03-23:
Just to be sure we are on the same page: What is the default way of flashing a new device with a new FW larger 128k? 

I'm running `ATC_v51.bin` the whole time and try to flash a zigbee fw. What should I do differently now?

pvvx, 2025-03-23:
> I'm running `ATC_v51.bin` the whole time and try to flash a zigbee fw. What should I do differently now?

Repeatedly made a firmware through OTA ATC_V51.bin?

benbender, 2025-03-23:
On the first device yes I guess. On the second device I think I've only flashed ATC_V51.bin once. The newer one is the one I'm testing with right now.

pvvx, 2025-03-23:
If the Zigbee and Ble firmware are recorded in a wrong place, for example, at 0x8000 using Tuya Zigbee OTA, they themselves will move to the right addresses. As a result, everything is much more complicated and made to correct user errors.

benbender, 2025-03-23:
I only used the TelinkMiFlasher. I only flashed ATC_V51.bin on this device successfully. Any recommendations what to try next?

pvvx, 2025-03-23:
Flash has the ability to make failures with normal tension if it is recorded at low voltage.
Have you already tried to flash atc_v51.bin again?

benbender, 2025-03-23:
The second device did only run on batteries with voltages above 3V. I did not reflash atc_v51.bin yet. Should I do that next?

The correct way to move forward would be:
- reflash atc_v51.bin
- try to switch to Z03MMC_zb0004.bin

correct?

pvvx, 2025-03-23:
> * reflash atc_v51.bin

benbender, 2025-03-23:
Soooo, finally success!

What worked:

- flash atc_v51.bin 2x times
- flash Z03MMC_v0124.bin

What didn't work:

- flash atc_v51.bin 2x times
- flash Z03MMC_zb0004.bin

(hangs as above)

Thank you again!

pvvx, 2025-03-23:
Your problem is the first in several years.
The reasons are unknown.
Options:
1. Some problems with the BT Adapter
2. Faulty Flash
3. Errors in the program or TelinkMiFlasher.html

benbender, 2025-03-23:
1. and 2. are unlikely imho as flashing atc_v51.bin worked consistently (so no fault of my BT adapter in the macbook air) and the problems were persistent across two different devices (so not a faulty device). 

Just because I'm unsure: "BLE version 5.1" is `atc_v51.bin`, right?

If you want to, you could close this issue but I'm also unsure if you want to track that issue further. If there is anything I can help with/test, I would gladly do so!

pvvx, 2025-03-23:
There are many more nuances. The check that the Flash sector is empty in the SDK for OTA is not fully checked. This can also affect, but doing a full read or full erase of all Flash sectors requires a lot of battery consumption. OTA checks only the first bytes and several others...
As a result, for example, if one or more bytes are written in the middle of the Flash sector, then OTA will not work. Then you will need a hardware firmware with preliminary cleaning of the entire Flash.
This feature is relevant for official Xiaomi firmware and there will be no help.

benbender, 2025-03-23:
So I will close this issue for now. 

Just as a hint: Evertime `OTA region: 0x000000` appeared, the update hang.
The successful updates did start with:

````
OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
````

as expected.

closed 2025-03-23 by benbender

hgontijo, 2025-04-19:
Having similar issue. Unable to flash Zigbee firmware `1141-020a-01263001-Z03MMC.zigbee` on `LYWSD03MMC`.
This device is currently flashed with ATC 5.1.

Before flashing ATC 5.1, last year it was flashed with 4.2:
```
Status: Hardware Version: LYWSD03MMC B2.0, Software Version: 4.2
TelinkMiFlasher.html:2046 Hardware Version: LYWSD03MMC B2.0, Software Version: 4.2

```

Current logs - TelinkMiFlasher.html:
```
8:43:20 AM: Searching for devices
8:43:28 AM: Connecting to: ATC_E427F0
8:43:50 AM: Hardware Revision String: B1.7
8:43:50 AM: Software Revision String: V5.1
8:43:50 AM: Firmware Revision String: github.com/pvvx
8:43:50 AM: Detected custom Firmware
8:43:50 AM: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
8:43:50 AM: Custom config HEX string: 55b70000062804a93131043c
8:45:04 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee'...
8:45:05 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
8:45:05 AM: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01263001
8:45:05 AM: APP Release: 0.1 Build: 2.6, Stack Version: 3.0 Release: 0.1
8:45:05 AM: File size: 131412 bytes
8:45:05 AM: Count: 8214
8:45:11 AM: Start DFU
8:45:14 AM: Clear Ext.OTA region (129kib)...
```

Chrome Console:
```
TelinkMiFlasher.html:2042 Load index file 'firmware.json'...
TelinkMiFlasher.html:2042 {version: 81, betaver: 82, zbjson: 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/firmware_z.json', zbdjson: 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/firmware_zb.json', custom: Array(40), …}
TelinkMiFlasher.html:2042 {version: '0.0.0.5', beta: '0.0.0.5', bin: Array(40), loaded: true}
TelinkMiFlasher.html:2042 {version: '0.1.2.6', beta: '0.1.2.6', bin: Array(40), loaded: true}
TelinkMiFlasher.html:2042 {optionalServices: Array(15), services: Array(11), acceptAllDevices: true}
TelinkMiFlasher.html:2046 Searching for devices
TelinkMiFlasher.html:2042 Status: Connecting to: ATC_E427F0 ...
TelinkMiFlasher.html:2042 Found GATT server
TelinkMiFlasher.html:2042 Services: 00001f10-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 00010203-0405-0607-0809-0a0b0c0d1912
TelinkMiFlasher.html:2042 Services: 0000181a-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 0000fe95-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 0000180a-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Found Telink OTA service
TelinkMiFlasher.html:2042 Found Telink OTA write characteristic
TelinkMiFlasher.html:2042 Found Device Information Service
TelinkMiFlasher.html:2042 Status: Detected custom Firmware
TelinkMiFlasher.html:2042 Found custom Main service (0x1f10)
TelinkMiFlasher.html:2042 Found custom read-write characteristic (0x1f1f)
TelinkMiFlasher.html:2042 Start Notifications (0x1f1f)
TelinkMiFlasher.html:2042 Add EventListener (0x1f1f)
TelinkMiFlasher.html:2042 Read Characteristics (0x1f1f)
TelinkMiFlasher.html:2042 Status: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
TelinkMiFlasher.html:2046 Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
TelinkMiFlasher.html:2046 Custom config HEX string: 55b70000062804a93131043c
TelinkMiFlasher.html:2042 Send cmd (3303): Query 3 measurements
TelinkMiFlasher.html:2042 Send cmd ok
TelinkMiFlasher.html:2042 Vbat: 3157 mV , Temp: 22.21°C, Humi: 66.23%, ID: 10, Switch: Open, Counter: 0, TRG Out: 0, flg: 0x05
TelinkMiFlasher.html:2042 Vbat: 3155 mV , Temp: 22.30°C, Humi: 63.29%, ID: 11, Switch: Open, Counter: 0, TRG Out: 0, flg: 0x05
TelinkMiFlasher.html:2046 Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee'...
TelinkMiFlasher.html:2046 File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
TelinkMiFlasher.html:2042 File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
TelinkMiFlasher.html:2042 File size = 0x20192
TelinkMiFlasher.html:2046 Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01263001
TelinkMiFlasher.html:2046 APP Release: 0.1 Build: 2.6, Stack Version: 3.0 Release: 0.1
TelinkMiFlasher.html:2042 Size in head = 0x20154
TelinkMiFlasher.html:2042 File CRC = 0x30fc6390, Check CRC = 0x30fc6390
TelinkMiFlasher.html:2046 File size: 131412 bytes
TelinkMiFlasher.html:2046 Count: 8214
TelinkMiFlasher.html:2046 Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2042 Status: Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2042 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
TelinkMiFlasher.html:2042 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
```

I also tried `Z03MMC_v0126.bin` but it fails with:
```
Size firmware is more 208 kbytes!
```

Any suggestions?

pvvx, 2025-04-19:
Try to do "Delete all records" before OTA

[image]

hgontijo, 2025-04-19:
Thanks for the quick response. Still no luck after doing `Delete all records`.

I'm using a MacBook Pro M1 + Chrome v135.0.7049.96.
Vbat: 2895 mV (brand new battery)

```
Bluetooth Controller:
  State:	On
  Chipset:	BCM_4378
  Discoverable:	Off
  Firmware Version:	22.1.536.3883
  Product ID:	0x4A01
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	PCIe
  Vendor ID:	0x004C (Apple)
```

```
11:22:22 AM: Searching for devices
11:22:37 AM: Get Advertising MAC (Web Experimental Features!)
11:22:37 AM: Connecting to: ATC_E427F0
11:22:38 AM: MAC: ?
11:22:42 AM: Hardware Revision String: B1.7
11:22:42 AM: Software Revision String: V5.1
11:22:42 AM: Firmware Revision String: github.com/pvvx
11:22:42 AM: Detected custom Firmware
11:22:42 AM: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
11:22:42 AM: Custom config HEX string: 55a70000062804a93131043c
11:23:03 AM: Settings 361234 was sent successfully
11:23:03 AM: Disconnected.
11:23:10 AM: Searching for devices
11:23:24 AM: Get Advertising MAC (Web Experimental Features!)
11:23:24 AM: Connecting to: ATC_E427F0
11:23:26 AM: MAC: ?
11:23:42 AM: Hardware Revision String: B1.7
11:23:42 AM: Software Revision String: V5.1
11:23:42 AM: Firmware Revision String: github.com/pvvx
11:23:42 AM: Detected custom Firmware
11:23:42 AM: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
11:23:42 AM: Custom config HEX string: 55b70000062804a93131043c
11:23:56 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee'...
11:23:56 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
11:23:56 AM: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01263001
11:23:56 AM: APP Release: 0.1 Build: 2.6, Stack Version: 3.0 Release: 0.1
11:23:56 AM: File size: 131412 bytes
11:23:56 AM: Count: 8214
11:24:00 AM: Start DFU
11:24:02 AM: Clear Ext.OTA region (129kib)...
```

Chrome console:
```
TelinkMiFlasher.html:2046 Searching for devices
TelinkMiFlasher.html:2046 Get Advertising MAC (Web Experimental Features!)
TelinkMiFlasher.html:2042 Catching advertisement from "ATC_E427F0"...
TelinkMiFlasher.html:2042 Status: Connecting to: ATC_E427F0 ...
TelinkMiFlasher.html:2042 Received advertisement from "ATC_E427F0"...
TelinkMiFlasher.html:2042 Uint8Array(10) [64, 0, 206, 12, 80, 11, 16, 1, 17, 1, buffer: ArrayBuffer(10), byteLength: 10, byteOffset: 0, length: 10, Symbol(Symbol.toStringTag): 'Uint8Array']
TelinkMiFlasher.html:2046 MAC: ?
TelinkMiFlasher.html:2042 Found GATT server
TelinkMiFlasher.html:2042 Services: 0000fe95-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 00010203-0405-0607-0809-0a0b0c0d1912
TelinkMiFlasher.html:2042 Services: 0000181a-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 0000180a-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Services: 00001f10-0000-1000-8000-00805f9b34fb
TelinkMiFlasher.html:2042 Found Telink OTA service
TelinkMiFlasher.html:2042 Found Telink OTA write characteristic
TelinkMiFlasher.html:2042 Found Device Information Service
TelinkMiFlasher.html:2042 Status: Detected custom Firmware
TelinkMiFlasher.html:2042 Found custom Main service (0x1f10)
[... 32 lines cut ...]
TelinkMiFlasher.html:2042 Start Notifications (0x1f1f)
TelinkMiFlasher.html:2042 Add EventListener (0x1f1f)
TelinkMiFlasher.html:2042 Read Characteristics (0x1f1f)
TelinkMiFlasher.html:2042 Status: Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
TelinkMiFlasher.html:2046 Hardware Version: LYWSD03MMC B1.7, Software Version: 5.1
TelinkMiFlasher.html:2046 Custom config HEX string: 55b70000062804a93131043c
TelinkMiFlasher.html:2042 Send cmd (3303): Query 3 measurements
TelinkMiFlasher.html:2042 Send cmd ok
TelinkMiFlasher.html:2042 Vbat: 2895 mV , Temp: 19.77°C, Humi: 66.47%, ID: 211, Switch: Open, Counter: 0, TRG Out: 1, flg: 0x2F
TelinkMiFlasher.html:2046 Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee'...
TelinkMiFlasher.html:2046 File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
TelinkMiFlasher.html:2042 File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01263001-Z03MMC.zigbee
TelinkMiFlasher.html:2042 File size = 0x20192
TelinkMiFlasher.html:2046 Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01263001
TelinkMiFlasher.html:2046 APP Release: 0.1 Build: 2.6, Stack Version: 3.0 Release: 0.1
TelinkMiFlasher.html:2042 Size in head = 0x20154
TelinkMiFlasher.html:2042 File CRC = 0x30fc6390, Check CRC = 0x30fc6390
TelinkMiFlasher.html:2046 File size: 131412 bytes
TelinkMiFlasher.html:2046 Count: 8214
TelinkMiFlasher.html:2046 Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2042 Status: Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2042 Vbat: 2895 mV , Temp: 19.78°C, Humi: 65.70%, ID: 212, Switch: Open, Counter: 0, TRG Out: 1, flg: 0x2F
TelinkMiFlasher.html:2042 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
TelinkMiFlasher.html:2042 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
```

benbender, 2025-04-19:
(just as a sidenote if it matters: My problem also occured on a M1 Macbook Air…)

pvvx, 2025-04-19:
Error here: "OTA region: 0x020000"
Reason unknown.
Should be: 0x400000

pvvx, 2025-04-19:
Apple uses its own recommended standards for Bluetooth. They do not comply with the generally accepted standards from the [Bluetooth SIG](https://www.bluetooth.com/specifications/specs/).

hgontijo, 2025-04-19:
I did a reset by connecting GND and Reset for 7 secs, then it entered in pair mode (bluetooth sign going on and off on display).
Try zigbee firmware flash again and got
```
Vbat: 2829 mV , Temp: 21.26°C, Humi: 60.81%, ID: 22, Switch: Open, Counter: 0, TRG Out: 0, flg: 0x25
TelinkMiFlasher.html:2053 Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2049 Status: Clear Ext.OTA region (129kib)...
TelinkMiFlasher.html:2049 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
TelinkMiFlasher.html:2049 OTA region: 0x020000, size: 128 kbytes -> Ext.OTA busy
```

What about hard code 0x40000 in [s](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html)?
```
	} else if(blkid == 0x73 && len >= 9) {
		let e = value.getUint8(1);
		let faddr = value.getUint32(2, true);
		let fszk = value.getUint32(6, true); // in kbytes
		let s = 'OTA region: 0x'+hex(faddr,6)+', size: '+fszk+' kbytes';
```

pvvx, 2025-04-19:
If Apple has rewritten the firmware and drivers of the BT adapter to their standards in your device, this means that there may be problems with many BLE devices.
Most likely, you lose connection with the device.
When executing the ext.OTA command, the "Latency" parameter in the connection settings switches from 49 to 3. And the adapter does not understand this...

> What about hard code 0x40000 in [s](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html)?

This is an incorrect decoding of the response. The correct addresses should be output when ext.OTA starts working.
Before that, values ​​from the standard OTA from the SDK are transmitted, purely for information.
I am already rewriting this part in TelinkMiFlasher...
But this is only an unspecified output of information and does not affect the operation of "ext.OTA".

hgontijo, 2025-04-19:
The device still connected to my laptop. Even after getting stuck, the
device showing “_ _ _ <bluetooth sign>”, I can click on “get config” and it
works. This means connection still established between the device and my
laptop.

pvvx, 2025-04-19:
This means that something in javascript doesn't work.

hgontijo, 2025-04-19:
I just got it working in another laptop. 
Chrome `135.0.7049.96`
MacBook Pro 14-inc 2021 M1 Max
```
Bluetooth Controller:
  State:	On
  Chipset:	BCM_4387
  Discoverable:	Off
  Firmware Version:	22.1.598.2575
  Product ID:	0x4A03
  Supported services:	0x392039 < HFP AVRCP A2DP HID Braille LEA AACP GATT SerialPort >
  Transport:	PCIe
  Vendor ID:	0x004C (Apple)
```
Thanks!

pvvx, 2025-04-19:
Beta Ver 5.2: Added additional checks for ext.OTA readiness state transfer.

elhef, 2025-06-09:
Not working on MacBook Pro 16-inch 2020 Intel i9 :/

No problem with a PC (Win 11) with ASUS TUF GAMING B760M-PLUS motherboard with bluetooth.