pvvx/ZigbeeTLc #176: LYWSD03MMC dead after flashing

chrneu, 2025-03-30:
Firstly thanks for the work you have put in the project! :-)

I have 4 pieces of the 03 version and used it the whole time with BLE. Flashing worked fine.

Now I added a zigbee bridge to my homeassistant and wanted to reflash them to zigbee.

3 out of 4 worked perfectly and they could be directly connected to HA. I am wondering now what happened with the last one. 

It flashed to 100%. Then, nothing happend. It still showed the three ---. Then I took out the battery. The display was empty and I put the battery back in but the display still shows nothing. 

I am sure I selected "Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/Z03MMC_v0125.bin'...".

Do you have an idea if I can repair it or is it bricked somehow?

pvvx, 2025-03-31:
If you flash the firmware on a dead battery, failures are possible.

chrneu, 2025-03-31:
Thanks for the addition. Even though it went to 100% it could have been an issue as the battery was used. So I guess to repair it  I would need UART, if that is even possible?

pvvx, 2025-03-31:
The programmer will definitely help.
But it's possible that everything works anyway.
Change the battery.

pvvx, 2025-04-09:
OTA loading from the original version has been adjusted in the [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) . 
Login data from MIHome is required.
After "Login", custom BLE is loaded first.
Then OTA on Zigbee...
If you need to restore the original, only the special OTA built into custom BLE is used.

It is not recommended to use anything other than [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html), since different FW are loaded via OTA to different Flash addresses, using ext.OTA, built only into custom BLE FW and working only with [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)!