pvvx/ZigbeeTLc #177: Updated SGS01Z by mistake to custom firmware

web2brain, 2025-03-31:
Hello,
I got a couple of plant sensors SGS01Z that identify as TS0601 _TZE284_aao3yzhs in ZHA (check 
this link more details about the device: https://github.com/zigpy/zha-device-handlers/issues/3305).

An update showed up and unfortunately I updated it, but it did not work properly afterwards. Trying to add it again to ZHA (without removing it) made it show up in "last seen" at least and I could read some values via Zigbee. Turned out, that it now identifies as Tuya TH03Z-z, so I guess ZigbeeTLc has been flashed on it that I am happy with with other devices. Unfortunately removing it from ZHA and trying to add it again did not work so I am afraid that I bricked this one.

Any idea how to restore the original firmware? I do have an USB-TTL adapter but not the firmware and no details to connect the device. I have a few more so I could try to get the firmware from it, but again do not know how.

I am aware that this was my mistake, but would appreciate some input on how to fix this. And how can I prevent this from happening again? I added "manufacturer_ids: [0x1141]" to the OTA config but am afraid that this will not fix it as the manufacturer probably is 4417 as well.

Kind regards

web2brain, 2025-03-31:
PS: After removing it from ZHA it was not possible to add it again :(

pvvx, 2025-03-31:
Try shorting GPIO_C0 to gnd (-Vbat) for 16 seconds. This will completely reset the bindings.
After that, a new registration is possible.
If a ZTU module is installed there, then its pinout is here: https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp

If this does not work, you will have to rewrite it using a programmer.

pvvx, 2025-03-31:
> And how can I prevent this from happening again? 

* Never update Tuya devices until you are absolutely sure that this firmware is compatible with this device. 
* Tuya devices in most cases have the same OTA ID numbers, and if you can't confirm by the device name, don't update.
* Some Tuya devices have OTA ID numbers that match other manufacturers' models...
* Tuya does not support Zigbee 3.0 formats b uses its own identification for OTA.

web2brain, 2025-03-31:
> Try shorting GPIO_C0 to gnd (-Vbat) for 16 seconds. This will completely reset the bindings.
> After that, a new registration is possible.
> If a ZTU module is installed there, then its pinout is here: [...]

Thanks a lot, i will give it a try and will let you know!

> Is it similar to this variant? https://pvvx.github.io/PLANT-MON/

It looks similar, I'll reconfirm once I've opened it.

web2brain, 2025-03-31:
> This is the Telink manufacturer ID. Tuya has its own number.

I'll verify whether filtering the ota with the telink id will prevent this in the future.

pvvx, 2025-03-31:
Now the device name and manufacturer are entered in text form into the general index.json...

pvvx, 2025-03-31:
In the [general index file](https://github.com/Koenkk/zigbee-OTA/blob/master/index.json) I did not find OTA for "manufacturerCode": 4417, "imageType": 542.
It is ID [1141-021e-01253001-TH03Z.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-021e-01253001-TH03Z.zigbee).

How did you manage to produce OTA?

web2brain, 2025-04-01:
> Is it similar to this variant? https://pvvx.github.io/PLANT-MON/

It looks different:
[image]
[image]
[image]

web2brain, 2025-04-01:
> In the [general index file](https://github.com/Koenkk/zigbee-OTA/blob/master/index.json) I did not find OTA for "manufacturerCode": 4417, "imageType": 542. It is ID [1141-021e-01253001-TH03Z.zigbee](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/1141-021e-01253001-TH03Z.zigbee). And all included from ZigbeeTlc have a signature in index.json: "otaHeaderString": "ZigbeeTLc OTA..."
> 
> How did you manage to produce OTA?

I have [ZigbeeTLc 0.1.2.4 index](https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/index0124.json) for several MiaoMiaoCe MHO-C401N-z and Tuya TS0201-z as an additional OTA source in my home assistant configuration.

web2brain, 2025-04-01:
Wouldn't it be possible to use https://github.com/pvvx/TlsrComProg825x to read the firmware from a non-modified sensor I still have and write this back to the one I accidentally updated?
As I really don't know much about electronics and only have the USB-TTL programmer and a couple of cables (I used https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html before to revive a TS0201): what is the resistor for and can I omit it? Or is there a simple reader analog to the HTML flasher?

Thanks a lot for your help so far, I really appreciate it!

pvvx, 2025-04-01:
> Wouldn't it be possible to use https://github.com/pvvx/TlsrComProg825x to read the firmware from a non-modified sensor I still have and write this back to the one I accidentally updated?

The easiest option for reading Flash:
https://github.com/pvvx/TlsrComSwireWriter -> [TLSR825xComFlasher.py](https://github.com/pvvx/TlsrComSwireWriter/blob/master/TLSR825xComFlasher.py)

Before writing the read FullFlash to another device, you need to change the MAC address.
If you don't understand how - post here the FullFlash you've read. I'll fix it...

web2brain, 2025-04-03:
I will need to get a 1k resistor, I guess there is no possibility to connect it without one? I will get back to you once I manage to find a resistor and the time.

web2brain, 2025-04-13:
With the help of some paperclips and the resistor, I finally managed to copy the firmware from an unused plant sensor using
`python TLSR825xComFlasher.py -p COM3 -t 600 rf 0 0x80000 SGS01Z.bin`

[SGS01Z.zip](https://github.com/user-attachments/files/19724289/SGS01Z.zip)

I attached it to this comment and it would be great if you could change or remove the MAC address. Can I flash the changed file using the web flasher or is the python flasher necessary?

pvvx, 2025-04-13:
There is no MAC/IEEE address in the transferred file.
Apparently, the chip has 1 MB Flash, and the MAC address is at the end of the second part of the Flash.
You can try to write only this half to the device being updated.
(If you erase the entire Flash before flashing, a new MAC/IEEE will be assigned...)

web2brain, 2025-04-29:
I successfully flashed the bricked sensor - just flashing the 512KB was enough and I could use it again.

Thanks a lot for your support @pvvx - I really appreciate it!

Would you be interested in the full 1 MB firmware? Then I could upload it.

pvvx, 2025-04-29:
The biggest problem with the [SGS01](https://developer.tuya.com/en/docs/iot/ZT5-Module-Datasheet?id=Kan388vhu20ar) is that the ZT5 module is not installed according to specification:

[image]

The result is that the signal level is 20 dB lower than, for example, the  [MJWSD06MMC](https://pvvx.github.io/MJWSD06MMC).
20 dB is approximately a 10-fold reduction in communication distance. 6 dB is twice as much.
For a vegetable garden, such an outrage with a 20 dB drop in reception and transmission levels is no good. It can only be used in a flower pot in the same room as the receiver or gateway.

closed 2025-06-18 by web2brain