pvvx/ZigbeeTLc #178: Can't get reset into pairing mode for LYWSD03MMC

Draghmar, 2025-04-01:
Hey. One of my LYWSD03MMC dropped from ZHA so I wanted to reset it into pairing status. From what the doc says, I should short GND and RESET pads for at least 7s. But it doesn't work for me for some reason. Actually nothing happens. I tried 2s for switching between C and F, but it doesn't work as well. So...I don't know what to do. Am I missing something? Maybe I'm using wrong pads? They are visible right away so I don't think so...

[image]

pvvx, 2025-04-01:
If you have removed the thermometer registration from ZHA, then it is completely turned off. It was sent a complete shutdown command. That is, after removing it from the ZigBee network, you must first remove and reinsert the battery.

Then it is possible to close the contacts...

Draghmar, 2025-04-01:
Well...I did remove it from ZHA. I think I did remove the battery although I'm not sure what was the order of the things I've done. Anyway, so the correct procedure is to:

1. Remove dropped entry from ZHA.
2. Remove battery.
3. Insert battery.
4. Short GND and RESET for at least 7s.
5. Hope for the best!

Is that correct?

Draghmar, 2025-04-01:
Ok, but I couldn't remove *working* device as it was dropped from ZHA and I couldn't get it to pairing state. So I removed it when ZHA wasn't able to see it (no data). Before that and after that I tried to remove battery. Then I read about getting those two pads together for 7s. 

But from what you're saying the only thing that is wrong with my ToDo list is switching point 1 with 2 so I'd end up with:
1. Remove battery.
2. Remove dropped entry from ZHA.
3. Insert battery.
4. Short GND and RESET for at least 7s.
5. Hope for the best!

Which for is impossible right now. But do mind that I'm not sure if shorting both pads works for some reason. Docs states that 2s connection should switch between C and F, right? It doesn't happen. I even thought that maybe my cable was broken. But it's not. :(

jperquin, 2025-05-14:
Just to make sure contact with the pads is optimal for a reset, I printed a small template that fits male dupont-connectors..

[image]

dani0219, 2025-06-17:
I have a similar issue. I was using Z2M and somehow the sensor disconnected. I can see the display changing when GND and RESET are shorted, but after the second 'ooo' screen, the Bluetooth logo doesn't show up and the device doesn't really enters pairing modee, hence permit join is enabled in Z2M. Reinserting the battery does nothing. Does anybody have any advice how to proceed?

pvvx, 2025-09-02:
For any ZigBee device, if there are problems with pairing, the steps are approximately as follows:

1. Remove the device registration from the ZigBee device list.
2. After a minute, turn off the ZigBee network. This can be done by rebooting the system.
3. Remove and reinsert the battery in the device.
4. Bring the device to the ZigBee coordinator and short-circuit the contacts or the device button for 15 seconds.
5. Enable the search for new devices in the ZigBee network.

For points 4 and 5, another sequence is possible - 5, then 4.

pvvx, 2025-09-13:
> In Zigbee2Mqtt I found it pretty inconvenient to de-register a device because this will also blacklist the device and prevent it from re-registering.

These are problems from the outside, not the thermometer firmware.

molnarti, 2025-11-01:
pairing these fuckers is indeed a nightmare. i have had some success with a multimeter, that at least shows me the circuit is closed - still it got a lot of attempts to close it at least. the first one i was able to pair, but with the rest no luck, despite the multimeter indicating the circuit is properly closed. 
maybe related to the fact my devices are still on the original Devbis firmware...