pvvx/ZigbeeTLc #179: Sensor values not updating after OTA Update to 0.1.2.5

Noir1992, 2025-04-05:
Hey,
First of all thank you for the great work.

I updated one of my sensors (LYWSD03MMC-z) to 0.1.2.5 and since then, Z2MQTT does Not get any updates from that sensor. I tried to repair it or even to remove it from Z2MQTT but it always fails because it Z2MQTT can't connect to the sensor. I removed and reinstalled the battery multiple times but that did not help. I also tried to get the sensor to repair by shorting GND and RESET for 7 seconds but even though the display showed 3 squares for a few seconds it imidiatly went back to Display the temperature.

My other sensors are still working fine with the old formware.

Anyone else having this problem after the update?

pvvx, 2025-04-05:
[image]

[image]

[image]

sephirothx, 2025-04-05:
Same issue here, using z2m and updating from 0123, 60s measurement interval. I have another one which I haven't updated yet, works well with the exact same settings

[image]

[image]

caiusseverus, 2025-04-05:
I did see this problem initially after update but it's now reporting regularly.

What I did was first take the battery out which made no difference. I then activated pairing in Z2M and shorted the reset pins on the sensor. The log has the message:

[image]

Since then it's been reporting regularly.

L0R3nz, 2025-04-05:
I also had some trouble with sensors. One of them I reset it like @caiusseverus suggested, but second one I leave and After some minutes it start to working, so I recommend to be pacent and then eventually try to reset end connect it again.

pvvx, 2025-04-05:
Since version 0.1.2.5, the SDK on which the firmware is compiled has changed. Therefore, old binding saves do not work. It is necessary to reset the device.
Connection of contacts "reset" and "gnd":
1. for 2 seconds - switching the display of temperature C and F
2. for 7 seconds - reset network bindings
3. for 15 seconds - reset network bindings and all user settings.

pvvx, 2025-04-05:
During the tests I came across a low-quality CR2032 battery.

[image]

At first, registration and operation were normal.
But after a few days, everything fell off and it was impossible to register the device again, although the data was displayed on the display. But when the contacts were closed, the display dimmed, showing that the battery was no longer enough.
Replacing the battery immediately corrected the situation.

sephirothx, 2025-04-05:
Definitely not simply a low-quality battery, I smell a red herring here.

Here's a month of data for my sensor's battery state of charge
* device was running version 0123 before the update
* after the update to 0125, the device went blank
* reset allowed the device to connect to z2m again (but only after removing it first)
* battery wasn't swapped after fw update

[image]

There's definitely something wrong in the new fw, either in sensing the supply voltage or something deeper; please let us know if providing logs or anything could help with the investigation.

My recommendation would be to revert the last changes and figure out the root cause of the issue

ThxAndBye, 2025-04-05:
I had the same issue, reconnecting them to ZHA (removing them in ZHA and then "pressing" the button for 7 sec) solved this for me.

pvvx, 2025-04-05:
> Definitely not simply a low-quality battery, I smell a red herring here.

I have many LYWSD03MMC-z for testing.
And all work successfully.

Your battery graph only says that Zigbee OTA has taken a lot of battery power.
The firmware uses averaging of battery voltage over 512 measurements. With a measurement period of 30 seconds, it will be 4.5 hours, when the battery voltage recovers a little.

pvvx, 2025-04-05:
> (removing them in ZHA and then "pressing" the button for 7 sec) 

[Action of the button](https://github.com/pvvx/ZigbeeTLc?tab=readme-ov-file#action-of-the-button)

pvvx, 2025-04-05:
If you don't make a patch that limits the minimum LongPollControl interval, ZHA will set the device polling to 6 seconds. This will drain the battery in two months.
Why all devices are set to a polling interval of 6 seconds - these are ZHA's problems.

Moreover, ZHA transmits the inclusion of a 6-second polling interval immediately upon biding. At the same time, it does not bother to read the minimum interval value for LongPollControl from the device. There are only two options:
Give an error - the parameter is incorrect, which will generally lead to the suspension of everything in ZHA, or set the period according to the limitation described in the special values ​​of the PollControl cluster.

Malzmeier, 2025-04-06:
After updating the firmware of my devices and removing and re-adding them I thought they were working normal.
But after some time I realised that the temperature isn’t updated as expected.
The temperature on the display shows 21.6 but is not reported to HA. HA still shows 21.4 reported two hours ago. Humidity seems to be updated.
The measurement interval is set to 10s.
I don’t really know what could be wrong or could be done to achieve a timely update in HA.

[image]
[image]

Malzmeier, 2025-04-06:
Ok measurement interval seems to be correct.

I did a reconfiguration and received `measured_value 30/900/50` for temperature and `30/900/100` for humidity.
In my interpretation this config will report a temperature change by 0,5° and humidity by 1%. I read default values should be 30/180/10 and 30/180/50.
If I want to change these values what Attributes in ClusterID: 0x0402 an 0x0405 have to be changed?

pvvx, 2025-04-06:
> If I want to change these values what Attributes in ClusterID: 0x0402 an 0x0405 have to be changed?

ZHA will set them to their values ​​anyway. 
This can only be changed by patching the ZHA files.

[ZHA patch](https://github.com/pvvx/ZigbeeTLc/issues/6)

Malzmeier, 2025-04-06:
I found a source for setting the right values for the reporting intervals using ZHA Toolkit. Now it seems to be working flawlessly.
 However, I need to have a look at the ZHA patch to avoid problems in the future

7w1z4rd, 2025-04-07:
I'm having the same issue with multiple devices + Z2M
Some of them started to work after multiple resets, but not all. I'm using brand new batteries after the OTA.
Also the displayed firmware version remained 0123 for some of them.

pvvx, 2025-04-07:
Added explicit full device reset with reboot when flashing FW with a new SDK version. That is, the device will be as if empty, requiring a new pairing with the coordinator... This was also in the current version, but the reboot was performed only by the "reset" contacts.

pvvx, 2025-04-07:
> (it is also a pita to short it for 7-15 seconds. what are you using for it?)

Is the "measure interval" set to 7-15 seconds?
Minimum "measure interval" is 3 seconds.
Interval for "report" is 0 or 3...65535.

7w1z4rd, 2025-04-07:
> > (it is also a pita to short it for 7-15 seconds. what are you using for it?)
> 
> Is the "measure interval" set to 7-15 seconds? (-> LongPollIntervalMin = 7-15 seconds) Corresponds to LongPollInterval: 12..7208960 qt -> 4..1802240 seconds (qt = 1/4 seconds). Minimum "measure interval" is 3 seconds. Interval for "report" is 0 or 3...65535 seconds.

Sorry, I was referring to shorting the gnd-reset pins.
I had 30-60sec measure intervals.
Finally I've managed to update my 7 devices, and all of them are working now.
Unfortunately I couldn't find a sequence that made them work. There was 1-2 pcs that was working right after resetting it, the others needed multiple resets, and sometimes I've just left it sitting on my desktop, and 0.5-1 hour later the "Reconfigure" action was working.

pvvx, 2025-04-07:
Ver 0.1.2.6. Fix - In some cases the timer did not turn off, which led to increased battery consumption. Changes in PollControl. Version 0.1.2.5 has been removed due to bugs.

kuchar91, 2025-04-08:
there's another 0.1.2.6 v2 version?

WarC0zes, 2025-04-08:
Work fine with 1.2.6 firmware ( 1.2.6_2 ) on Z2M. Thank for the fix.

kuchar91, 2025-04-08:
> Work fine with 1.2.6 firmware ( 1.2.6_2 ) on Z2M. Thank for the fix.

Did you update manual from file or OTA?

WarC0zes, 2025-04-08:
> > Work fine with 1.2.6 firmware ( 1.2.6_2 ) on Z2M. Thank for the fix.
> 
> Did you update manual from file or OTA?

I manually updated on Z2M by ota, because not yet available on Z2M.
https://github.com/Koenkk/zigbee2mqtt/issues/27027#issuecomment-2784534399

pvvx, 2025-04-18:
> Oh, so i should not set the "measurement interval" in Z2M to 60 sec?

Why do you need such a large measurement interval?
This will not result in a significant reduction in battery consumption.
The reporting period in Zigbee is set in the "Report" settings.

wlfbck, 2025-04-18:
>Why do you need such a large measurement interval?
>This will not result in a significant reduction in battery consumption.

I don't really need data more often, so i thought i'd just set the interval to value that is easy to work with. 

>The reporting period in Zigbee is set in the "Report" settings.

Aha, learned something today! I assumed this was the same thing.

closed 2026-04-09 by Noir1992