pvvx/ZigbeeTLc #180: z2m showing wrong version

propi62, 2025-04-06:
after every update some sensors show a wrong version number.
i am pretty sure, the sensors showing the old version were updated succesfull.
repairing them does not solve the problem.

[image]

pvvx, 2025-04-07:
[image]

pvvx, 2025-04-07:
```
zha_toolkit_version: v1.1.27
zigpy_version: 0.77.1
zigpy_rf_version: 0.43.0
ieee_org: update.xiaomi_lywsd03mmc_z_firmware_2
ieee: a4:c1:38:d8:40:5b:e6:8d
command: scan_device
command_data: null
start_time: "2025-04-07T09:05:36.572945+00:00"
errors: []
params:
  dir: 0
  tries: 1
  expect_reply: true
  args: []
  kwargs: {}
  read_before_write: true
  read_after_write: true
scan:
  ieee: a4:c1:38:d8:40:5b:e6:8d
  nwk: "0x9e84"
  model: LYWSD03MMC-z
  manufacturer: Xiaomi
  manufacturer_id: "0x4417"
  endpoints:
    - id: 1
      device_type: "0x0302"
      profile: "0x0104"
      in_clusters:
        "0x0000":
          cluster_id: "0x0000"
          title: Basic
          name: basic
          attributes:
            "0x0000":
              attribute_id: "0x0000"
              attribute_name: zcl_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ
              access_acl: 1
              manf_id: 4417
              attribute_value: 3
            "0x0001":
              attribute_id: "0x0001"
              attribute_name: app_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ
              access_acl: 1
              manf_id: 4417
              attribute_value: 37
            "0x0002":
              attribute_id: "0x0002"
              attribute_name: stack_version
              value_type:
                - "0x20"
                - uint8_t
                - Analog
              access: READ
              access_acl: 1
              manf_id: 4417
              attribute_value: 49
            "0x0003":
              attribute_id: "0x0003"
              attribute_name: hw_version
[... 617 lines cut ...]
              attribute_id: "0x0009"
              attribute_name: minimum_block_req_delay
              value_type:
                - "0x21"
                - uint16_t
                - Analog
              access: READ
              access_acl: 1
              manf_id: 4417
              attribute_value: 0
            "0xfffd":
              attribute_id: "0xfffd"
              attribute_name: cluster_revision
              value_type:
                - "0x21"
                - uint16_t
                - Analog
              access: READ
              access_acl: 1
              manf_id: 4417
              attribute_value: 1
          commands_received: {}
          commands_generated: {}
success: true
```

pvvx, 2025-04-07:
OTA periodic request interval after binding = 15 minutes + random delay.
The OTA periodic query also reports the version number.

propi62, 2025-04-07:
after device resete and repair the version shows correct.

closed 2025-04-12 by propi62