pvvx/ZigbeeTLc #181: LYWSD03MMC-z after flashing Z2M to 1.2.5

pawelgalerczykcat, 2025-04-07:
Hi

I have a problem with one of my devices. I update from 1.2.4 to 1.2.5 with Zigbee2Mqtt OTA, and after that device didn't refresh temp & humidity 

I've tried to reset ( hold longer than 7 sec gnd & rst ) remove the battery, remove from the z2m repair again nothing works :( 

I have connected the device but when I have read the temp:

`zh:ember:ezsp: Received network/route error ROUTE_ERROR_INDIRECT_TRANSACTION_EXPIRY for "60577".`

[image]

Battery is 100% 

How can i downgrade to the previous version in Z2M ? 

I now know that I need to create a custom OTA file for Z2M, but which file and setting do I need to pass for that? 

[image]

On the other hand, another device works fine after reset & repair

littlej956, 2025-04-08:
I updated 2 of my devices, one works correctly but the other one it does not update temp/humi anymore :(

yongliqi, 2025-04-08:
Force upgrade to 1.2.6.

pvvx, 2025-04-09:
> Why 1.2.6 is not in the releases?

Because I alone support over 150 different firmware and device versions (HW). And you don't help, only questions.
And checking new firmware versions is delegated to users. You haven't checked all versions yet and haven't notified that all new versions of ZigbeeTLc and ZBdevice work.

nobwyn, 2025-04-10:
> Force upgrade to 1.2.6.

Can I do this via z2m? My device seems to not respond to z2m requests for new version check …

csutcliff, 2025-04-10:
@pvvx MHO-C401N & MHO-C401 are both working perfectly with 1.2.6 "z" firmware across 12 devices.