pvvx/ZigbeeTLc #182: Can't compile: undefined FLASH_READ_UID_CMD_GD_PUYA_ZB_TH

KipK, 2025-04-08:
I'm trying to compile last version has I have few TS0201 with different pinout. 
Compile faills because of missing definition for FLASH_READ_UID_CMD_GD_PUYA_ZB_TH

```
Building file: src/patch_z_sdk/flash.c
src/patch_z_sdk/flash.c: In function 'flash_read_uid':
src/patch_z_sdk/flash.c:439:14: error: 'FLASH_READ_UID_CMD_GD_PUYA_ZB_TH' undeclared (first use in this function)
src/patch_z_sdk/flash.c:439:14: note: each undeclared identifier is reported only once for each function it appears in
src/patch_z_sdk/flash.c: In function 'flash_read_mid_uid_w
```

pvvx, 2025-04-09:
Updated: Replacing SDK zip.

KipK, 2025-04-09:
Thanks it compiled ok and works all good on TS0201 bguser20 variant after remapping the pinout.  👍

pvvx, 2025-04-09:
[TH03 TS0201(_TZ3000_gohcgzj7, _TZ3000_bguser20): Tuya ZigBee "Wireless Temp & Humidity Sensor"](https://pvvx.github.io/TS0201_TZ3000_TH03/)

https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/1141-d3a3-01993001-TH03Zz_v0126.zigbee

KipK, 2025-04-09:
Another tuya mess. Mine use CHT8305, no LCD.
Here is my pinout:

https://github.com/KipK/ZigbeeTLc/blob/newbguser20/src/board_ts0201_tz3000.h