pvvx/ZigbeeTLc #186: OTA for TS0201 _TZ3000_fllyghyj to ZigbeeTlc

web2brain, 2025-04-13:
Hello,
I have a TS0201 _TZ3000_fllyghyj that I would like to convert to ZigbeeTlc but including `https://github.com/pvvx/ZigbeeTLc/raw/refs/heads/master/bin/ts0201_tuya2z.json` for OTA updates does not result in an update that is offered. Is there a way to convert it to ZigbeeTlc via OTA or do I have to flash it? If the latter, which one would be the correct firmware? I guess `ZTS0201Z3000_v0126.bin`?

I would like to avoid another #177 :)

web2brain, 2025-04-16:
The file `ts0201_tuyaz.json` referencing `1141-d3a3-01983001-TS0201z_v0126.zigbee` was the correct one, flashing worked fine. Thanks!

closed 2025-04-16 by web2brain

antt0n, 2025-06-16:
Hello @web2brain !
Sorry to dig up this issue, but I have the exact same model and am currently looking to flash ZigbeeTLc on it.
How did you finally manage to flash from the Z2M OTA?
Thanks in advance!