pvvx/ZigbeeTLc #187: LYWSD03MMC-z TABLE_FULL

CZonin, 2025-04-13:
Recently updated all of my sensors from 0.1.2.3 -> 0.1.2.5 -> 0.1.2.6 and I've been seeing the following error for all of them. I saw this issue that references the same error: https://github.com/pvvx/ZigbeeTLc/issues/114

I tried doing a hard reset on 2 of them by connecting "reset" to "gnd" for 15 seconds and then restarting both my coordinator and Z2M, and then re-adding the sensor, but the issue persisted.

```
error:    z2m: Failed to configure 'Nursery Climate', attempt 3 (Error: Bind 0xa4c138f44b6aed67/1 msTemperatureMeasurement from '0x9035eafffec39038/1' failed (Status 'TABLE_FULL')
    at Endpoint.bind (/app/node_modules/.pnpm/zigbee-herdsman@3.4.11/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:574:23)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at setupAttributes (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:158:13)
    at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:209:17)
    at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@23.20.1/node_modules/zigbee-herdsman-converters/src/index.ts:363:21)
    at Configure.configure (/app/lib/extension/configure.ts:124:13)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))
```

pvvx, 2025-04-14:
> I also can't enable BLE ota mode.

BLE works only in [ZBdevice](https://github.com/pvvx/ZigbeeTLc/blob/master/ZBdevices.md) firmware. And it has version 0.0.0.5.

RobTS, 2025-04-22:
Same here, can't even change reporting times any more on new, 1.2.6 flashed out of the box thermometers:

[image]

pvvx, 2025-04-22:
Read the "binding" table from the device and show here how many "bindings" you have.

pvvx, 2025-04-22:
The firmware uses a maximum of 8 reports (bindings) in the table. Of these, 4 are never used.
Because there is no point in binding to immutable data in clusters.

pvvx, 2025-04-22:
> Is there a specific command to send for z2m to get this report?

I don't use z2m. Ask about it on specialized or educational resources on z2m.

mrmaximas, 2025-04-22:
@RobTS what version of the coordinator are you using?

VivantSenior, 2025-04-24:
> Is there a specific command to send for z2m to get this report? This is what I see atm:
> 
> <img alt="Image" width="1703" src="https://private-user-images.githubusercontent.com/1608080/435930089-f909f34f-1ece-4a00-aa16-4c03ed7acc74.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU0ODgxMTQsIm5iZiI6MTc0NTQ4NzgxNCwicGF0aCI6Ii8xNjA4MDgwLzQzNTkzMDA4OS1mOTA5ZjM0Zi0xZWNlLTRhMDAtYWExNi00YzAzZWQ3YWNjNzQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MjRUMDk0MzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWNjZmQxNGYwNDc3NTU3MjFlZDFjNDNiMTI5MDAxNDEwNDZkN2ZmMTU4ZDg4M2VhY2IzZDg4NDdjNmEyMThhNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.jJ13jce9uiq6WLlZ-8lkt3VnssZEukCqKVljQOzKabo"> <img alt="Image" width="1692" src="https://private-user-images.githubusercontent.com/1608080/435930090-07059c86-42fc-425d-90c9-cae32a2ed231.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDU0ODgxMTQsIm5iZiI6MTc0NTQ4NzgxNCwicGF0aCI6Ii8xNjA4MDgwLzQzNTkzMDA5MC0wNzA1OWM4Ni00MmZjLTQyNWQtOTBjOS1jYWUzMmEyZWQyMzEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDQyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTA0MjRUMDk0MzM0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2Q4Y2E5NWM0MWYyODc3NjlhZWQ0MmMyZDYwNjY2NjBhNzUzY2I1NzI0OWJkYmRjNzNlNzYyM2I5NjRmNWFmZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.7LN07DtqNrJNG1WMDrc3Y0EwlZL89T_hTKan9DnDLZU">

I had similar issue, two steps helped in my case:
1. Unbind all existing bindings:
[image]
2. Force Interview / Reconfigure (don't recall which one)

If second step won't help try to do a hard reset (enable join and 15 seconds GND + RST)

closed 2025-06-09 by CZonin